SpinalHDL/SpinalHDL #1137: axi4Crossbar address out of range

ZhengJuzhong666, 2023-06-08:
Hello! I try to connect an axi bus to the slave side of a AxiCrossbar.  The address width is 32 bits.  When I allocate 2GiB of address space to this axi bus, there is no problem. It ranges from 0x80000000 to 0xFFFFFFFF.  But when I want to allocate 2560 MiB of address space from 0x60000000 to 0xFFFFFFFF for it, I get the “OUT OF RANGE CONSTANT” error.
This is my code：
~~~scala
  axi4Crossbar.addSlaves(
    io.highSpeedBus -> (0x60000000l, 2560 MiB)
  )
~~~

And this is the error：
[image]

When I change the range to 2559MiB, it works.  I don't understand why this problem occurs in AxiCrossbar, Could you help me? Thank you！

ZhengJuzhong666, 2023-06-08:
Thank you， My SpinalHDL version is 1.8.0

Dolu1990, 2023-06-08:
I think you need to update to 1.8.1, the fix is likely that one : 
https://github.com/SpinalHDL/SpinalHDL/commit/018e3128e4dec180e9faf4c7c35998d4133657cb

ZhengJuzhong666, 2023-06-08:
Thank you！I will do the updata.

closed 2023-06-08 by ZhengJuzhong666