SpinalHDL/SpinalHDL #1142: create more analog signal than need

yportne13, 2023-06-14:
```scala
class InOut extends Component {
  val b = new MyBlack
  val a = new MyBlack
  val c = new MyBlack
  val ret = Analog(Bits(2 bit)) 
  b.io.x := ret
  c.io.x := ret
  a.io.x := ret
}
class MyBlack extends BlackBox {
  val io = new Bundle {
    val x = inout(Analog(Bits(2 bits)))
  }
}
```
in this example, i have create a signal **ret**, but spinal still create a new signal **analog_wrap_0**, and use **assign** to connect, which is illegal for some synthesis tool:
```verilog
assign ret[1 : 0] = analog_wrap_0[1 : 0];
```
if i do not create the signal on my own, then there is no way to change the signal name **analog_wrap_0**
maybe this can be solved by:
in **phase.scala: 387**: **val count = filtred.count(e => e.isInOut && e.component == c)**
change e.isInOut to e.isAnalog

yportne13, 2023-06-15:
change e.isInOut to e.isAnalog will leads to **MULTIPLE INOUT** bug
ignore this, on official 1.8.2 version, i found another bug:
```scala

class InOut extends Component {
  val io = new Bundle {
    val x = inout(Analog(Bits(2 bits)))
    val xx = inout(Analog(Bits(2 bits)))
  }

  val b = new MyBlack
  val a = new MyBlack
  val c = new MyBlack
  b.io.x := io.xx
  c.io.x := io.xx
  a.io.x := io.xx
  b.io.x := io.x
  c.io.x := io.x
  a.io.x := io.x

}
class MyBlack extends BlackBox {
  val io = new Bundle {
    val x = inout(Analog(Bits(2 bits)))
  }
}

object top extends App {
  SpinalVerilog(new InOut)
}
```
the result is:
```verilog
module InOut_1 (
  inout      [1:0]    io_x,
  inout      [1:0]    io_xx
);

  MyBlack b (
  );
  MyBlack a (
  );
  MyBlack c (
  );

endmodule
```

Dolu1990, 2023-06-15:
Hi,

```scala
  val io = new Bundle {
    val x = inout(Analog(Bits(2 bits)))
    val xx = inout(Analog(Bits(2 bits)))
  }

  val b = new MyBlack
  val a = new MyBlack
  val c = new MyBlack
  b.io.x := io.xx
  c.io.x := io.xx
  a.io.x := io.xx
  b.io.x := io.x
  c.io.x := io.x
  a.io.x := io.x
  ```
  
  You can't realy have that in VHDL, and i guess verilog ?
  
  The main challange with tristate stuff is to elect one true signal which will host all the tristate stuff.
  
  Issue if you connect two inout which are io of the toplevel is that both want to host it.
  
  How would you want the verilog of that code example to be ?
  
  (main issue i think is io.x connected to io.xx)

yportne13, 2023-06-16:
i don't know, maybe the second case is also no possible to write in verilog, but if so, maybe spinal should raise error instead of generate this strange code?
for the first case, some synth tool will insert digital buffer to this analog signal when meet with this assign, so i want to remove this useless analog_wrap. for my first try, i change isInOut to isAnalog in `phase.scala: line 387 and line 417`, it works fine in this simple case , but i found that a lot of other case will goes wrong, for example:
```scala
class InOut extends Component {
  val x = inout(Analog(Bits(3 bits))
  val a = new MyBlack
  a.io.x := x(1 downto 0)
  x(2) := True
}
class MyBlack extends BlackBox {
  val io = new Bundle {
    val x = inout(Analog(Bits(2 bits)))
  }
}
```
when phase check, there will be one more signal [???(2 bits)] which will lead to `MULTIPLE INOUT` error, then i have no idea how to solve this problem

Dolu1990, 2023-06-16:
>  maybe spinal should raise error instead of generate this strange code?

Yes :)

> when phase check, there will be one more signal [???(2 bits)] which will lead to MULTIPLE INOUT error, then i have no idea how to solve this problem

Hmm i have to investigate it.

Dolu1990, 2023-06-16:
I just tried your last example on spinalhdl dev, seems fine : 

```scala
module miaou (
  inout      [2:0]    x
);

  MyBlack a (
    .io_x ({x[1 : 0]}) 
  );
  assign x[2] = 1'b1 ? 1'b1 : 1'bz;

endmodule
```

can you try with SpinalHDL 1.8.2 ?

Dolu1990, 2023-06-16:
The "val count = filtred.count(e => e.isInOut && e.component == c)" fix unfortunatly break some cases (detected by AnalogConnectionTest)

I agree that in some case, the current implementation will generate some additional signals (to have a analog aggregation signal), which are unecessary. Not sure it is realy worth trying to improve, as it will have no synthesis impact and Analog stuff is kind of very corner case ?

yportne13, 2023-06-16:
when you write such a code in verilog:
```verilog
//rtl.v 
assign  analog_a   =  analog_b  ;
```
it is possible to get something looks like this after synthesis:
```verilog
//netlist.v (DC/RC_SYNC)
BUFFD4BWP169H3P48CPDSVT  u0 (
  .I(analog_b),
  .Z(analog_a)
) ; 
```
a digital buffer on analog
so, a lot of company do not allow coder to write `assign` on a analog signal

Dolu1990, 2023-06-16:
> assign  analog_a   =  analog_b  ;

Agree, should never be done.
SpinalHDL is designed to avoid it in verilog. Have you seen SpinalHDL emiting such broken verilog ?

yportne13, 2023-06-19:
> Have you seen SpinalHDL emiting such broken verilog ?

yes, look at the beginning of this issue, this case generate such a line

> ```verilog
> assign ret[1 : 0] = analog_wrap_0[1 : 0];
> ```

Dolu1990, 2023-06-19:
in that example, i think that SpinalHDL transformed ret into a read only signal. So ret isn't realy a analog thing anymore.
analog_wrap_0 was "elected" as host of the analog connection.