SpinalHDL/SpinalHDL #1143: Is the selIndex in A庞博

ZhengJuzhong666, 2023-06-19:


ZhengJuzhong666, 2023-06-19:
I noticed that the selIndex variable here uses a register. I wanna know why need to use registers instead of direct combinatorial  logic here.  I think index will change the PSEL signal synchronously.  The current implementation looks like if the selected slave's PREAD signal is affected by other slaves in the first cycle.
~~~ scala
  for((output,index) <- io.outputs.zipWithIndex){
    output.PADDR   := io.input.PADDR
    output.PENABLE := io.input.PENABLE
    output.PSEL(0) := io.input.PSEL(index)
    output.PWRITE  := io.input.PWRITE
    output.PWDATA  := io.input.PWDATA
  }

  val selIndex = RegNext(OHToUInt(io.input.PSEL))
  io.input.PREADY := io.outputs(selIndex).PREADY
  io.input.PRDATA := io.outputs(selIndex).PRDATA
~~~

andreasWallner, 2023-06-19:
When you look at the APB timings:
[image]
you can see that both PREADY and PRDATA are only valid one cycle into the transaction. The master has to ignore the state of those lines during the initial cycle. The router can therefore split the combinatorial path from address decode, SEL generation (in ApbDecoder) and mux'ing the response signals with the register stage that you mention.

closed 2023-06-20 by ZhengJuzhong666

ZhengJuzhong666, 2023-06-20:
Thank you! Since the other slaves in my Router return valid PREADY signals during the first cycle, it seems that I should modify the Slave side code.