SpinalHDL/SpinalHDL #1144: Current StreamFifo not suitable for high clock rates, propose StreamFifoHighPerformance

likewise, 2023-06-19:
1. The `StreamFifo` (SpinalHDL 1.8.1) has substantial logic in it's `full` and `empty` signals (addition, followed by comparison in the read/write pointer vector width) which are combinatorial signals that drive the `ready` output combinatorial.
2. The backing RAM is not registered on read. Although this can be mitigated by manually adding a `.s2mPipe().m2sPipe()` (also know as a dual skid buffer), the first item remains the critical path.
3. On ICE, the above limits the clock rate to 50 MHz, as reported by andreasWallner.
4. On US+, the clock rate is limited to 400 MHz.

I am aware and have experience with several high-speed, well-designed FIFOs that are low resource count, and high clock speed. Their worst case latency might be one clock more, but this is of little concern (at least to me).

I propose we develop StreamFifoHighPerformance, I can start with this somewhere next week.

I propose we port Dan Gisselquist's FIFO to SpinalHDL. Mr Gisselquist has kindly provided his FIFO design to the public domain.

I am attaching his Verilog source code obtained from https://gitlab.com/ZipCPU/wb2axip/-/raw/formal/rtl/sfifo.v on Monday 19th June 2023, which has the following header:

```
// Written and distributed by Gisselquist Technology, LLC
// }}}
// This design is hereby granted to the public domain.
// {{{
// This program is distributed in the hope that it will be useful, but WITHOUT
// ANY WARRANTY; without even the implied warranty of MERCHANTIBILITY or
// FITNESS FOR A PARTICULAR PURPOSE.
```
and had to rename it from `sfifo.v` to `sfifo.v.txt.` for the issue tracker to accept it.

The FIFO has formal testing, which probably matches ours.

[sfifo.v.txt](https://github.com/SpinalHDL/SpinalHDL/files/11788106/sfifo.v.txt)

likewise, 2023-06-19:
For high-performance, instead of:
```
		parameter [0:0]	OPT_ASYNC_READ = 1'b1,
```
I would like to target
```
```
		parameter [0:0]	OPT_ASYNC_READ = 1'b0,
```
so that the read data from RAM is registered. (This is requires to use block RAM on ICE, and is required for high-speed in general.)

Dolu1990, 2023-06-19:
Sure, alternative implementation could be added ^^
Mainly : 
- 2 cycle ram read
- always power of 2 depth, allowing faster logic (current implementation allow non power of 2)

>  the first item remains the critical path.

Even with an output s2mPipe the timings aren't good ?
I mean that should cut the read path.

How big is your test fifo ?

likewise, 2023-06-19:
The main (remaining) issue which limits performance is not (only) the 2-cycle RAM read, and power of two. The remaining issue after that will be an adder and compare in one path, which is composed of `StreamFifo` and `CounterUpDown` logic:

```
    io.pop.valid := !empty & !(RegNext(popPtr.valueNext === pushPtr, False) & !full) //mem write to read propagation
```
and
```
  if (isPow2(stateCount)) {
    valueNext := (value + finalIncrement).resized
```
and a non-registered but combinatorial compare result on the output path:
```
    val ptrMatch = pushPtr === popPtr
    val empty = ptrMatch & !risingOccupancy
    val full = ptrMatch & risingOccupancy
```

likewise, 2023-06-19:
Also, I would be happy to port the `sfifo.v` design later this week, I am not expecting anyone else to pick this up per se.
This issue is just to track it, as I was not the only person to see the critical path as worst path in their design.

Dolu1990, 2023-06-20:
Ahhh i see,
So, the Counter used in the current fifo design could avoid the value + finalIncrement by insted using a "enable" on the value register to sample an "always +1 adder"

This would cut the path quite well i think. Should be good then it think ? (assuming using s2mPipe / m2sPipe on the pop port)

> and a non-registered but combinatorial compare result on the output path:

I think that one is fully solved by using a s2mPipe on the push side.

Dolu1990, 2023-06-20:
What FPGA are you on / what frequency are you targeting ?

likewise, 2023-06-21:
Xilinx "xcu50-fsvh2104-2-e" which is Alveo U50 FPGA (Xilinx Ultrascale+), my FIFO needs to run at 500 MHz at least, but I typically design for 20% extra speed to have some slack in P&R.

`sfifo.v` reaches 700 MHz in 128 bit wide x 128 deep using LUTRAM
`sfifo.v` reaches 600 MHz in 512 bit wide x 512 deep using 2 BRAMs

likewise, 2023-06-21:
> If the critical path of StreamFIFO is the path from addr to full or empty, then how about just import a register for full and empty? Just like what sfifo did? Maybe there would not that many changes are required.

Yes, it might be as easy as that, but the difficult cases are (always) around the situation when the FIFO becomes empty or full, and if you want to allow a 2^N fully filled FIFO. And in the case of the fully filled FIFO, if you want to support simultaneous read/write. (I do not care about the latter, a FIFO should typically be never full ideally.)

It is for this reason I wanted to make formal work first :-)

likewise, 2023-06-21:
In the `sfifo.v` design, a bypass is used, but even this is not needed. The tutorial describes a second approach on page 65 of https://zipcpu.com/tutorial/lsn-10-fifo.pdf.

Dolu1990, 2023-06-23:
Got the design which can scale from 0, 1, 2 cycle latency : 

```scala
    assert(isPow2(depth))

    val ram = Mem(dataType, depth)

    val ptr = new Area{
      val push = Reg(UInt(log2Up(depth)+1 bits)) init(0)
      val pop  = Reg(UInt(log2Up(depth)+1 bits)) init(0)
      val empty = push === pop
      val full = (push ^ pop ^ depth) === 0
    }

    val push = new Area {
      io.push.ready := !ptr.full
      val bypassed = False
      when(io.push.fire && !bypassed) {
        ptr.push := ptr.push + 1
      }
      val write = ram.writePort()
      write.valid := io.push.fire
      write.address := ptr.push.resized
      write.data := io.push.payload
    }

    val pop = new Area{
      val addressGen = Stream(ram.addressType)
      addressGen.valid := !ptr.empty
      addressGen.payload := ptr.pop.resized

      when(addressGen.fire){
        ptr.pop := ptr.pop + 1
      }

      val sync = !withAsyncRead generate new Area{
        val readArbitation = addressGen.m2sPipe(flush = io.flush)
        val readPort = ram.readSyncPort
        readPort.cmd := addressGen.toFlowFire

        io.pop << readArbitation.translateWith(readPort.rsp)
      }

      val async = withAsyncRead generate new Area{
        io.pop << addressGen.translateWith(ram.readAsync(addressGen.payload))

        if(withBypass){
          when(ptr.empty){
            io.pop.valid := io.push.valid
            io.pop.payload := io.push.payload
            push.bypassed := io.pop.ready
          }
        }
      }
    }

    io.occupancy := ptr.push - ptr.pop
    io.availability := depth - io.occupancy

    when(io.flush){
      ptr.push := 0
      ptr.pop := 0
    }
``` 

using val withAsyncRead : Boolean = true, val withBypass : Boolean = true parameters for 0 latency

Dolu1990, 2023-06-23:
I pushed an implementation proposal, StreamFifoV2

Note that the occupancy behaviour isn't exactly the same than the original StreamFifo, so it isn't necessarly a drop in replacement

likewise, 2023-06-23:
Thanks, I am also working on a sfifo.v port. I will share it soon.

Let's then evaluate what is the best way to get one stabilized later.

Dolu1990, 2023-06-23:
@likewise ok :)

In case of, i was using https://github.com/SpinalHDL/SpinalHDL/blob/855c5df41328893cdf380b98b879610335471795/lib/src/main/scala/spinal/lib/eda/bench/Bench.scala#L106

to check the FMax / area usage.

It works using some environnement variable. For instance : 
export QUARTUS_CYCLONE_IV_BIN="/home/rawrr/tools/intelFPGA_lite/19.1/quartus/bin"
export QUARTUS_CYCLONE_V_BIN="/home/rawrr/tools/intelFPGA_lite/19.1/quartus/bin"
export VIVADO_ARTIX_7_BIN="/media/data/Xilinx/Vivado/2022.2/bin"

Dolu1990, 2023-06-23:
@likewise I pushed a fix which now make that StreamFifoV2 behave exactly like the original stream fifo
All regression tests passed.

Also, i improved the test bench to also check the flush / occupancy / availabilty values : 
https://github.com/SpinalHDL/SpinalHDL/commit/342b6afb49abbc8a0dd34ccf9c13e20400be2728#diff-2354fc9d9bbf3f1844752c220549fe6690c15b7dc82105d18a2b66a80aaae585R43

likewise, 2023-06-23:
@Dolu1990 

Wow, impressive work. Thanks.

```scala
      val push = Reg(UInt(log2Up(depth)+1 bits)) init(0)
      val pop  = Reg(UInt(log2Up(depth)+1 bits)) init(0)
```
```scala
      val empty = push === pop
```
In my design, I still have `empty` registered, but I realized this makes the design complex, as there is a 2-cycle latency on the memory reads, so `popping` will be 2 cycles late (and the same address will be read twice if incrementing `pop(Ptr)` on `popping`). Instead the `pop(Ptr)` must be incremented on RAM reads rather than StreamFifo pops. So I was thinking of moving the 2-cycle latency FIFO into a 1-cycle latency skid buffer first, and only reading the RAM when there is enough room in the skid buffer. 

```scala
      val full = (push ^ pop ^ depth) === 0
```
Clever. I can see this works for power of 2 FIFO sizes, but I do not understand why it works for non-power of 2 FIFO sizes.
I trust formal verification, I just cannot get my brains to understand this.

Why is the `AddressGen` needed? Is this your solution to deal with the 2-cycle memory latency?

Dolu1990, 2023-06-24:
Thanks :)

> In my design, I still have empty registered, but I realized this makes the design complex, as there is a 2-cycle latency on the memory reads

On its default configuration (2 cycle latency, with block ram) It is also registred (empty not exposed to the io.pop port). Empty is used as valid of the addressGen stream, then this goes into a Stream m2sStage, then goes to io.pop.

> I can see this works for power of 2 FIFO sizes, but I do not understand why it works for non-power of 2 FIFO sizes.

If for non power of two, the pop / push had extra msb bit which toggle on overflow, that would work too.
The alternative is to have a "wentUp" register to distinguish if the fifo is full or empty when push and pop match
full := push === pop && wentUp

> Why is the AddressGen needed? Is this your solution to deal with the 2-cycle memory latency?

yes, that's mostly for ram blocks support with 2 cycle memory latency. But the good thing is that this way of doing things also work well with 0 and 1 latency design.

likewise, 2023-06-24:
Nice. Here is what I tried so far: My attempt was different, a RAM FIFO of 2^N, followed by two buffers `a` and `b` that should take care of the two-cycle read latency.

My ideas was: Calculate the room in `a` and `b`, as well as the number of outstanding memory reads, only read memory once there is room to store the read result in `a` or `b` (or if io.pop.ready, as then `a` will be ready.)

It is failing in formal verification still, but I know the formal verification is yet unaware of `a` and `b` buffers.

`mem_read` has 6 boolean inputs, so should map nicely to LUT6.

```scala
class StreamFifoHighPerformance[T <: Data](val dataType: HardType[T], val depth: Int, performance: Boolean = true) extends Component {
  require(depth >= 0)
  val io = new Bundle {
    val push = slave Stream (dataType)
    val pop = master Stream (dataType)
    val flush = in Bool() default(False)
    val occupancy    = out UInt (log2Up(depth + 1) bits)
    val availability = out UInt (log2Up(depth + 1) bits)
  }

  val bypass = (depth == 0) generate new Area {
      io.push >> io.pop
      io.occupancy := 0
      io.availability := 0
    }
  val oneStage = (depth == 1) generate new Area{
      io.push.m2sPipe(flush = io.flush) >> io.pop
      io.occupancy := U(io.pop.valid)
      io.availability := U(!io.pop.valid)
    }

  val logic = (depth > 1)  generate new Area {

  // fully registered FIFO logic
  // This is based on Dan Gisselquists' sfifo.v, public domain 2023-06-22
    require(isPow2(depth))
    // 1 bit extra, this allows to distinguish with empty/full easily
    // NOTE that we have to resize back to log2Up(depth) for pointer use
    val ptrWidth = log2Up(depth) + 1
    val ram = Mem(dataType, depth)
    // (matches: sfifo.v wr_addr)
    val pushPtr = Reg(UInt(ptrWidth bits)) init(0)
    // (matches: sfifo.v rd_addr)
    val popPtr = Reg(UInt(ptrWidth bits)) init(0)

    val mem_out = Stream(dataType)

    val pushing = io.push.fire
    val popping = mem_out.fire
    // (matches: sfifo.v r_full)
    val full = RegInit(False)
    // (matches: sfifo.v r_empty)
    val empty = RegInit(True)
    val empty_d1 = RegNext(empty) init(True)

    val pushing_d1 = RegNext(pushing) init(False)
    val popping_d1 = RegNext(popping) init(False)

    val filled = Reg(UInt(ptrWidth bits)) init(0)

    val a = Stream(dataType)
    val b = Stream(dataType)

    io.push.ready := !full

    // indicates if we read (a valid word from) memory in current cycle
    val mem_read = Bool()
    // keep track of past outstanding valid reads
    val mem_read_d1 = RegNext(mem_read) init(False)
    val mem_read_d2 = RegNext(mem_read_d1) init(False)

    // room after the FIFO, in the skid buffers 'a' and 'b', 1 extra if streaming
    val room = U(io.pop.ready).resize(2) + U(!a.valid).resize(2) + U(!b.valid).resize(2)
    // inflight (outstanding) memory reads
    val inflight = U(mem_read_d1).resize(2) + U(mem_read_d2).resize(2)
    // how many skid buffers have no matching outstanding memory read
    val unreserved = room - inflight
    
    mem_read := (unreserved > 0) && !empty

    mem_out.valid := Delay(mem_read, 2, init=False)
    mem_out.payload := RegNext(ram.readSync(popPtr.resized))

    a << mem_out.m2sPipe().s2mPipe()
    b << a.m2sPipe().s2mPipe()
    io.pop << b

    // update fill level of RAM
    when (!pushing & mem_read) {
      filled := filled - 1
    } elsewhen (pushing & !mem_read) {
      filled := filled + 1
    } otherwise {
      filled := pushPtr - popPtr
    }

    // update full flag of RAM
    when (!pushing & popping) {
      full := False
    } elsewhen (pushing & !popping) {
      // filled with one empty entry left?
      full := (filled === U(depth - 2, ptrWidth bits))
    } otherwise {
      // completely filled
      full := (filled === U(depth - 1, ptrWidth bits))
    }

    // update write address pointer
    when(pushing) {
      pushPtr := pushPtr + 1
    }
    // write to memory
    when(pushing) {
      ram(pushPtr.resized) := io.push.payload
    }

    // update read address pointer
    when(mem_read) {
      popPtr := popPtr + 1
    }

    // update empty flag of RAM
    when (!pushing & mem_read) {
      empty := (filled <= 1)
    } elsewhen (pushing & !mem_read) {
      empty := False
    }

    io.occupancy := RegNext(pushPtr - popPtr)
    io.availability := RegNext(popPtr - pushPtr)

    when(io.flush){
      pushPtr := 0
      popPtr := 0
      filled := 0
      full := False
      empty := True
    }    
  }
  ```

Also, I am figuring out how to make formalCheck() also deal with checking the conditions in buffers `a` and `b`, that should allow some formal checking.

Dolu1990, 2023-06-24:
Ahhh for the updown counter, you may get better performance with things like : 

```scala
      val internalOccupancy = Reg(UInt(log2Up(depth+1) bits)) init(1 << log2Up(depth+1)-1)
      internalOccupancy := (doPush ## doPop).mux(
        0 -> internalOccupancy,
        1 -> KeepAttribute(internalOccupancy + 1),
        2 -> KeepAttribute(internalOccupancy - 1),
        3 -> internalOccupancy
      )
      when(io.flush) { internalOccupancy := 1 << log2Up(depth+1)-1 }
      empty := internalOccupancy.msb
      
      Basicaly, using the MSB of the counter as a empty / full
```

Dolu1990, 2023-06-24:
Hmm seems like having up down counters with used as full / empty will provide the best timings possible.

likewise, 2023-06-24:
> Hmm seems like having up down counters with used as full / empty will provide the best timings possible.

I do not understand this part of the sentence: "with used as full / empty". Can you please elaborate?

likewise, 2023-06-24:
This is my latest status. It does not pass formal yet, and I am trying to find out if my design is buggy, or formal is incorrect.
I modified `formalCheck` to deal with my `a` and `b` buffers.

```scala

object StreamFifoHighPerformance{
  def apply[T <: Data](dataType: T, depth: Int) = new StreamFifoHighPerformance(dataType,depth)
}

class StreamFifoHighPerformance[T <: Data](val dataType: HardType[T], val depth: Int) extends Component {
  require(depth >= 0)
  val io = new Bundle {
    val push = slave Stream (dataType)
    val pop = master Stream (dataType)
    val flush = in Bool() default(False)
    val occupancy    = out UInt (log2Up(depth + 1) bits)
    val availability = out UInt (log2Up(depth + 1) bits)
  }

  val bypass = (depth == 0) generate new Area {
      io.push >> io.pop
      io.occupancy := 0
      io.availability := 0
    }
  val oneStage = (depth == 1) generate new Area{
      io.push.m2sPipe(flush = io.flush) >> io.pop
      io.occupancy := U(io.pop.valid)
      io.availability := U(!io.pop.valid)
    }

  val logic = (depth > 1)  generate new Area {

  // fully registered FIFO logic
  // This is based on Dan Gisselquists' sfifo.v, public domain 2023-06-22
    require(isPow2(depth))
    // 1 bit extra, this allows to distinguish with empty/full easily
    // NOTE that we have to resize back to log2Up(depth) for pointer use
    val ptrWidth = log2Up(depth) + 1
    val ram = Mem(dataType, depth)
    // (matches: sfifo.v wr_addr)
    val pushPtr = Reg(UInt(ptrWidth bits)) init(0)
    // (matches: sfifo.v rd_addr)
    val popPtr = Reg(UInt(ptrWidth bits)) init(0)

    val mem_out = Stream(dataType)

    val pushing = io.push.fire
    val popping = mem_out.fire
    // (matches: sfifo.v r_full)
    val full = RegInit(False)
    // (matches: sfifo.v r_empty)
    val empty = RegInit(True)
    val empty_d1 = RegNext(empty) init(True)
    //val empty_d2 = RegNext(empty_d1) init(True)

    val pushing_d1 = RegNext(pushing) init(False)
    val popping_d1 = RegNext(popping) init(False)

    // number of entries readable from ram
    val filled = Reg(UInt(ptrWidth bits)) init(0)

    // two buffers after the RAM
    val a = Stream(dataType)
    val b = Stream(dataType)

    io.push.ready := !full

    // indicates if we read (a valid word from) memory in current cycle
    val mem_read = Bool()
    // keep track of past outstanding valid reads
    val mem_read_d1 = RegNext(mem_read) init(False)
    val mem_read_d2 = RegNext(mem_read_d1) init(False)

    // room after the FIFO, in the skid buffers 'a' and 'b', 1 extra if streaming
    val room = U(!a.valid).resize(2) + U(!b.valid).resize(2)
    // inflight (outstanding) memory reads
    val inflight = U(mem_read_d1).resize(2) + U(mem_read_d2).resize(2)

    // read if the RAM is not empty, and we can store the data in buffers a and b, or
    // the buffers are streaming towards the pop output
    mem_read := (room > inflight || io.pop.ready) && !empty
    // 2-cycle read from RAM
    mem_out.valid := Delay(mem_read, 2, init=False)
    mem_out.payload := RegNext(ram.readSync(popPtr.resized))

    a << mem_out.m2sPipe()//.s2mPipe()
    b << a.m2sPipe()//s.s2mPipe()
    io.pop << b

    // update fill level of RAM
    when (!pushing & mem_read) {
      filled := filled - 1
    } elsewhen (pushing & !mem_read) {
      filled := filled + 1
    } otherwise {
      filled := pushPtr - popPtr
    }

    // update full flag of RAM
    when (!pushing & popping) {
      full := False
    } elsewhen (pushing & !popping) {
      // filled with one empty entry left?
      full := (filled >= U(depth - 2, ptrWidth bits))
    } otherwise {
      // completely filled
      full := (filled >= U(depth - 1, ptrWidth bits))
    }

    // update RAM write address pointer
    when(pushing) {
      pushPtr := pushPtr + 1
    }
    // write to RAM
    when(pushing) {
      ram(pushPtr.resized) := io.push.payload
    }

    // update RAM read address pointer
    when(mem_read) {
      popPtr := popPtr + 1
    }

    // update RAM empty flag
    when (!pushing & mem_read) {
      empty := (filled <= 1)
    } elsewhen (pushing & !mem_read) {
      empty := False
    }

    io.occupancy := RegNext(pushPtr - popPtr)
    io.availability := RegNext(popPtr - pushPtr)

    when(io.flush){
      pushPtr := 0
      popPtr := 0
      filled := 0
      full := False
      empty := True
    }    
  }

  def formalCheck(cond: T => Bool): Vec[Bool] = this.rework {
    // (old) approach 1: extend two beyond the FIFO RAM depth to also cover buffers a and b
    ///val condition = (0 until depth + 2).map(x => cond(if (x < depth) logic.ram(x) else if (x == depth) logic.a.payload else logic.b.payload))
    ///// mask which memory locations are valid in the FIFO RAM and buffers a and b
    ///val mask = Vec(True, depth + 2)
    ///mask(depth) := logic.a.valid
    ///mask(depth + 1) := logic.b.valid

    val condition = (0 until depth).map(x => cond(logic.ram(x)))
    // mask which memory locations are valid in the FIFO RAM
    val mask = Vec(True, depth)
    val popMask = (~((U(1) << logic.popPtr.resize(depth)) - 1)).asBits
    val pushMask = ((U(1) << logic.pushPtr.resize(depth)) - 1).asBits
    when(logic.popPtr.resize(depth) < logic.pushPtr.resize(depth)) {
      mask.assignFromBits(pushMask & popMask)
    }.elsewhen(logic.popPtr.resize(depth) > logic.pushPtr.resize(depth)) {
      mask.assignFromBits(pushMask | popMask)
    }.elsewhen(logic.empty) {
      mask := mask.getZero
    }
    val check = mask.zipWithIndex.map{case (x, id) => x & condition(id)}
    // approach 2: add check_a and check_b into the Vec() here, easier:
    val check_a = logic.a.valid & cond(logic.a.payload)
    val check_b = logic.b.valid & cond(logic.b.payload)
    Vec(check ++ Vec(check_a) ++ Vec(check_b))
  }
```

Dolu1990, 2023-06-25:
> This is my latest status. It does not pass formal yet, and I am trying to find out if my design is buggy, or formal is incorrect.
I modified formalCheck to deal with my a and b buffers.

Formal  isn't passing for me aswell, but it seems to be a false positive on some additional buffers. The formal seems to do some whiteboxing which isn't true anymore with the new design.
I would say, if you run the SpinalSimStreamFifoTester from the branch i'm on, that's should be quite safe.

> I do not understand this part of the sentence: "with used as full / empty". Can you please elaborate?
(I did not focus on optimizing occupancy or availability in my design yet.)

The goal isn't only to optimize the occupancy availabilty io timing, but to also use similar logic to handle the full / empty internal signals.

See this area : 
https://github.com/SpinalHDL/SpinalHDL/blob/46fb70e8b031cdc0de83a4a50b812641c22d57f2/lib/src/main/scala/spinal/lib/Stream.scala#L1176

It use two updown counter, when those counters hit : MSB = True, others = False   the MSB is used as empty or full.

With this design, the bottleneck for 1024 depth fifo on artix 7 for FMax is the blockram fmax XD

```
fifo128 ->
Artix 7 -> 437 Mhz 52 LUT 65 FF 
Artix 7 -> 447 Mhz 59 LUT 65 FF 
Cyclone V -> 275 Mhz 50 ALMs

fifo1024 ->
Artix 7 -> 382 Mhz 70 LUT 77 FF 
Artix 7 -> 454 Mhz 79 LUT 77 FF 
Cyclone V -> 275 Mhz 68 ALMs
```

likewise, 2023-06-25:
I have specifically named the extra buffer in sync read case for formal purposes:

```scala
        val buffer = Stream(readPort.rsp)
        buffer << readArbitation.translateWith(readPort.rsp)
        io.pop << buffer
```

then with RAM-normalized pointers `.resize(log2Up(depth))`:
```
  def formalCheck(cond: T => Bool): Vec[Bool] = this.rework {
    val condition = (0 until depth).map(x => cond(logic.ram(x)))
    val mask = Vec(True, depth)
    val popMask = (~((U(1) << logic.ptr.pop.resize(log2Up(depth))) - 1)).asBits
    val pushMask = ((U(1) << logic.ptr.push.resize(log2Up(depth))) - 1).asBits
    when(logic.ptr.pop.resize(log2Up(depth)) < logic.ptr.push.resize(log2Up(depth))) {
      mask.assignFromBits(pushMask & popMask)
    }.elsewhen(logic.ptr.pop.resize(log2Up(depth)) > logic.ptr.push.resize(log2Up(depth))) {
      mask.assignFromBits(pushMask | popMask)
    }.elsewhen(logic.ptr.empty) {
      mask := mask.getZero
    }
    val check = mask.zipWithIndex.map{case (x, id) => x & condition(id)}
    // only for sync, also check sync buffer
    val check_sync_buffer = logic.pop.sync.buffer.valid & cond(logic.pop.sync.buffer.payload)
    Vec(check :+ check_sync_buffer)
  }
```
this portion now passes formal:
```
class FormalFifoTester extends SpinalFormalFunSuite {
  //TODO fix me
  test("fifo-verify all") {
    val initialCycles = 2
    val inOutDelay = 2
    val coverCycles = 10
    FormalConfig
      .withBMC(10)
      //.withProve(10)
      .withCover(coverCycles)
      // .withDebug
      .doVerify(new Component {
        val depth = 4
        val dut = FormalDut(new StreamFifo(UInt(7 bits), depth, withAsyncRead=false, withBypass=false, occupancyFromRamOnly=false, forFMax=true))
```

We should iterate through options there too, I think.

I first trying to understand why induction fails. (`withProve`). I see `d2` is in the buffer register initially somehow, so it pops out first and the test fails.

likewise, 2023-06-25:
I am not sure how we can make `formalCheck()` depend on `withAsyncRead`. Any ideas or hints?

likewise, 2023-06-25:
Duh. `withAsyncRead` automatically becomes a member field of the class object, right?

Change to result of `formalCheck(cond: T => Bool): Vec[Bool]` to be like this, and we are passing more combinations.

```scala
    if (!withAsyncRead) {
      // only for sync, also check in s2mPipe() (on io.pop)
      val check_buffer = io.pop.valid & cond(io.pop.payload)
      Vec(check :+ check_buffer)
    } else {
      Vec(check)
    }
  }
```

likewise, 2023-06-25:
Also in `formalCheck`, the check for `empty` RAM must be to compare the pop and push register, because `empty` might include the extra room in the buffer.

i.e. new code:

```
    // RAM empty?
    }.elsewhen(logic.ptr.pop === logic.ptr.push) {
```

likewise, 2023-06-25:
I am checking with below `formalCheck()` against StreamFifo (a7e2f7b44ef12deaee1b952debeb1a6f3eb09c44).

```scala
 def formalCheck(cond: T => Bool): Vec[Bool] = this.rework {
    // check condition against all words in RAM
    val condition = (0 until depth).map(x => cond(logic.ram(x)))
    // create a mask of all valid words in RAM, assume RAM to be full
    val mask = Vec(True, depth)
    val popMask = (~((U(1) << logic.ptr.pop.resize(log2Up(depth))) - 1)).asBits
    val pushMask = ((U(1) << logic.ptr.push.resize(log2Up(depth))) - 1).asBits
    when(logic.ptr.pop.resize(log2Up(depth)) < logic.ptr.push.resize(log2Up(depth))) {
      mask.assignFromBits(pushMask & popMask)
    }.elsewhen(logic.ptr.pop.resize(log2Up(depth)) > logic.ptr.push.resize(log2Up(depth))) {
      mask.assignFromBits(pushMask | popMask)
    // empty?
    }.elsewhen(logic.ptr.pop === logic.ptr.push) {
      mask := mask.getZero
    }
    val check = mask.zipWithIndex.map{case (x, id) => x & condition(id)}
    if (!withAsyncRead) {
      // only for sync, also check in m2sPipe() which is directly on io.pop
      val check_buffer = io.pop.valid & cond(io.pop.payload)
      Vec(check :+ check_buffer)
    } else {
      Vec(check)
    }
  }
```

I am seeing BMC and Cover pass, but induction (Prove) fails.

I think the issue might be you only have one buffer while the RAM latency is two, but I am still checking the waveforms to understand what happens.

Readon, 2023-06-26:
> ut induction (Prove) FAIL. Investigating why, but it's hard to track down, names are scrambled and hard to debug in Verilog or waveforms.

Yes, the prove is most difficult one. I could have a look at it when I found time. :)

Readon, 2023-06-26:
One method is to increase the steps that BMC runs, default is 10?. Some company do that.

likewise, 2023-06-26:
How can I keep the names of the vars of formalCheck in the RTL output?

So that I can debug the formal verification itself a bit better?

likewise, 2023-06-26:
Can you update your branch with below formal stuff?

I suggest not exposing the one extra "register" to the interface (`availability`) and hide it as an internal design decision.

One change in `StreamFifo.scala`:
```
          val fullTracker = new CounterUpDownFmax(1 << counterWidth, (1 << counterWidth - 1) - depth) {
            incr := doPush //io.push.fire
            decr := doPop //io.pop.fire
            full := value.msb
          }
```
I have adapted the existing formal verification code, without deleting any checks (adding a few) to pass to your new `StreamFifo`, initially only for the `withAsyncRead=false, withBypass=false, forFMax=true` cases, but should be a good starting point to adapt for the other cases. (I can quickly do this with some help on the Gitter chat.)

New formal code for `StreamFifo.scala`:

```scala
// check a condition against all valid payloads in the FIFO RAM
  def formalCheckRam(cond: T => Bool): Vec[Bool] = this.rework {
    val condition = (0 until depth).map(x => cond(logic.ram(x)))
    // create mask for all valid payloads in FIFO RAM
    // inclusive [popd_idx, push_idx) exclusive
    // assume FIFO RAM is full with valid payloads
    //           [ ...  push_idx ... ]
    //           [ ...  pop_idx  ... ]
    // mask      [ 1 1 1 1 1 1 1 1 1 ]
    val mask = Vec(True, depth)
    val push_idx = logic.ptr.push.resize(log2Up(depth))
    val pop_idx = logic.ptr.pop.resize(log2Up(depth))
    // pushMask(i)==0 indicates location i was popped
    val popMask = (~((U(1) << pop_idx) - 1)).asBits
    // pushMask(i)==1 indicates location i was pushed
    val pushMask = ((U(1) << push_idx) - 1).asBits
    // no wrap   [ ... popd_idx ... push_idx ... ]
    // popMask   [ 0 0 1 1 1 1  1 1 1 1 1 1 1 1 1]
    // pushpMask [ 1 1 1 1 1 1  1 1 0 0 0 0 0 0 0] &
    // mask      [ 0 0 1 1 1 1  1 1 0 0 0 0 0 0 0]
    when(pop_idx < push_idx) {
      mask.assignFromBits(pushMask & popMask)
    // wrapped   [ ... push_idx ... popd_idx ... ]
    // popMask   [ 0 0 0 0 0 0  0 0 1 1 1 1 1 1 1]
    // pushpMask [ 1 1 0 0 0 0  0 0 0 0 0 0 0 0 0] |
    // mask      [ 1 1 0 0 0 0  0 0 1 1 1 1 1 1 1]
    }.elsewhen(pop_idx > push_idx) {
      mask.assignFromBits(pushMask | popMask)
    // empty?
    //           [ ...  push_idx ... ]
    //           [ ...  pop_idx  ... ]
    // mask      [ 0 0 0 0 0 0 0 0 0 ]
    }.elsewhen(logic.ptr.empty) {
      mask := mask.getZero
    }
    val check = mask.zipWithIndex.map{case (x, id) => x & condition(id)}
    Vec(check)
  }

  def formalCheckm2sPipe(cond: T => Bool): Bool = this.rework {
    io.pop.valid & cond(io.pop.payload)
  }

  // verify this works, then we can simplify below
  //def formalCheck(cond: T => Bool): Vec[Bool] = this.rework {
  //  Vec(formalCheckm2sPipe(cond) +: formalCheckRam(cond))
  //}

  def formalContains(word: T): Bool = this.rework {
    formalCheckRam(_ === word.pull()).reduce(_ || _) || formalCheckm2sPipe(_ === word.pull())
  }
  def formalContains(cond: T => Bool): Bool = this.rework {
    formalCheckRam(cond).reduce(_ || _) || formalCheckm2sPipe(cond)
[... 86 lines cut ...]
          val id = OHToUInt(dut.formalCheckRam(_ === x.pull()).asBits)
          // offset all RAM order indexes by 1, as order index 0 is reserved for m2sPipe()
          // assume wrapped; add RAM size (depth) to location
          val extId = id +^ U(depth + 1)
          val compId = CombInit(extId)
          // for all non wrapped locations
          when(id >= (dut.logic.ptr.pop.resize(log2Up(depth)))) {
          // offset all RAM order indexes by 1, as order index 0 is reserved for m2sPipe()
            compId := (id +^ U(1)).resized
          }
          if (!asyncRead) {
            when(dut.formalCheckm2sPipe(_ === x.pull())) {
              // order index 0 for m2sPipe register
              compId := 0
            }
          }
          compId
        }
        when(d1_in && d2_in && !d1_out && !d2_out) {
          assert(getCompId(d1) < getCompId(d2))
        }
      })
  }
}
```

Dolu1990, 2023-06-26:
@likewise 

> One change in StreamFifo.scala:  val fullTracker = new CounterUpDownFmax...

The reason why it use : 
            incr := io.push.fire
            decr := io.pop.fire
instead of             
            incr := doPush //io.push.fire
            decr := doPop //io.pop.fire
            
is to ensure that never more than "depth" transaction can be in between io.push / io.pop, so it include the memory storage but also the additional 1 entry buffer that is created by the RAM read port (when using syncronous read port).

I did it like it to ensure backward compatibility. For instance, the SPI controller put the fifo availability readable by the software, and the software is allowed to assume that then when the occupancy decremented, it mean that the transaction is fully completted. 

Else, without that consideration, yes i agree.

> How can I keep the names of the vars of formalCheck in the RTL output?

Composite are your friend ^^

```scala
  def formalCheckRam(cond: T => Bool): Vec[Bool] = this rework new Composite(this){
     ...
  }  
````

I pushed the formal changes

likewise, 2023-06-26:
I think using doPop and doPush for fullTracker is fully backward compatible. The StreamFifo is full when 'depth' elements are inside.

This hides the extra (optional) stage.

I think that available should not be able to exceed depth.

likewise, 2023-06-26:
Thanks, yes sounds fine.

I find it strange that it passes withProve without that assertion.

I will try to adapt the assertion to the io.{push,pop}.valid case for fullTracker.

I thought it was needed to make withProve() work, but it may have been my getCompId fixes that solved that.

Dolu1990, 2023-06-26:
> I find it strange that it passes withProve without that assertion.

I only tried the SpinalSim thing, i didn't tried the formal one, as i'm on a PC without formal stuff installed right now.

likewise, 2023-06-26:
With your `fullTracker`,  here is the new `assert()`  that guarantees the relationship between `dut.logic.ptr.push`, `dut.logic.ptr.pop` and `logic.ptr.arb.fmax.fullTracker.value`.

This is needed for induction proof (`.withProve()`) to pass, but is also a good check in general.

```scala
        // initially (1 << log2Up(depth)), minus what is inside the StreamFifo
        assert(dut.logic.ptr.arb.fmax.emptyTracker.value === ((1 << log2Up(depth)) - (dut.logic.ptr.push - dut.logic.ptr.pop)))
        // initially (1 << log2Up(depth)) - depth, plus what is inside the StreamFifo and the output stage
        assert(dut.logic.ptr.arb.fmax.fullTracker.value ===  ((1 << log2Up(depth)) - depth + (dut.logic.ptr.push - dut.logic.ptr.pop) + U(dut.io.pop.valid)))
```

Dolu1990, 2023-06-26:
Ahh nice ^^
Thanks :)

likewise, 2023-06-26:
The Bench approach does not give a senseful fmax is the requested frequency is too low, as Vivado will not optimize after `WNS>0.000` was reached.

Asking for `400 MHz`
```
[info] fifo128 ->
[info] Virtex UltraScale+ -> 612 Mhz 26 LUT 59 FF 
[info] fifo1024 ->
[info] Virtex UltraScale+ -> 588 Mhz 33 LUT 68 FF
```

Asking for `850 MHz`
```
[info] fifo128 ->
[info] Virtex UltraScale+ -> 912 Mhz 26 LUT 59 FF 
[info] fifo1024 ->
[info] Virtex UltraScale+ -> 836 Mhz 33 LUT 68 FF
```

Nice result:
825 MHz for 32 bit wide, 128 bit deep
725 MHz for 128 bit wide, 512 deep.

Dolu1990, 2023-06-27:
:D 
And that is without the forFMax argument set to true right ?

likewise, 2023-06-27:
That is with forFmax=true.

In my fork, fifo-rework branch, I pushed some more formal stuff.

It still only passes for the sync fmax case.

Would you like to keep all new StreamFifo options, or limit
them?

I think it is important to choose this, before spending much time on FV.

Dolu1990, 2023-06-27:
> That is with forFmax=true

Ahhh ok

> Would you like to keep all new StreamFifo options, or limit
them?

I think all the options have a use case. Ultimately we could even replace the StreamFifoLowLatency with the proposed StreamFifo implemenation. (i will take a look about it)

> It still only passes for the sync fmax case.

Is It due to the formal verification testbench itself or the fifo design ?

Dolu1990, 2023-06-27:
StreamFIfoLowLatency is now replaced by the StreamFifo.
It passes all the SpinalSim tests

likewise, 2023-06-27:
You changed:

```scala
    Vec(check)
  }
```

to:

```scala
    val vec = Vec(check)
  }.vec
```

Can you explain why this is needed? (It is also in `dev` still).

likewise, 2023-06-27:
Suggested change:
```scala
  if(!withAsyncRead) assert(!withBypass)
```
to:
```
  // bypass not supported for sync read case
  require(withAsyncRead || !withBypass)
```
so that it is clear that this is a requirement on the arguments, not a code failure condition.

likewise, 2023-06-27:
`popStorageMatter` and thus the `occupancyFromRamOnly` are never used?

likewise, 2023-06-27:
`val allowExtraMsb : Boolean = true`

What would be the use case for `false`? You always need one bit to keep the condition anyway, you use `wentUp` otherwise.

What if we use this internally always?
```
  val withExtraMsb =  isPow2(depth)
```

Dolu1990, 2023-06-27:
```
  new Composite(xxx){
    val vec = Vec(check)
  }.vec
```

So, the Composite create a name space, but the function we are in want to turn the vec.
Instead of returning the composite, we return the vec attribute of the Composite.

> popStorageMatter and thus the occupancyFromRamOnly argument are never used?

Hoo right, i just removed it.

> What would be the use case for false? (allowExtraMsb)

So, on some hardware it may provide better area / timings, maybe. I provided the argument to let the user have the choice as it was a not changing much in the code. But that may be a bit overkill XD

> val withExtraMsb =  isPow2(depth)

Yes, could be done. Still, i think that as the option chance very little in the code base, it isn't bad to keep allowExtraMsb to let the user the choice / freedoom to play around.
To be clear, i do not realy expect people to dig that far, but it is kind of a "interresting" fifo design option.

Dolu1990, 2023-06-27:
Should i merge your branch in my one ?

likewise, 2023-06-27:
Please do so: https://github.com/SpinalHDL/SpinalHDL/pull/1146

likewise, 2023-06-27:
I'm hitting some Java stack overflow on iterating through the List options in  this update of `FormalFifoTester.scala`, so I have reduced the depth List to prevent the overflow. Your `SpinalSimStreamFifoTester` iterator does not give me this problem, that's where I simply copied it from.

Dolu1990, 2023-06-27:
merged :)

andreasWallner, 2023-06-27:
Just tried it for ice40, the improvement is amazing:
```
Name              f max      size        
----------------------------------------
FIFO 8x256 fMax   196 MHz    255/7680 LCs  **********
FIFO 8x256        132 MHz    187/7680 LCs  *******

likewise, 2023-06-27:
Yeah nice, you had 50 MHz before, right? On AMD/Xilinx, we are hitting the BRAM limits as the critical path. I am seeing two logic levels max between registers, unless I make the FIFO very wide or deep (i.e. many BRAMs have to be combined).

Dolu1990, 2023-06-28:
I think we can merge the branch into dev.
Good for you ?

likewise, 2023-06-28:
Hard to say without formal verification for me :) The hard part is checking for all combinations that you built XD.

If found one formal failure I think for withAsync withBypass forFmax, depth 3.

In this case:
```scala
ptr.empty=1
io.pop.valid=1
io.pop.ready=0
```
then `io.push.ready=1` which is not expected, as the push and pop ports should be combinatorially connected together.

```scala
      val async = withAsyncRead generate new Area{
        if(withBypass){
          when(ptr.empty){
            io.pop.valid := io.push.valid
            io.pop.payload := io.push.payload
            ptr.doPush clearWhen(io.pop.ready)
          }
```

Dolu1990, 2023-06-29:
Merged, i think, verification speaking, it is ok. The non-formal SpinalSim one is probably providing a good coverage too :)

likewise, 2023-06-29:
Yes, agreed.

I probably do not have time short-period to also make formal tests for the `!forFMax` cases to support formal verification induction proofs, but have documented what needs to be done. (Basically, assert relationship between push/pop/fill-level.)

One question:

For the `withAsyncRead` && `withBypass` case, when/how is the `io.push.ready` connected to `io.pop.ready` in bypass? I cannot find the corresponding code.

Contrary to my earlier statement, I will **not** continue on my `StreamFifoHighPerformance` work, because I think all ideas (and more!) are already in your implementation. I especially like the {empty, full}Tracker approach to make the `empty` and `full` bits fully registered outputs.

Dolu1990, 2023-06-29:
> For the withAsyncRead && withBypass case, when/how is the io.push.ready connected to io.pop.ready in bypass? I cannot find the corresponding code.

It doesn't need to be an additional condition.
Reason is : 
The bypass only need to activated when the fifo is empty. When the fifo is empty, there is always some room in the fifo. => 
no need care about the io.push.ready durring bypass, as it is already high (empty fifo)

> Contrary to my earlier statement, I will not continue on my StreamFifoHighPerformance work, because I think all ideas (and more!) are already in your implementation. I especially like the {empty, full}Tracker approach to make the empty and full bits fully registered outputs.

No worries, the implementation in the SpinalHDL/fifo-rework branch would not have been at the point it is without you :)
I'm quite happy you pushed this issue forward.

likewise, 2023-06-29:
The f.max reports from the VivadoFlow are not (always) representative. The scripting lacks checking for timing violations such as worst pulse width slack. I found it strange I could achieve f.max higher than the datasheet BRAM f.max for `-2` devices.

Still we are hitting BRAM limits, I think 650-700 MHz is doable.

Dolu1990, 2023-06-30:
Merged in dev !

@likewise Did you tried the new design in your production design ?

[4 more replies]

closed 2023-07-10 by Dolu1990