SpinalHDL/SpinalHDL #1147: inout interface always cause "NO DRIVEN" problems  ,spinalHDL version 1.8.0b

Mido-sun, 2023-06-28:
upup

Dolu1990, 2023-06-28:
Hi, 
Have you tried 1.8.2 ?
Do you have a code sample to reproduce ?

Mido-sun, 2023-06-29:
case class lbsInf() extends Bundle{

    val PAD_GLB_RST        =in    Bool()
    val PAD_EVB_LED        =in Bool()
    val PAD_CPU_DATA0      =in Bool()
    val PAD_CPU_DATA1      =in Bool()
    val PAD_CPU_DATA2      =in Bool()
    val PAD_CPU_DATA3      =in Bool()
    val PAD_CPU_DATA4      =in Bool()
    val PAD_CPU_DATA5      =in Bool()
    val PAD_CPU_DATA6      =in Bool()
    val PAD_CPU_DATA7      =in Bool()
    val PAD_CPU_DATA8      =in Bool()
    val PAD_CPU_DATA9      =in Bool()
    val PAD_CPU_DATA10     =in Bool()
    val PAD_CPU_DATA11     =in Bool()
    val PAD_CPU_DATA12     =in Bool()
    val PAD_CPU_DATA13     =in Bool()
    val PAD_CPU_DATA14     =in Bool()
    val PAD_CPU_DATA15     =in Bool()
    val PAD_CPU_ALE_I      =in    Bool()
    val PAD_CPU_CS_N_I     =in    Bool()
    val PAD_CPU_OE_N_I     =in    Bool()
    val PAD_CPU_WE_N_I     =in    Bool()

  def connect(that:lbsInf): Unit = {
      that.PAD_EVB_LED          := this.PAD_EVB_LED
      that.PAD_GLB_RST          := this.PAD_GLB_RST
      that.PAD_CPU_DATA0        := this.PAD_CPU_DATA0
      that.PAD_CPU_DATA1        := this.PAD_CPU_DATA1
      that.PAD_CPU_DATA2        := this.PAD_CPU_DATA2
      that.PAD_CPU_DATA3        := this.PAD_CPU_DATA3
      that.PAD_CPU_DATA4        := this.PAD_CPU_DATA4
      that.PAD_CPU_DATA5        := this.PAD_CPU_DATA5
      that.PAD_CPU_DATA6        := this.PAD_CPU_DATA6
      that.PAD_CPU_DATA7        := this.PAD_CPU_DATA7
      that.PAD_CPU_DATA8        := this.PAD_CPU_DATA8
      that.PAD_CPU_DATA9        := this.PAD_CPU_DATA9
      that.PAD_CPU_DATA10       := this.PAD_CPU_DATA10
      that.PAD_CPU_DATA11       := this.PAD_CPU_DATA11
      that.PAD_CPU_DATA12       := this.PAD_CPU_DATA12
      that.PAD_CPU_DATA13       := this.PAD_CPU_DATA13
      that.PAD_CPU_DATA14       := this.PAD_CPU_DATA14
      that.PAD_CPU_DATA15       := this.PAD_CPU_DATA15
      that.PAD_CPU_ALE_I        := this.PAD_CPU_ALE_I
      that.PAD_CPU_CS_N_I       := this.PAD_CPU_CS_N_I
      that.PAD_CPU_OE_N_I       := this.PAD_CPU_OE_N_I
      that.PAD_CPU_WE_N_I       := this.PAD_CPU_WE_N_I
  }

}
this is the definition ，one blackbox instance it
and  in main files 
   val io = new Bundle {

        val lbsInfa = lbsInf().setName("")

   }
    noIoPrefix()

val 
io.serdesInfa.connect(u_xxxxx_top.io.lbsInf)

run and "NO DRIVEN XXXX"

Mido-sun, 2023-06-29:
do not pay attention the name of the val

Mido-sun, 2023-06-29:
1.8.2 reports the same error info

Dolu1990, 2023-06-29:
Ahh inout Bool()   should inout(Analog(Bool()))

That maybe part of the issue.

Can you provide a fully self contained example ?

Mido-sun, 2023-06-29:
thanks a lot ,inout(Analog(Bool())) solves my problem   ,another problem, if I want to add this ---->
(* M0_PLL_ENABLE=1 *)
(* M0_PLL_REF_CLK_DIV=1 *)
(* M0_PLL_POSTDIV1=7 *)
(* M0_PLL_POSTDIV2=7 *)

in the head of the top module , like this 

(* M0_PLL_ENABLE=1 *)
(* M0_PLL_REF_CLK_DIV=1 *)
(* M0_PLL_POSTDIV1=7 *)
(* M0_PLL_POSTDIV2=7 *)
moudule topLevel(

)
endmodule

"addAttribute("M0_PLL_ENABLE","1")" wont add it to the top of the generated verilog .

Does SPINAL support the function ?

Mido-sun, 2023-06-29:
I am sorry that the fully code is forbidden to upload to the internet because of the company policy , sorry

Dolu1990, 2023-06-29:
Yes it is supported  in that way : 
```scala

object Debug1237654 extends App{
  SpinalVerilog(new Component{
    definition.addAttribute("miaou", "rawrr")
    definition.addAttribute("miaou2", "rawrr2")
  })
}
```

```verilog
(* miaou = "rawrr" , miaou2 = "rawrr2" *) module unamed (
);

endmodule
```

Mido-sun, 2023-06-29:
wow  I really learn something from you ! 
I don't want to write it everytime when I generate diff verilog ,CAN iT  make Encapsulation？

Dolu1990, 2023-06-29:
I'm not sure to understand exactly, could you provide an example of what you are looking for ?

Mido-sun, 2023-06-29:
val gen = new MySpinalConfig(targetDirectory).generateVerilog(new toplevel{
        definition.addAttribute("PROJECT_DEVICE","xxxx")
        definition.addAttribute("SOC_REF_CLK_FREQ","50.00")
        definition.addAttribute("M0_PLL_ENABLE","1")
        definition.addAttribute("M0_PLL_REF_CLK_DIV","1")
        definition.addAttribute("M0_PLL_POSTDIV1","7")
        definition.addAttribute("M0_PLL_POSTDIV2","7")
        definition.addAttribute("M0_PLL_FBDIV","16'h31")
        definition.addAttribute("M0_PLL_FRAC","24'h0")
        definition.addAttribute("M0_PLL_PLL_MODE","1")
        definition.addAttribute("M0_PLL_FN_PLL_CFG0","16'h1400")
        definition.addAttribute("M1_PLL_ENABLE","1")
        definition.addAttribute("M1_PLL_REF_CLK_DIV","1")
        definition.addAttribute("M1_PLL_POSTDIV1","1")
        definition.addAttribute("M1_PLL_POSTDIV2","1")
        definition.addAttribute("M1_PLL_FBDIV","16'h28")
        definition.addAttribute("M1_PLL_FRAC","24'h0")
        definition.addAttribute("M1_PLL_PLL_MODE","1")
        definition.addAttribute("M1_PLL_FN_PLL_CFG0","16'h1400")
        definition.addAttribute("TEST_MUX0_ENABLE","0")
        definition.addAttribute("TEST_MUX1_ENABLE","0")
        definition.addAttribute("TEST_MUX2_ENABLE","0")
        definition.addAttribute("TEST_MUX3_ENABLE","0")
        definition.addAttribute("TEST_MUX4_ENABLE","0")
        definition.addAttribute("TEST_MUX5_ENABLE","0")
        definition.addAttribute("TEST_MUX6_ENABLE","0")
        definition.addAttribute("TEST_MUX7_ENABLE","0")
        definition.addAttribute("TEST_MUX7_ENABLE","0")
        definition.addAttribute("PPI_CONFIG_MODE","X8")
        definition.addAttribute("PPI_CONFIG_DATA_IO_REUSE_ENABLE","0")

    }.setDefinitionName("xxxx"))
}

it generate

////
(* PROJECT_DEVICE = "HFD2H1000" , SOC_REF_CLK_FREQ = "50.00" , M0_PLL_ENABLE = "1" , M0_PLL_REF_CLK_DIV = "1" , M0_PLL_POSTDIV1 = "7" , M0_PLL_POSTDIV2 = "7" , M0_PLL_FBDIV = "16'h31" , M0_PLL_FRAC = "24'h0" , M0_PLL_PLL_MODE = "1" , M0_PLL_FN_PLL_CFG0 = "16'h1400" , M1_PLL_ENABLE = "1" , M1_PLL_REF_CLK_DIV = "1" , M1_PLL_POSTDIV1 = "1" , M1_PLL_POSTDIV2 = "1" , M1_PLL_FBDIV = "16'h28" , M1_PLL_FRAC = "24'h0" , M1_PLL_PLL_MODE = "1" , M1_PLL_FN_PLL_CFG0 = "16'h1400" , TEST_MUX0_ENABLE = "0" , TEST_MUX1_ENABLE = "0" , TEST_MUX2_ENABLE = "0" , TEST_MUX3_ENABLE = "0" , TEST_MUX4_ENABLE = "0" , TEST_MUX5_ENABLE = "0" , TEST_MUX6_ENABLE = "0" , TEST_MUX7_ENABLE = "0" , TEST_MUX7_ENABLE = "0" , PPI_CONFIG_MODE = "X8" , PPI_CONFIG_DATA_IO_REUSE_ENABLE = "0" *) module virtualPallasTop (
////

 all the parameters Put on one line which is really badlooking

I wish it would be like this ：
(* PROJECT_DEVICE="HFD2H1000" *)
(* SOC_REF_CLK_FREQ="50.00" *)
(* M0_PLL_ENABLE=1 *)
(* M0_PLL_REF_CLK_DIV=1 *)
(* M0_PLL_POSTDIV1=7 *)
(* M0_PLL_POSTDIV2=7 *)
(* M0_PLL_FBDIV=16'h31 *)
(* M0_PLL_FRAC=24'h0 *)
(* M0_PLL_PLL_MODE=1 *)
(* M0_PLL_FN_PLL_CFG0=16'h1400 *)
(* M1_PLL_ENABLE=1 *)
(* M1_PLL_REF_CLK_DIV=1 *)
(* M1_PLL_POSTDIV1=1 *)
(* M1_PLL_POSTDIV2=1 *)
(* M1_PLL_FBDIV=16'h28 *)
(* M1_PLL_FRAC=24'h0 *)
(* M1_PLL_PLL_MODE=1 *)
(* M1_PLL_FN_PLL_CFG0=16'h1400 *)
(* TEST_MUX0_ENABLE=0 *)
(* TEST_MUX1_ENABLE=0 *)
(* TEST_MUX2_ENABLE=0 *)
(* TEST_MUX3_ENABLE=0 *)
(* TEST_MUX4_ENABLE=0 *)
(* TEST_MUX5_ENABLE=0 *)
(* TEST_MUX6_ENABLE=0 *)
(* TEST_MUX7_ENABLE=0 *)
(* PPI_CONFIG_MODE="X8" *)
(* PPI_CONFIG_DATA_IO_REUSE_ENABLE=0 *)

module  xxxxx
  (
）endmodule

Mido-sun, 2023-06-29:
Can these codes be optimized so that I can call them conveniently next time?

val gen = new MySpinalConfig(targetDirectory).generateVerilog(new toplevel{
    definition.addAttribute("PROJECT_DEVICE","xxxx")
    definition.addAttribute("SOC_REF_CLK_FREQ","50.00")
    definition.addAttribute("M0_PLL_ENABLE","1")
    definition.addAttribute("M0_PLL_REF_CLK_DIV","1")
    definition.addAttribute("M0_PLL_POSTDIV1","7")
    definition.addAttribute("M0_PLL_POSTDIV2","7")
    definition.addAttribute("M0_PLL_FBDIV","16'h31")
    definition.addAttribute("M0_PLL_FRAC","24'h0")
    definition.addAttribute("M0_PLL_PLL_MODE","1")
    definition.addAttribute("M0_PLL_FN_PLL_CFG0","16'h1400")
    definition.addAttribute("M1_PLL_ENABLE","1")
    definition.addAttribute("M1_PLL_REF_CLK_DIV","1")
    definition.addAttribute("M1_PLL_POSTDIV1","1")
    definition.addAttribute("M1_PLL_POSTDIV2","1")
    definition.addAttribute("M1_PLL_FBDIV","16'h28")
    definition.addAttribute("M1_PLL_FRAC","24'h0")
    definition.addAttribute("M1_PLL_PLL_MODE","1")
    definition.addAttribute("M1_PLL_FN_PLL_CFG0","16'h1400")
    definition.addAttribute("TEST_MUX0_ENABLE","0")
    definition.addAttribute("TEST_MUX1_ENABLE","0")
    definition.addAttribute("TEST_MUX2_ENABLE","0")
    definition.addAttribute("TEST_MUX3_ENABLE","0")
    definition.addAttribute("TEST_MUX4_ENABLE","0")
    definition.addAttribute("TEST_MUX5_ENABLE","0")
    definition.addAttribute("TEST_MUX6_ENABLE","0")
    definition.addAttribute("TEST_MUX7_ENABLE","0")
    definition.addAttribute("TEST_MUX7_ENABLE","0")
    definition.addAttribute("PPI_CONFIG_MODE","X8")
    definition.addAttribute("PPI_CONFIG_DATA_IO_REUSE_ENABLE","0")

}.setDefinitionName("xxxx"))

Dolu1990, 2023-06-29:
> all the parameters Put on one line which is really badlooking

Yes. 
I wasn't aware of the multiline syntax.
So currently multiline isn't implemented. What's about having : 

```verilog
(* PROJECT_DEVICE = "xxxx" , 
    SOC_REF_CLK_FREQ = "50.00" , 
    M0_PLL_ENABLE = "1" , 
   ..
    M0_PLL_REF_CLK_DIV = "1" *)
````

Is it a legal syntax ?

> reusing

Ahh yes sure : 

```scala
object Tuner{
  def on[T <: Component](c : T) = {
    c.definition.addAttribute("PROJECT_DEVICE","xxxx")
    ...
    c
  }
}

SpinalVerilog(Tuner.on(new MyComponent))
```

Should be good

Mido-sun, 2023-06-29:
> 
 (* PROJECT_DEVICE="HFD2H1000" *)
(* SOC_REF_CLK_FREQ="50.00" *)
....
(* M0_PLL_ENABLE=1 *)

I konw this is a legal way

closed 2023-07-14 by Mido-sun