SpinalHDL/SpinalHDL #1152: Are there some methods to access signals in Blackbox in testbench?

yuqi-ali, 2023-07-06:
If I init a blackbox in SpinalHDL code,Are there some methods to access signals in Blackbox in testbench?
Just like access the signal with tag simPublic()?

Readon, 2023-07-09:
Do you mean signals not defined in io?

yuqi-ali, 2023-07-10:
Yes, signals in blackbox internal

Readon, 2023-07-10:
> Yes, signals in blackbox internal

I think there would not be info about blackbox internal signals.

Dolu1990, 2023-07-10:
As long as you define a signal in the blackbox itself which map to a existing one in the blackbox implementation, that's fine.
In the blackbox : 
```scala
    val miaou = UInt(8 bits).simPublic()
``` 

In the verilog : 

```verilog
  wire       [7:0]    miaou  /* verilator public */  = 42 ;
```

in the testbench : 

```scala
      println(dut.blackBoxtoTest.miaou.toInt)
```

it worked for me.

yuqi-ali, 2023-07-11:
@Dolu1990 It's work for a single blackbox，how to deal with black about the nesting blackbox just like this:

```
case class A() extends BlackBox{
  val io = new Bundle {
    val data_in, data_in2 = in Bits (16 bits)
    val data_out = out Bits (16 bits)
  }
  noIoPrefix()
  setInlineVerilog(
    """
      |module A(
      | input [15:0] data_in,
      | input [15:0] data_in2,
      | output[15:0] data_out
      |);
      |B B_Inst(
      |.data_in(data_in),
      |.data_in2(data_in2),
      |.data_out(data_out)
      |);
      |endmodule
      |
      |module B(
      | input [15:0] data_in,
      | input [15:0] data_in2,
      | output[15:0] data_out
      |);
      |wire [15:0] sum /* verilator public */;
      |assign sum=data_in+data_in2;
      |assign data_out=sum-1'b1;
      |endmodule
      |""".stripMargin
  )
}
```
    If I want to access the sum signal in module B?

Dolu1990, 2023-07-11:
I guess that if in your scala blackbox you create : 

```scala
val A =  new Component{
 val miaou = UInt(8 bits).simPublic()
}
```

it would work aswell ?

yuqi-ali, 2023-07-13:
It works，Thanks！

closed 2023-07-13 by yuqi-ali