SpinalHDL/SpinalHDL #1158: SpinalHDL generates illegal VHDL identifiers (GHDL says)

likewise, 2023-07-10:
In a larger project, this fails. I'm not sure if it is (only) related to the #1144 work.

```
  val io = new Bundle with StreamFifoInterface[T]{
    <...>
    override def pushOccupancy = occupancy
    override def popOccupancy = occupancy
  }
```
fails with GHDL:
```
[info] [Progress] GHDL compilation started
[info] [Progress] GHDL compilation done in 16.502 ms
[info] /project-on-host/Finka/tmp/job_1/Finka.vhd:734:8: two underscores can't be consecutive    zz__1 : in std_logic;       ^/project-on-host/Finka/tmp/job_1/Finka.vhd:735:7: an identifier cannot finish with '_'    zz_ : in std_logic
```
Debugging the issue a bit:
```
entity StreamFifo_40 is
  port(
    io_push_valid : in std_logic;
    <...>
    io_occupancy : out unsigned(5 downto 0);
    io_availability : out unsigned(5 downto 0);
    zz__1 : in std_logic;   <---------------------------------------- due to "override def pushOccupancy = occupancy" in Bundle??
    zz_ : in std_logic   <---------------------------------------- due to "override def popOccupancy = occupancy" in Bundle??
  );
end StreamFifo_40;
```

My Simulator configuration is:
```
    val simConfig = SimConfig
    // synchronous resets, see Config.scala
    .withConfig(Config.spinal)
    .allOptimisation
    .withGhdl.addRunFlag("--unbuffered").addRunFlag("--ieee-asserts=disable").addRunFlag("--assert-level=none").addRunFlag("--backtrace-severity=warning")

```
and the `Config.spinal` is as follows:

```
object Config {
  def spinal = SpinalConfig(
    targetDirectory = "build/rtl",
    mode = VHDL,
    // synchronous resets for Xilinx devices
    defaultConfigForClockDomains = ClockDomainConfig(
      resetKind = SYNC_RESET
    ),
    // change (un)signed I/Os at toplevel into std_logic_vector
    onlyStdLogicVectorAtTopLevelIo = true
  )
  def syncConfig = ClockDomainConfig(
    resetKind = SYNC_RESET
  )
}
```

I am trying to build a minimal example, but did not succeed yet.
For example, I tried copying the above Sim to `class SpinalSimStreamFifoTester` but cannot get that to fail yet.

likewise, 2023-07-11:
Simulator is not the issue, problems happens in generated VHDL also without Simulator. Further zooming in it seems to have to do with 2 clock domains (RX and TX), and *per* clock domain I set up two parallel instances: 
```
  val io = new Bundle{
    // main clock for CPU and AXI interconnect, 250 MHz
    val clk     = in Bool()
    val rst     = in Bool()

    // high speed clock specifically for crypto
    val turbo_clk     = in Bits(2 bits)
    val turbo_rst     = in Bits(2 bits)
```
here we create two clockdomains:
```
  val rxCryptoClockDomain = ClockDomain(
    clock = io.turbo_clk.apply(0),
    reset = io.turbo_rst.apply(0),
    config = Config.syncConfig
  )

  val txCryptoClockDomain = ClockDomain(
    clock = io.turbo_clk.apply(1),
    reset = io.turbo_rst.apply(1),
    config = Config.syncConfig
  )
```
which gets passed by argument:
```
  // packet rx area
  val packetRx = new ClockingArea(axiClockDomain) {
    val rx = CryptoFunction(cryptoCD = rxCryptoClockDomain)
```

```
case class CryptoFunction(cryptoCD : ClockDomain) extends Component {
  // two parallel pipelines gets instantiated, within the same clock domain
  val crypto_areas = Array.tabulate(2)(instanceNr => {
    new Area {
      val crypto_turbo = new ClockingArea(cryptoCD) {
```

Resulting VHDL:
```
entity Finka is
  port(
    clk : in std_logic;
    rst : in std_logic;
    turbo_clk : in std_logic_vector(1 downto 0);
    turbo_rst : in std_logic_vector(1 downto 0);
    ...
```
```
  zz_zz_ <= pkg_extract(turbo_clk,0);
  zz_zz__1 <= pkg_extract(turbo_rst,0);
```
Nowhere do I see `pkg_extract(turbo_clk,1);` or `pkg_extract(turbo_rst,1);
```
  packetRx_rx : entity work.CryptoFunction
    port map ( 
      ...
      zz_ => zz_zz__1,
      zz__1 => zz_zz_,   <----------------- illegal identifier gets introduced
      resetCtrl_axiResetDelayed => resetCtrl_axiResetDelayed,
      clk => clk 
    );
```

Dolu1990, 2023-07-11:
Hi,

So something is wrong in the way the name are provided in some ways.

I tried to reproduce, but without success, i can't reproduce the issue. I tried with : 
```scala
  SimConfig.withGhdl.withConfig(SpinalConfig(
    defaultConfigForClockDomains = ClockDomainConfig(
      resetKind = SYNC
    ),
    onlyStdLogicVectorAtTopLevelIo = true
  )).compile(new Component{
    val miaou = False
    val turbo_clk     = in Bits(2 bits)
    val turbo_rst     = in Bits(2 bits)
    val cd0 = ClockDomain(
      clock = turbo_clk.apply(0),
      reset = turbo_rst.apply(0)
    )

    val cd1 = ClockDomain(
      clock = turbo_clk.apply(1),
      reset = turbo_rst.apply(1)
    )

    val c0 = new Component {
      val x = cd0 on out(RegNext(True) init (False))
    }
    val c1 = new Component {
      val c2 = new Component {
        val x = cd1 on out(RegNext(True) init (False))
      }
    }
    setDefinitionName("miaou")
  }).doSim{dut =>
    simSuccess()
  }
```

Do you have a self contained way for me to reproduce ?

likewise, 2023-07-11:
I'm trying to create one based on my SpinalHDL hierarchy (see below), but fail to get it to fail. I am afraid I will have to copy and break down the large project. This will take me some effort.

```
package corundum

import spinal.core._
import spinal.lib._
import spinal.sim._
import spinal.core.sim._

import scala.util.Random

case class decryptPipe(instanceNr : Int = 0) extends Component {
  val sf = StreamFifo(Bits(8 bits), 8)
}

case class receiveDual(cd : ClockDomain) extends Component {
  val my_areas = Array.tabulate(2)(instanceNr => {
    new Area {
      val cd_area = new ClockingArea(cd) {
        val decrypt = decryptPipe(instanceNr)
      }
    }
  })
}

object UnderscoresSim {
  def main(args: Array[String]) : Unit = {
  SimConfig.withGhdl.withConfig(SpinalConfig(
    defaultConfigForClockDomains = ClockDomainConfig(
      resetKind = SYNC
    ),
    onlyStdLogicVectorAtTopLevelIo = true
  )).compile(new Component{
    val miaou = False

    val clk     = in Bool()
    val rst     = in Bool()

    val turbo_clk     = in Bits(2 bits)
    val turbo_rst     = in Bits(2 bits)

    val axiClockDomain = ClockDomain(
      clock = clk,
      reset = rst
    )

    val cd0 = ClockDomain(
      clock = turbo_clk.apply(0),
      reset = turbo_rst.apply(0)
    )

    val cd1 = ClockDomain(
      clock = turbo_clk.apply(1),
      reset = turbo_rst.apply(1)
    )

    val packetRx = new ClockingArea(axiClockDomain) {
      val x = receiveDual(cd = cd0)
    }

    setDefinitionName("miaou")
  }).doSim{dut =>
    simSuccess()
  }
}
}
```
gives
```
entity StreamFifo_1 is
  port(
    io_push_valid : in std_logic;
    io_push_ready : out std_logic;
    io_push_payload : in std_logic_vector(7 downto 0);
    io_pop_valid : out std_logic;
    io_pop_ready : in std_logic;
    io_pop_payload : out std_logic_vector(7 downto 0);
    io_flush : in std_logic;
    io_occupancy : out unsigned(3 downto 0);
    io_availability : out unsigned(3 downto 0);
    zz_1_1 : in std_logic;
    zz_2_1 : in std_logic
  );
end StreamFifo_1;

Somehow the first index (first number `1`) seems to disappear, such that `zz_1_1` ends up as `zz__1`.

Dolu1990, 2023-07-11:
As a workaround until we can find a way to reproduce it, you can do : 

```scala
    val cd0 = ClockDomain(
      clock = turbo_clk.apply(0).setName("miaou0"),
      reset = turbo_rst.apply(0).setName("wuff0")
    )
```

likewise, 2023-07-11:
Thanks for the workaround.

likewise, 2023-07-11:
Here is a stripped failing case. (It could make it smaller, but that hides the use-case.) We are running some blackboxed VHDL crypto in a dual parallel path, in a different clock domain.  (Hope I didn't make a user error there, however in that it would be nice if an error in Spinal.)

It seems related to the use of the Blackbox.

```scala
package underscores

import spinal.core._
import spinal.lib._

class cryptoBlackBox() extends BlackBox {
  // Define IO of the VHDL entity / Verilog module
  val io = new Bundle {
    val clk = in Bool()
    val rst = in Bool()
  }
  // Map the current clock domain to the io.clk and io.rst pins
  mapClockDomain(clock = io.clk, reset = io.rst)
}

case class decryptPipe() extends Component {
  val sf1 = StreamFifo(Bits(8 bits), 8)
  val vhdl = new cryptoBlackBox()
}

case class receiveDual(cryptoCD : ClockDomain) extends Component {
  val crypto_areas = Array.tabulate(2)(instanceNr => {
    new Area {
      val crypto_turbo = new ClockingArea(cryptoCD) {
        val decrypt = decryptPipe()
      }
    }
  })
}

case class TestCaseUnderscores() extends Component {
  val io = new Bundle{
    val clk     = in Bool()
    val rst     = in Bool()
    val turbo_clk     = in Bits(2 bits)
    val turbo_rst     = in Bits(2 bits)
  }

  val axiClockDomain = ClockDomain(
    clock = io.clk,
    reset = io.rst
  )

  val rxCryptoClockDomain = ClockDomain(
    clock = io.turbo_clk.apply(0),
    reset = io.turbo_rst.apply(0)
  )

  val packetRx = new ClockingArea(axiClockDomain) {
    val rx = receiveDual(cryptoCD = rxCryptoClockDomain)
  }
}

// will generate a VHDL files with identifiers with double underscore, or underscore at end
// those are marked illegal by GHDL
object TestCaseUnderscores {
  def main(args: Array[String]) : Unit = {
    val vhdlReport = SpinalConfig().generateVhdl(new TestCaseUnderscores())
  }
}

import scala.util.Random
import spinal.core.sim._
import scala.collection.mutable.ListBuffer

// will use GHDL to simulate wit the VHDL files with identifiers with double underscore, or underscore at end
// those are marked illegal by GHDL
object TestCaseUnderscoresSim {

  def main(args: Array[String]): Unit = {
    val simSlowDown = false

    val simConfig = SimConfig
    .withGhdl.addRunFlag("--unbuffered").addRunFlag("--ieee-asserts=disable").addRunFlag("--assert-level=none").addRunFlag("--backtrace-severity=warning")

    simConfig.compile{
      val dut = new TestCaseUnderscores()
      dut
    }
    .doSim{dut =>
    }
  }
}
```

Dolu1990, 2023-07-13:
Thanks for the self contained example, it realy helped ^^
I pushed a fix.

closed 2023-07-13 by Dolu1990

likewise, 2023-07-13:
Thanks!  I couldn't recognize the (pushed?) fix yet. What commit was it in?

Dolu1990, 2023-07-13:
Ahhh damned, i pushed it to the wrong branch XDXD
it was in https://github.com/SpinalHDL/SpinalHDL/commit/c3b0546d4418f02789fcf7b4089ef4940754dc78
i just pushed it in dev now.

commit c3b0546d (2023-07-13): Fix #1158 PhasePropagateNames now cleanup unnamed signals before naming stuff
commit 008c73f1 (2023-07-13): Fix #1158 PhasePropagateNames now cleanup unnamed signals before naming stuff