SpinalHDL/SpinalHDL #1159: RegIf AxiLite4 Problem:Data should be avaliable when Valid and Ready is active

zyn810039594, 2023-07-11:
[image]
This is a R channel of axilite4 generated by regif.....

likewise, 2023-07-11:
> [image]
> 
> This is a R channel of axilite4 generated by regif.....

Looks good. The read data is value 0.

Or is there a reason to think that value is not expected? We cannot tell from the screenshot.

zyn810039594, 2023-07-11:
[image]
The data after DoRead is 0000023d

zyn810039594, 2023-07-11:
> > [image]
> > This is a R channel of axilite4 generated by regif.....
> 
> Looks good. The read data is value 0.
> 
> Or is there a reason to think that value is not expected? We cannot tell from the screenshot.

[image]
The Data after DoRead is 0000023D.

zyn810039594, 2023-07-11:
For signal with same names,
[image]

andreasWallner, 2023-07-11:
Since the data send to be changing, what type of bitfield is this? Could you provide a small example that reproduces this behaviour so that we can debug it?

zyn810039594, 2023-07-12:
Ok, I've tested as this module:
```scala
class regAxiLite extends Component {
    val io = new Bundle {
        val axiLite = slave(AxiLite4(32, 32))
    }
    val busIf   = AxiLite4BusInterface(io.axiLite, (0x000, 48 Bytes), "Test_axilite")
    val test0   = busIf.newReg("Test_reg")
    val test0_0 = test0.field(Bool(), RW, 1)
    val test0_1 = test0.field(Bool(), RW, 0)
    val test0_2 = test0.field(Bool(), RW, 1)
    val test0_3 = test0.field(Bits(3 bit), RW, 7)
    val test0_4 = test0.fieldAt(pos = 8, UInt(8 bit), RW, 2)
    busIf.accept(RalfGenerator("header"))
}
```
This is my test code:
```scala
object regAxiLiteSim extends App {
    Config.sim.compile(new regAxiLite).doSim { dut =>
        // Fork a process to generate the reset and the clock on the dut
        dut.clockDomain.forkStimulus(period = 10)
        SimTimeout(100000)
        val port = dut.io.axiLite

        port.aw.valid #= false
        port.w.valid #= false
        port.ar.valid #= false
        port.r.ready #= false
        port.b.ready #= false

        sleep(1000)
        dut.clockDomain.waitActiveEdge()
        port.ar.valid #= true
        port.ar.addr #= 0
        port.ar.prot #= 0
        dut.clockDomain.waitActiveEdge()
        while (!port.ar.ready.toBoolean) {
            dut.clockDomain.waitActiveEdge()
            println("Wait for ar ready active...")
        }
        port.ar.valid #= false
        dut.clockDomain.waitActiveEdge()
        while (!port.r.valid.toBoolean) {
            dut.clockDomain.waitActiveEdge()
            println("Wait for r valid active...")
        }
        //Wait for a extra clk, it's in line with the rules, right?
        println("Wait for a extra clk...")
        dut.clockDomain.waitActiveEdge()
        port.r.ready #= true
        println("Data is " + port.r.data.toBigInt.toString())
        dut.clockDomain.waitActiveEdge()
        port.r.ready #= false
        sleep(1000)

    }
}
```
(Sim with Verilator)

The wave is:
[image]

cheerryio, 2023-07-17:
My way to solve the problem is to go back to spinalHDL version 1.8.0b or earlier

jijingg, 2023-07-22:
> My way to solve the problem is to go back to spinalHDL version 1.8.0b or earlier. And this works for me. Or you may wait for the issue to be fixed in later spinalHDL versions or you can develop on dev version.

This problem was not introduced by 1.8.0. In fact, there is a bug. It’s just that the preservation circuit of readData did not expose this problem
**old way** 
``` scala
  private def readGenerator() = {
    when(askRead){
        readData  :=  .....    //readData hold
    }.otherwise{
      readError := False
    }
  }
}
```
**after 1.8.0**   
``` scala
  private def readGenerator() = {
    when(askRead){
        readData  :=  ..... 
    }.otherwise{
      readData  := readDefaultValue  //rdata cannot be retained, otherwise there will be security risks , SO KEEP 0 or defualtValue
      readError := False
    }
  }
}
```
 so the fundamental solution is to modify `lib/bus/regif/BusIfAdapter/AxiLite4BusInterfa`

https://github.com/SpinalHDL/SpinalHDL/blame/dev/lib/src/main/scala/spinal/lib/bus/regif/BusIfAdapter/AxiLite4BusInterface.scala#L52

likewise, 2023-07-22:
> This problem was not introduced by 1.8.0. In fact, there is a bug. It’s just that the preservation circuit of readData did not expose this problem **old way**
I'm not convinced this is the correct fix.

Do you have a minimal example exposing the bug?

likewise, 2023-07-22:
I see, the new code resets readData in the wrong state of the transaction. 

See my comment on https://github.com/SpinalHDL/SpinalHDL/commit/230ab60ba549be816929661a27d7d4d7f08f1b54.
My suggestion is to revert that commit.

jijingg, 2023-07-23:
fix by expand  askread cycle  for workround . 
```
  val askRead   = axiAr.valid || (axiR.valid && !axiR.ready)
```
[image]

closed 2023-08-03 by jijingg