SpinalHDL/SpinalHDL #1161: Feature: translateWith() for Stream(Fragment())

likewise, 2023-07-14:
This is not my code, but was posted on Gitter once, I think it should be in lib.

```
  // translateWith() for Stream(Fragment())
  implicit class FragmentPimper[T <: Data](v: Fragment[T]) {
    def ~~[T2 <: Data](trans: T => T2) = {
      val that = trans(v.fragment)
      val res = Fragment(cloneOf(that))
      res.fragment := trans(v.fragment)
      res.last := v.last
      res
    }
  }
```
Example usages
```
  io.source << io.sink.~~(_.~~(_.asBits.subdivideIn(8 bits).reverse.asBits))
```

```
  // such that both sink and sink_length are skid buffered
  val x1 = Stream(Fragment(Bits(dataWidth + 12 bits)))
  x1 << io.sink.~~(_.~~(io.sink_length.asBits ## _)).s2mPipe().m2sPipe()
```

Readon, 2023-07-15:
use translateWith might be better option

likewise, 2023-07-15:
> use translateWith might be better option

Do you mean we should also add translateWith() (instead of ~~) for Stream Fragments? 

And that it looks clearer in code?

Dolu1990, 2023-07-19:
Ahhh i noticed that in Stream, there is a "def map" which is equivalent to ~~

Overall the ~~ syntax was very very much just a "throwed in"

likewise, 2023-07-19:
As the previous remarks were confused about ~~, let me re-summarize this issue:

We have wihTranslate for Stream(), but this does not work for Stream(Fragment()).

Dolu1990, 2023-07-19:
Ahhhh got it XD

Yes you are right, so a "fragmentMap" function could be added. that would solve the issue right ?

likewise, 2023-07-19:
I don't know -- could be :)

I think it should be the same interface as Stream .withTranslate() but now for the Fragment data (i.e. not the last signal).

The snippet of code I posted works for me, but I think we should have this functionality in the lib.

Dolu1990, 2023-07-24:
Ahhh i see, yes you right, the Fragment class itself could have the same API.
(I was thinking more from a Stream[Fragment[XXX]] perspective)