SpinalHDL/SpinalHDL #1162: How to use Ram_1wrs

zhuxihuaizuo, 2023-07-16:
I try to use Ram_1wrs like this
```scala
class TestRam extends Component {
  val io = new Bundle {
    val clk = in Bool()
    val rst_n = in Bool()
  }

  val mem = new Ram_1wrs(32, 1024, auto, writeFirst, dontCare, 0, false)
  mem.io.wr := False
  mem.io.en := True
  mem.io.wrData := 0
  mem.io.addr := 0
  mem.io.clk := io.clk

}
```
but there is something error
```
Error detected in phase PhaseCheckHiearchy
********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/[Ram_1wrs]/wr : in Bool) is driven by Bool(x), but isn't accessible in the null component.
    TestRam.<init>(TestRam.scala:16)
    TestRam$$anonfun$main$1.apply(TestRam.scala:26)
    TestRam$$anonfun$main$1.apply(TestRam.scala:26)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/[Ram_1wrs]/en : in Bool) is driven by Bool(x), but isn't accessible in the null component.
    TestRam.<init>(TestRam.scala:17)
    TestRam$$anonfun$main$1.apply(TestRam.scala:26)
    TestRam$$anonfun$main$1.apply(TestRam.scala:26)
    spinal.sim.JvmThread.run(SimManager.scala:51)
```
I want to know how to solve the problem.
Please give me a right example, thank you very much!

Dolu1990, 2023-07-17:
Hi,

I got 
```scala
class TestRam extends Component {
  val io = new Bundle {

  }
  
  val mem = new Ram_1wrs(32, 1024, auto, writeFirst, dontCare, 0, false)
  mem.io.wr := False
  mem.io.en := True
  mem.io.wrData := 0
  mem.io.addr := 0
}
```

To work fine for me. Note that the clockdomain signals are implicit, you don't need to define them in the toplevel, a default one is already provided to you. Also, the Ram_1wrs bind itself automaticaly to the current clock domain

You may have a issue a project setup issue ?
How did you run the generation ?

zhuxihuaizuo, 2023-07-18:
Thank you very much!

closed 2023-07-18 by zhuxihuaizuo