SpinalHDL/SpinalHDL #1170: Could someone can provide the instance of spi  use

Mido-sun, 2023-07-25:


Mido-sun, 2023-07-25:
I want to study   thanks

Dolu1990, 2023-07-25:
Do you mean how to use it in SpinalHDL ?
Or practical usage of SPI ? (i use it for SDCARD and nor flash access for instance on SaxonSoc)

Mido-sun, 2023-07-25:
> 

I want to know  how to use it in SpinalHDL. 
For example, some devices require SPI configuration, how do I  use it

Mido-sun, 2023-07-25:
It seems a little complex, maybe I dont get the points

Dolu1990, 2023-07-27:
i don't have simple example in hands, excepted the testbench, but that's in python cocotb : 
https://github.com/SpinalHDL/SpinalHDL/blob/48e7e15da1125edfb055ba53deb8f9719dc707b0/tester/src/test/scala/spinal/tester/scalatest/Apb3SpiMasterCtrlTester.scala#L26
https://github.com/SpinalHDL/SpinalHDL/blob/48e7e15da1125edfb055ba53deb8f9719dc707b0/tester/src/test/python/spinal/SpiTester/Apb3SpiMasterCtrlTester/Apb3SpiMasterCtrlTester.py#L4

Mido-sun, 2023-07-28:
> cocotb

thanks a lot , I learn a new sim way .
I will try .Any question will leave here. Thank you for you helping

Dolu1990, 2023-07-28:
cocotb was used by the past as a way to handle simulation for both Verilog and VHDL with a single Python testbench.

It isn't used anymore, as SpinalSim was implemented, and a few people added GHDL / IVerilog backends :)

Mido-sun, 2023-07-28:
oh I see. I have a problem installing spinalsim env in my company’network . the gcc version incompated with ventilator and I can’t change the version of gcc .🙁 back to older version

Mido-sun, 2023-07-29:
[image]
The other problem is that spinal_lib will not be pulled down for versions above 1.70 on the company network via sbt.

Mido-sun, 2023-07-29:
> val spinalVersion = "1.8.0b"
> 
> 
> scalaVersion := "2.11.12"
> 
> 
> 
> libraryDependencies ++= Seq(
> 
>   "org.scala-lang.modules" %% "scala-parser-combinators" % "1.1.2",
>   
>   "com.github.spinalhdl" % "spinalhdl-lib_2.11" % spinalVersion,
>   
>   "com.github.spinalhdl" % "spinalhdl-core_2.11" % spinalVersion,
> 
> 
> 
>   compilerPlugin("com.github.spinalhdl" % "spinalhdl-idsl-plugin_2.11" % spinalVersion)
> 
> )
> 
> fork := true

this is my sbt

Dolu1990, 2023-07-31:
> The other problem is that spinal_lib will not be pulled down for versions above 1.70 on the company network via sbt.

What error do you get when you run it ? (in the terminal)

> oh I see. I have a problem installing spinalsim env in my company’network . the gcc version incompated with ventilator and I can’t change the version of gcc .slightly_frowning_face back to older version

If you can, aim for verilator >= 4.041

Mido-sun, 2023-08-01:
> What error do you get when you run it ? (in the terminal)

nothing error show 
[info] welcome to sbt 1.8.2 (Oracle Corporation Java 18.0.2)
[info] loading global plugins from D:\sbt\.sbt\plugins
[info] loading settings for project bs_prj_xg-build from idea10.sbt ...
[info] loading project definition from D:\project\XG\XG1.0\code\03_spinalHDL\01_BS\bs_prj_XG\project
[info] loading settings for project bs_prj_xg from build.sbt ...
[warn] Discarding 1 session setting.  Use 'session save' to persist session settings.
[info] set current project to bs_prj_xg (in build file:/D:/project/XG/XG1.0/code/03_spinalHDL/01_BS/bs_prj_XG/)
[info] Defining Global / sbtStructureOptions
[info] The new value will be used by Global / ssOptions
[info] Reapplying settings...
[info] set current project to bs_prj_xg (in build file:/D:/project/XG/XG1.0/code/03_spinalHDL/01_BS/bs_prj_XG/)
[warn] sbt 0.13 shell syntax is deprecated; use slash syntax instead: Global / dumpStructureTo
[info] Writing structure to C:\Users\s00779302\AppData\Local\Temp\sbt-structure.xml...
[info] Done.
[success] Total time: 1 s, completed 2023楠�?閺�?閺�?娑撳﹤宕�11:34:04
4
[info] Reapplying settings...
[info] set current project to bs_prj_xg (in build file:/D:/project/XG/XG1.0/code/03_spinalHDL/01_BS/bs_prj_XG/)
[info] Defining Global / ideaPort
[info] The new value will be used by Compile / compile, Test / compile
[info] Reapplying settings...
[info] set current project to bs_prj_xg (in build file:/D:/project/XG/XG1.0/code/03_spinalHDL/01_BS/bs_prj_XG/)
[IJ]

[image]

Mido-sun, 2023-08-01:
everything pull down excepts spinal.lib

Mido-sun, 2023-08-02:
and about spinal sim  I install the 5.004 verilator   it shows the error below 

> [Progress] Verilator compilation started
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity___024root__DepSet_h7f1f0f6e__0__Slow.o] 错误 1
> make: *** 正在等待未完成的任务....
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity__Syms.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity___024root__DepSet_hb11b3ea2__0__Slow.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:239：VIdentity___024root__DepSet_hb11b3ea2__0.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity___024root__Slow.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity__Trace__0__Slow.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:239：VIdentity__Trace__0.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:239：VIdentity___024root__DepSet_h7f1f0f6e__0.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:239：VIdentity.o] 错误 1
> make: *** [VIdentity.mk:77：VIdentity__spinalWrapper.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:245：verilated_threads.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:245：verilated_vcd_c.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:245：verilated.o] 错误 1
> Exception in thread "main" java.lang.AssertionError: assertion failed: Verilator C++ model compilation failed
> 	at scala.Predef$.assert(Predef.scala:170)
> 	at spinal.sim.VerilatorBackend$$anonfun$compileVerilator$2.apply$mcV$sp(VerilatorBackend.scala:651)
> 	at spinal.sim.VerilatorBackend.cacheSynchronized(VerilatorBackend.scala:75)
> 	at spinal.sim.VerilatorBackend.compileVerilator(VerilatorBackend.scala:616)
> 	at spinal.sim.VerilatorBackend.<init>(VerilatorBackend.scala:735)
> 	at spinal.core.sim.SpinalVerilatorBackend$.apply(SimBootstraps.scala:144)
> 	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:975)
> 	at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:908)
> 	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:885)
> 	at TestIdentity$.delayedEndpoint$TestIdentity$1(practice.scala:14)
> 	at TestIdentity$delayedInit$body.apply(practice.scala:12)
> 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
> 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
> 	at scala.App$$anonfun$main$1.apply(App.scala:76)
> 	at scala.App$$anonfun$main$1.apply(App.scala:76)
> 	at scala.collection.immutable.List.foreach(List.scala:392)
> 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
> 	at scala.App$class.main(App.scala:76)
> 	at TestIdentity$.main(practice.scala:12)
> 	at TestIdentity.main(practice.scala)

Mido-sun, 2023-08-02:
gcc version 13.2.0 (Rev1, Built by MSYS2 project)

Mido-sun, 2023-08-03:
@Dolu1990

Dolu1990, 2023-08-03:
I would say, do not use verilator 5 for now, it is very fresh and has major new features.

I will take a look at verilator 5

Else all good right ?

Mido-sun, 2023-08-03:
> I would say, do not use verilator 5 for now, it is very fresh and has major new features.
> 
> I will take a look at verilator 5
> 
> Else all good right ?

except sim problem ,the spinal lib wont pull still exist .  information above

Dolu1990, 2023-08-03:
Just tried verilator head (Verilator 5.013 devel rev v5.012-75-g9caa79a7e)
works fine on debian.

So you may have a verilator instalation issue on your env ?
Did you followed the SpinalDoc for the install of verilator ?

For the sbt, can you try with : 
https://github.com/SpinalHDL/SpinalTemplateSbt/blob/master/build.sbt  ?

Mido-sun, 2023-08-03:
> Just tried verilator head (Verilator 5.013 devel rev v5.012-75-g9caa79a7e) works fine on debian.
>
> So you may have a verilator instalation issue on your env ? Did you followed the SpinalDoc for the [...]

I BACK the verilator version to 4.228 , still

>[Progress] Verilator compilation started

> make: *** [/mingw64/share/verilator/include/verilated.mk:245：VIdentity___024root__Slow.o] 错误 1
> make: *** 正在等待未完成的任务....
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity___024root__DepSet_hb11b3ea2__0.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:245：VIdentity___024root__DepSet_hb11b3ea2__0__Slow.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:245：VIdentity__Syms.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity__Trace__0.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:245：VIdentity__Trace__0__Slow.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:242：VIdentity.o] 错误 1
> make: *** [VIdentity.mk:77：VIdentity__spinalWrapper.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:248：verilated_vcd_c.o] 错误 1
> make: *** [/mingw64/share/verilator/include/verilated.mk:248：verilated.o] 错误 1
> Exception in thread "main" java.lang.AssertionError: assertion failed: Verilator C++ model compilation failed
> 	at scala.Predef$.assert(Predef.scala:170)
> 	at spinal.sim.VerilatorBackend$$anonfun$compileVerilator$2.apply$mcV$sp(VerilatorBackend.scala:651)
> 	at spinal.sim.VerilatorBackend.cacheSynchronized(VerilatorBackend.scala:75)
> 	at spinal.sim.VerilatorBackend.compileVerilator(VerilatorBackend.scala:616)
> 	at spinal.sim.VerilatorBackend.<init>(VerilatorBackend.scala:735)
> 	at spinal.core.sim.SpinalVerilatorBackend$.apply(SimBootstraps.scala:144)
> 	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:975)
> 	at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:908)
> 	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:885)
> 	at TestIdentity$.delayedEndpoint$TestIdentity$1(practice.scala:14)
> 	at TestIdentity$delayedInit$body.apply(practice.scala:12)
> 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
> 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
> 	at scala.App$$anonfun$main$1.apply(App.scala:76)
> 	at scala.App$$anonfun$main$1.apply(App.scala:76)
> 	at scala.collection.immutable.List.foreach(List.scala:392)
> 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
> 	at scala.App$class.main(App.scala:76)
> 	at TestIdentity$.main(practice.scala:12)
> 	at TestIdentity.main(practice.scala)

Mido-sun, 2023-08-03:
[image]

Mido-sun, 2023-08-03:
> 

is there any relation with the gcc version ? mine gcc 13.2.0

Dolu1990, 2023-08-03:
Yes, Such a recent GCC version may rise some verilator incompatibility. I'm on 10.1.0

Main issue i have is that your error message doesn't tell much :(
maybe, try to delete the simWorkspace folder between try with different tool (to purge the cache)

about the lack of spinallib,
what is the log of "sbt clean compile" ?

Mido-sun, 2023-08-03:
> what is the log of "sbt clean compile" ?

how can I check this log   
thanks

Dolu1990, 2023-08-03:
just run it in a terminal at the root of your project, and copy past the messages in the terminal

Mido-sun, 2023-08-03:
> sbt thinks that server is already booting because of this exception:
> sbt.internal.ServerAlreadyBootingException: java.io.IOException: Could not create lock for \\.\pipe\sbt-load1674937071948918792_lock, error 5
>         at sbt.internal.BootServerSocket.newSocket(BootServerSocket.java:356)
>         at sbt.internal.BootServerSocket.<init>(BootServerSocket.java:295)
>         at sbt.xMain$.getSocketOrExit(Main.scala:150)
>         at sbt.xMain$.bootServerSocket$lzycompute$1(Main.scala:77)
>         at sbt.xMain$.bootServerSocket$1(Main.scala:77)
>         at sbt.xMain$.withStreams$1(Main.scala:85)
>         at sbt.xMain$.run(Main.scala:121)
>         at java.base/jdk.internal.reflect.DirectMethodHandleAccessor.invoke(DirectMethodHandleAccessor.java:104)
>         at java.base/java.lang.reflect.Method.invoke(Method.java:577)
>         at sbt.internal.XMainConfiguration.run(XMainConfiguration.java:57)
>         at sbt.xMain.run(Main.scala:46)
>         at xsbt.boot.Launch$.$anonfun$run$1(Launch.scala:149)
>         at xsbt.boot.Launch$.withContextLoader(Launch.scala:176)
>         at xsbt.boot.Launch$.run(Launch.scala:149)
>         at xsbt.boot.Launch$.$anonfun$apply$1(Launch.scala:44)
>         at xsbt.boot.Launch$.launch(Launch.scala:159)
>         at xsbt.boot.Launch$.apply(Launch.scala:44)
>         at xsbt.boot.Launch$.apply(Launch.scala:21)
>         at xsbt.boot.Boot$.runImpl(Boot.scala:78)
>         at xsbt.boot.Boot$.run(Boot.scala:73)
>         at xsbt.boot.Boot$.main(Boot.scala:21)
>         at xsbt.boot.Boot.main(Boot.scala)
> Caused by: java.io.IOException: Could not create lock for \\.\pipe\sbt-load1674937071948918792_lock, error 5
>         at org.scalasbt.ipcsocket.Win32NamedPipeServerSocket.<init>(Win32NamedPipeServerSocket.java:129)
>         at org.scalasbt.ipcsocket.Win32NamedPipeServerSocket.<init>(Win32NamedPipeServerSocket.java:48)
>         at sbt.internal.BootServerSocket.newSocket(BootServerSocket.java:350)
>         ... 21 more
> Create a new server? y/n (default y)
> y
> [info] welcome to sbt 1.8.2 (Oracle Corporation Java 18.0.2)
> [info] loading global plugins from D:\sbt\.sbt\plugins
> [info] loading project definition from D:\project\XG\XG1.0\code\03_spinalHDL\01_BS\bs_prj_XG\project
> [info] loading settings for project projectname from build.sbt ...
> [info] set current project to projectname (in build file:/D:/project/XG/XG1.0/code/03_spinalHDL/01_BS/bs_prj_XG/)
> [info] Executing in batch mode. For better performance use sbt's shell
> [success] Total time: 1 s, completed 2023骞?鏈?鏃?涓嬪崍5:07:28
> [success] Total time: 1 s, completed 2023骞?鏈?鏃?涓嬪崍5:07:28
> [0J

Mido-sun, 2023-08-04:
the spinal sim problem solved by installing older verilator  thank you very much

Dolu1990, 2023-08-04:
Hmm did you tried to delete the lock file ? 
(Could not create lock for \.\pipe\sbt-load1674937071948918792_lock, error 5)

Mido-sun, 2023-08-04:
i tried  but  still no work for pull spinal.lib
（no ）

> Could not create lock for .\pipe\sbt-load1674937071948918792_lock, error 5

Mido-sun, 2023-08-07:
> hmm maybe, try with another sbt version, can you try 1.6.0 ? (the version is set in one of the file in the project folder)
 It is not OK  
I pull  1.7.0 and older version  the lib will be pulled  
but later version is not OK 
now I install higher version spinal using https://oss.sonatype.org/content/groups/public/com/github/spinalhdl/spinalhdl-tester_2.12/1.8.2/ to download jar  lmao

Dolu1990, 2023-08-07:
XD sometime, the sbt / maven stuff break, and fix himself over time.

So all good now ?

Mido-sun, 2023-08-08:
yeah , it is ok that I install higher version of spinal manually  <_>

Mido-sun, 2023-08-09:
I found when the scalaVersion == 2.13.11  ，the lib will be pulled  .But it cause the incompability of spinal code .the code I write before shows error

> ThisBuild / version := "1.0"
> ThisBuild / scalaVersion := "2.13.11"
> ThisBuild / organization := "org.example"
> 
> val spinalVersion = "1.8.2"
> val spinalCore = "com.github.spinalhdl" %% "spinalhdl-core" % spinalVersion
> val spinalLib = "com.github.spinalhdl" %% "spinalhdl-lib" % spinalVersion
> val spinalIdslPlugin = compilerPlugin("com.github.spinalhdl" %% "spinalhdl-idsl-plugin" % spinalVersion)
> 
> lazy val projectu6g3rd = (project in file("."))
>     .settings(
>         Compile / scalaSource := baseDirectory.value / "hw" / "spinal",
>         libraryDependencies ++= Seq(spinalCore, spinalLib, spinalIdslPlugin)
>     )
> 
> fork := true

Mido-sun, 2023-08-09:
@Dolu1990

Mido-sun, 2023-08-09:
why I update spinal with 2.13.11  ， some syntax mistakes shows  ， like

>   val a = UInt(7 bits ) setAsReg()

> can't supply unit value with infix notation because nullary method setAsReg: (): spinal.core.UInt takes no arguments; use dotted invocation instead: spinal.core.`package`.UInt(spinal.core.`package`.IntToBuilder(6).bits).setAsReg()

Mido-sun, 2023-08-09:
[image]

and this

Dolu1990, 2023-08-09:
Scala 2.13 is kinda of a proxy to scala 3, and break some compatibility.
Better staying on scala 2.11 or 2.12

Mido-sun, 2023-08-10:
that is weried that 2.12 or 2.11 ,the spinal LIB will not pulled . nevermind I quit

closed 2023-10-10 by Mido-sun