SpinalHDL/SpinalHDL #1171: Supporting generate saif file with VCS backend

hanm2019, 2023-07-30:
SAIF file is important for power estimation. Would we implement an flag for vcs backend, such as `withSAIF` to generate the saif file?

I found some verilog example to generate SAIF file.
https://github.com/CMU-SAFARI/NOCulator/blob/6a58c1223dc7424f11a40a3244c07b73e8dc3030/hring/hw/calf/tb2.v#L69

It seems that at the beginning of simulation, we need to set the toggle region and start the toggle.
 
```
initial begin
                $set_toggle_region(tb.r);
                $toggle_start();
		end
```

and before the end of simulation, we need to stop toggle and generate the saif file.

```
$toggle_stop();
$toggle_report("./calf_backward_2.saif", 1.0e-9, "tb.r");
$finish
```

I also found that the FSDB wave file can convert to SAIF file by `FSDB2SAIF` command, but this slution is high cost when the hardware design is quite large and we don't want to store the whole wavefile.

Dolu1990, 2023-07-31:
So, mostly, the primitive you would need is to be able to add lines of verilog in a initial begin in the toplevel ?

hanm2019, 2023-07-31:
Certainly.

Dolu1990, 2023-08-04:
One question is, those initial begin statements need to be in the toplevel ? Or they can be in a sub component just fine ?

hanm2019, 2023-08-09:
I found two solutions: 

1. the begin statement and report statement are in the testbench file:

wrapper.v

```
`timescale 1ns / 1ns
module Wrapper( 
input a,
input b,
output reg c,
input clk);

always@(posedge clk) begin
	c <= a | b;
end

endmodule
```

tb.v

```
`timescale 1ns/1ns
module tb();
reg clk;
reg a,b;
wire result;
initial begin
	clk = 1'b0;
	forever begin
		#1 clk = ~clk;
	end
end

Wrapper DUT(.a(a), .b(b), .c(result), .clk(clk));

initial begin
	a = 1'b0;
	b = 1'b1;
	#10
	a = 1'b1;
	b = 1'b1;
	#10
	a = 1'b0;
	b = 1'b0;
	#10
	a = 1'b1;
	b = 1'b0;
end

initial begin
 $set_toggle_region(tb.DUT);
 $toggle_start();
 #100
 $toggle_stop();
 $toggle_report("Wrapper.saif", 1.0e-9, "tb.DUT");
 $finish;

end

endmodule

```

2. the begin statement in the top-level module while the report statement in the testbench file:

Wrapper.v

```
`timescale 1ns / 1ns
module Wrapper( 
input a,
input b,
output reg c,
input clk);

always@(posedge clk) begin
	c <= a | b;
end

initial begin
	$set_toggle_region(Wrapper);
	$toggle_start();
end

endmodule 
```

tb.v

```
`timescale 1ns/1ns
module tb();
reg clk;
reg a,b;
wire result;
initial begin
	clk = 1'b0;
	forever begin
		#1 clk = ~clk;
	end
end

Wrapper DUT(.a(a), .b(b), .c(result), .clk(clk));

initial begin
	a = 1'b0;
	b = 1'b1;
	#10
	a = 1'b1;
	b = 1'b1;
	#10
	a = 1'b0;
	b = 1'b0;
	#10
	a = 1'b1;
	b = 1'b0;
end

initial begin
 #100
 $toggle_report("Wrapper.saif", 1.0e-9, "tb.DUT");
 $finish;

end

endmodule
```