SpinalHDL/SpinalHDL #1172: Is there any easy way to generate ascii signal?

chenbo-again, 2023-08-03:
```scala
val type: Bits = S2B"OK"
```

It's useful for debug complex protocol.

Dolu1990, 2023-08-03:
Agree, ASCII to Bits convertion is missing.
I would say, it doesn't need to be a interpolator itself (to avoid being too narrow)
But could be instead a cast like thing : 
B("OK")
?

andreasWallner, 2023-08-03:
I have a small snippet that I was considering to open a PR for if there is some interest, it adds a signal with ASCII content but ATM only with a `#=` not yet with a `:=` operator (since I mainly used it for debugging from testcases)

https://github.com/andreasWallner/spinalStuff/blob/7ba86e97255cfe39e23443eb32b8068e371ea370/src/main/scala/andreasWallner/sim/package.scala#L298

yportne13, 2023-08-03:
why not convert to BigInt?
```scala
def stringToBigInt(x: String): BigInt = {
  x.getBytes()
    .map(x => BigInt(x))
    //.reverse
    .reduce(_ * 256 + _)
}
```

andreasWallner, 2023-08-17:
> Agree, ASCII to Bits convertion is missing. I would say, it doesn't need to be a interpolator itself (to avoid being too narrow) But could be instead a cast like thing : B("OK") ?

Wouldn't that be a problem insofar that it clashes with the current `B("110000")`?

kleinai, 2023-08-17:
Just my two cents, doesn't Java/Scala already have ways to convert text into bytes? That would side step text encoding headaches.