SpinalHDL/SpinalHDL #1176: How to get init value of Reg type

kellenwang1017, 2023-08-08:
I wanna create a bunch of Regs in serial that follows a certain Reg R (just to create N clock delayed copy of R), and copy the init value of R. I checked the code or Reg.init(), it uses assignInit() to initialise the Reg, and no public val or any other method provided to the the value once passed to init() method. Any suggestion?

likewise, 2023-08-08:
Hmm, is `Delay()` what you want?

Dolu1990, 2023-08-08:
There is currently no real API to retreive the value used to init something.
But it can be done : 

```scala
    val miaou = Reg(UInt(8 bits)) init(42)
    
    miaou.foreachStatements{
      case s : InitAssignmentStatement => {
        println(s)
        s.source match {
          case bt : UInt => {
            println(bt)
            bt.foreachStatements{
              case s : DataAssignmentStatement => {
                println(s)
                s.source match {
                  case lit : UIntLiteral => {
                    println(lit.value)
                  }
                }
              }
            }
          }
        }
      }
      case _ =>
    }
    
    will print 
    (toplevel/miaou :  UInt[8 bits]) := (U"101010" 6 bits)
    (U"101010" 6 bits)
    (U"101010" 6 bits) := (U"101010" 6 bits)
    42
```

kellenwang1017, 2023-08-08:
Dolu you are the best!!!!!!!!!!!!!!!!!

closed 2023-08-08 by kellenwang1017