SpinalHDL/SpinalHDL #1177: Syntax error in IDEA 2023.2

dahuwei, 2023-08-15:
Hi, when I upgrade IDEA to 2023.2, the editor report many syntax errors for the scala code using spinal. But the source code can be executed successfully to generate verilog code. Do I need to change any IDEA configurations to remove the incorrect error reports in the editor?

Dolu1990, 2023-08-15:
Hi,

So, IDEA had kind of a history of having quite a few false positive red marking. 5 years ago it was kind of a fight XD, then it went ok / good.
But seems like they messedup things again XD
I would say, can you use a older version of IDEA ?
Until they fix things :)

dahuwei, 2023-08-15:
> Hi,
> 
> So, IDEA had kind of a history of having quite a few false positive red marking. 5 years ago it was kind of a fight XD, then it went ok / good. But seems like they messedup things again XD I would say, can you use a older version of IDEA ? Until they fix things :)

Thank you, I have rolled back idea to 2023.1.5, this version do not have the false positive red marking.

closed 2023-08-18 by dahuwei

dlmiles, 2023-08-22:
FWIW I am attempting a bisect of this issue, with the project https://github.com/JetBrains/intellij-scala

What I see is:
* this issue was introduced sometime before 2023.2.19
* this issue affects all SpinalHDL `Reg` constructor use in a project, its methods and the following type inference it broken as it does not resolve the type as every point of use (there maybe other issues, but due to the volume of Reg use it is not easy to see).

Scala Plugin Version bisect (so far):

2023.2.13 failed
2023.2.12 failed
2023.2.7 failed
2023.2.4 failed
2023.2.3 ????
2023.2.2 pass

kleinai, 2023-08-22:
I was on plugin 2023.2.17 when I first noticed this myself.  Easy workaround is to explicitly provide a type when using the `Reg` patterns.