SpinalHDL/SpinalHDL #1179: Spi

Amegodvivado, 2023-08-16:


Amegodvivado, 2023-08-16:
Hi,author,I want to know whether 3-wire,interface is supported？another one, In SpiMasterCtrl.scala and SpiSlaveCtrl.scala ：line9：`Created by PIC32F_USER on 02/08/2017` 
 and line 32：
   `  * In short, it has one command fifo (write/read/chip-select commands) and one read fifo.
     .............
     * config -> 0x08
   * - cpol -> W[0]
   * - cpha -> W[1]
   **/`,
Which protocol standard is used to configure the registers here, is it NXP?

andreasWallner, 2023-08-17:
Hi,

CPOL plainly encodes the idle state of the clock.
CPHA 0 should mean "sample first" (check lines 106, 125 and 128)
It should therefore match the graphic on wikipedia: https://de.wikipedia.org/wiki/Serial_Peripheral_Interface#/media/Datei:SPI_timing_diagram2.svg

When you ask about 3-wire, do you mean an interface with CLK, select and half-duplex data? That is currently not supported.

Amegodvivado, 2023-08-18:
Thank you for your answer.I wonder which standard protocol is the SPI in the library based on, is it the SPI Block Guide V04.01 standard or other PICF series?For 3-wire interface,it likes use only one port mosi to half-duplex,without miso port.It seems like this is not supported.

Dolu1990, 2023-08-18:
the SpiXdrMasterCtrl does support it i think, it's just about calling the addFullDuplex function in its Parameter class.

Amegodvivado, 2023-08-18:
Oh，thank you for your answer,i miss the SpiXdrMasterCtrl.By the way  What's the meaning of "xdr".I can't search the explanation.
so we can implement 3-wire half-duplex mode and 4-wire full-duplex mode,right?like Standard SPI、Dual SPI and Queued SPI.

Dolu1990, 2023-08-21:
Xdr mean Xdefined Data Rate  meaning that the PHY layer could go faster than the controller itsefl (SDR, DDR, QDR, ..)

> so we can implement 3-wire half-duplex mode and 4-wire full-duplex mode,right?like Standard SPI、Dual SPI and Queued SPI.

Yes, all at once in a single controller, given the hardware generation parameters given specify the correspondings mods

Amegodvivado, 2023-08-30:
Thanks！By the way，can burst transmission be implemented in SPI-related libs?

andreasWallner, 2023-09-03:
For both of the controllers the SW can choose when the SS signals should change, so you can send as many bytes if you want w/o changing SS. Does that answer your question?

Amegodvivado, 2023-09-04:
thanks  for your  answer ，the burst transmission like axi interface to  spi（xilinx ip-AXI Quad Spi） ，It is mainly used for data interaction with FIFO 。by the way， no baudrate configration in spi ctrl lib  right？

Dolu1990, 2023-09-04:
>  no baudrate configration in spi ctrl lib right

That's a runtime configuration, via the register accesses bus

andreasWallner, 2023-09-26:
@Amegodvivado
did that answer your question? in that case can we close this?

Amegodvivado, 2023-10-16:
> @Amegodvivado did that answer your question? in that case can we close this?

ok,thank you for your answer

Amegodvivado, 2023-11-07:
In SpiSlaveCtrl.scala  of  tx logic and rx logic. blow:
tx logic:` val (stream, fifoAvailability) = streamUnbuffered.queueWithAvailability(rxFifoDepth)`
rx logic:`val (stream, fifoOccupancy) = rx.takeWhen(listen).queueWithOccupancy(txFifoDepth)`

Are the transmission parameters here reversed? in other words，Why in tx logic  use rxFifoDepth，rx logic use txFifoDepth

andreasWallner, 2023-11-08:
@Amegodvivado yes, looks to me like we flipped those