SpinalHDL/SpinalHDL #1180: Exception in thread "main" java.lang.Exception: ISim engine error: Failed to Load up XSI.

zwxb, 2023-08-16:
Software operating environment：SpinalHdl 1.9.0，vivado2018，OS：win10

```
case class add() extends Component {
  val io = new Bundle {
    val a = in UInt (8 bits)
    val b = in UInt (8 bits)
    val c = out UInt (8 bits)
  }
  noIoPrefix()

  io.c := io.a + io.b
}
```

```
object Simadd {
  def main(args: Array[String]): Unit = {
    val a = SimConfig
    a.withWave.withXSim.compile(add()).doSim { dut =>

      dut.clockDomain.forkStimulus(10000)

    }
  }
}
```
spinal_xsim.h

```
#define SIM_KERNEL "librdi_simulator_kernel.dll"
#define SIM_DESIGN "xsim.dir/add_behav/xsimk.dll"
#define SIM_WAVE   "test.wdb"
#define SIM_HAS_WAVE true
```
```
[Progress] at 1.226 : Generate Verilog
[Done] at 1.480
[Progress] Simulation workspace in D:\PC_bak\TestXsim\.\simWorkspace\add
[Progress] XSIM compilation started
Exception in thread "main" java.lang.Exception: ISim engine error: Failed to Load up XSI.
	at spinal.sim.xsi.JNIXSIIfaceJNI.new_XSIIface(Native Method)
	at spinal.sim.xsi.XSIIface.<init>(XSIIface.java:40)
	at spinal.sim.XSimBackend.getInterface(XSimBackend.scala:327)
	at spinal.sim.SimXSim.<init>(SimXSim.scala:9)
	at spinal.core.sim.SpinalSimConfig$$anon$8.newSimRaw(SimBootstraps.scala:1089)
	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:581)
	at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:551)
	at Simadd$.main(Simadd.scala:5)
	at Simadd.main(Simadd.scala)
```

likewise, 2023-08-16:
On Linux for XSim (Xilinx Simulator, right?) to start was to set LD_LIBRARY_PATH.

zwxb, 2023-08-16:
xsim is Xilinx Simulator, and the OS I use is win10.

environment variable and PATH

[image]

zwxb, 2023-08-21:
I tried 4 versions of vivado on my own computer and finally found that 2018 2019 could not be used, and 2021 and 2022 could be used, and I initially suspected that the problem was a version drop.

closed 2023-08-21 by zwxb

Readon, 2023-08-21:
If it is applicable, would you please add this to the documentation. It seems that someone has asked about the minimum version that is supported before.

Readon, 2023-08-21:
#816 is a similar issue, but did not test as much as this.

Mido-sun, 2023-11-07:
@Dolu1990

andreasWallner, 2023-11-07:
@Mido-sun the linked issue looks like @kleinai couldn't reproduce the issue there, and I'm not sure we have another user that is on Windows & using Vivado 2018.3 - so you might have to look into that yourself, fixes are always welcome

Mido-sun, 2023-11-08:
> Could not load XSI simulation shared library (xsim.dir/Hmc7044CtrlrTB_behav/xsimk.dll): Failed to load shared library "xsim.dir/Hmc7044CtrlrTB_behav/xsimk.dll". �Ҳ���ָ����ģ�顣
> 
> Exception in thread "main" java.lang.Exception: ISim engine error: Failed to Load up XSI.

I don't use dump wave like them

Readon, 2023-11-10:
Another question is that where can I find the log of XSI native interface execution.
It is found that there are standard output in XSIIface.cpp as below
``` cpp
XSIIface::XSIIface(): loader{SIM_DESIGN, SIM_KERNEL} {
...
   std::cout << "Design lib: " << sim_design << std::endl;
```
But no log is found.

On other side, I find that in this constructor get_sim_time_precision is called, which I could not tell is support by vivado from which version.