SpinalHDL/SpinalHDL #1182: False assertion error after elaboration failure

nmk456, 2023-08-21:
I've been getting the following error in a project that uses VexRiscv:

```
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 1.947 : Elaborate components
[error] Exception in thread "main" java.lang.AssertionError: assertion failed
[error]         at scala.Predef$.assert(Predef.scala:156)
[error]         at spinal.core.package$.assert(core.scala:497)
[error]         at vexriscv.plugin.DecoderSimplePlugin.addDefault(DecoderSimplePlugin.scala:67)
[error]         at vexriscv.plugin.DBusSimplePlugin.setup(DBusSimplePlugin.scala:348)
[error]         at vexriscv.plugin.DBusSimplePlugin.setup(DBusSimplePlugin.scala:285)
[error]         at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
[error]         at vexriscv.Pipeline$$anonfun$build$2.apply(Pipeline.scala:47)
...
```

It appears that this error is entirely unrelated to the actual problem. Instead, since DecodeSimplePlugin is getting called twice, it's triggering an [assertion](https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/DecoderSimplePlugin.scala#L67) and failing, preventing the real problem from being reach and throwing the correct error. I was able to work around this by commenting out the assert but it doesn't solve the real problem of the state not being properly reset on this restart. This seems like more of SpinalHDL issue than a VexRiscv issue because this issue could potentially show up in any module.

Dolu1990, 2023-08-21:
Hi,
Yes right.

This can occure if you defined the VexRiscv parameters outside the SpinalVerilog{... here ...} body.
How is your scala main ?

Dolu1990, 2023-08-21:
> Is this the best way to do it?

Yes :)

Else  you can define the config as a def instead of a val

andreasWallner, 2023-10-26:
Looks like @Dolu1990's response fixed this, please reopen if there is something else.

closed 2023-10-26 by andreasWallner