SpinalHDL/SpinalHDL #1184: Is here any easy way to connect two bus with bits only?

nanoric, 2023-08-25:
Hi, im new to FPGA, and found that SpinalHDL is really a good library to design with FPGA.
But something still confuse me a lot.

--- 
consider this example:
```
object Test {
  def main(args: Array[String]) {
    val config = SpinalConfig(
      mode = Verilog)
      .generate(new Component {
        val o = out UInt (8 bits) // Dummy output against pruning

        val mySubComponent = new Component {
          val busA1 = in UInt (8 bits) // The relevant input
          val busA2 = in UInt (8 bits) // The relevant input
          val busA3 = out Bool()

          val busB1 = out UInt (16 bits)
          val busB2 = in Bool()

          Vec(busA1, busA2, busA3) <> Vec(busB1, busB2)
        }
      })
      .printPruned()
      .printPrunedIo()
      .printUnused()
  }
}
```

This doesn't work because auto connect with Vec has two requirements: 
1. same length
2. same type

But in my opinion they are just **bits** of a **same length** with **in/out** so they should be able to auto-connected.

In fact i use the Cat operator first and found this issue:
https://github.com/SpinalHDL/SpinalHDL/issues/275

Then I turn to Vec. But it seems that Vec couldn't fit my requirement.

---
Why I need it:
I have a blackbox busA, it has two channel:
```
bit [32*2-1:0] data;
bit [1:0] valid;
bit [1:0] ack;
# In fact it can be split into two channels that are completely independent of each other.:
bit [31:0] data0;
bit valid0;
bit ack0;
bit [31:0] data1;
bit valid1;
bit ack1;
```
I can of course separate it easily in the varilog source.
But since I have used SpinalHDL, I want all my code (except BlackBox) to be implemented by SpinalHDL.

Because if one day I want to make any changes on this, it must be easier and more enjoyable to do it on SpinalHDL than on Verilog/SystemVerilog.

andreasWallner, 2023-08-25:
I'll try to answer in two steps, first the philosophical part, then solutions to your questions.

> But in my opinion they are just bits of a same length with in/out so they should be able to auto-connected.
SpinalHDL is/wants to be a strictly typed language. This means (among other things) that even if two hardware signals would have the same width, you can't assign them if they have different interpretation of the data. (i.e. two 8-bit UInt vs. one 16-bit UInt).
The hope is that this makes most code easier to understand (e.g. because you explicitly state what the signals in an interface mean in the types rather than in some comments) and less error prone (you can't accidentially assign a signed signal to an input that expects an unsigned signal). This may come at some cost if you open the hatch and want to do something that you know is correct, but does not look like it in the model of the HDL.

Maybe this also answers (to a part) your question in #275: we wouldn't want assignment through functions that convert the type, as this would create an "alias" that breaks typing.

But: there are sometimes corner cases (especially with stuff like verilog BlackBoxes which are simply missing some of the type information that SpinalHDL has) where you need to work around this.

One possible solution to the initial example that you provided would be:
```scala
        val busA1 = in UInt (8 bits) // The relevant input
        val busA2 = in UInt (8 bits) // The relevant input
        val busA3 = out Bool ()
        val busA4 = out Bool ()

        val busB1 = out UInt (16 bits)
        val busB2 = in UInt (2 bits)

        (busB1, busA3, busA4) := (busA1, busA2, busB2).asBits
```

This makes use of tuple-assignment, a convenience syntax that you can read up on [here](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Semantic/assignments.html#assignments)). The `asBits` conversion is basically you opting in that you know what you are doing ;-).

With the example that you have given I know of no way how you'd be able to use `<>` for assignment.
It depends a bit on the circumstances, but if I wanted something more streamlined then I think I'd define two Bundles (for the two blocks of signals that you've shown) and provide a function on one of them to convert to the other. (If this comes up more often, if it's a one-off I might not if I don't already have the Bundles)

On the topic of your BlackBox:
When I need to use them then I'll often make a Bundle that represents the BlackBox I/Os in the appropriate SpinalHDL types (you blackbox looks like I'd be `Stream(Bits(32 bit))` in SpinalHDL). Sometimes you can stick that Bundle in the interface and just use `.setName()` to fix the names to match what you have. 
In you case this does not work, so I'd most likely put a member function on the BlackBox that creates the Bundle in question, assigns to it and returns it (using caching if needed)
```scala
// guessing on the directions
case class BlackBoxIO() {
  val data0 = Stream(Bits(32 bit))
  val data1 = Stream(Bits(32 bit))
}
case class MyBlackBox() extends BlackBox {
  protected val data = in port Bits(64 bit)
  protected val valid = in port Bits(2 bit)
  protected val ack = out port Bits(2 bit)

  def io = signalCache(this -> "io") {
    val io = BlackBoxIO()
    data := (io.data0.payload, io.data1.payload).asBits
    valid := (io.data0.valid, io.data1.valid).asBits
    (io.data0.ready, io.data1.ready) := ack
    io
  }
}

object Test extends App {
  val config = SpinalConfig(mode = Verilog)
    .generate(new Component {
      val bb = MyBlackBox()
      bb.io.data0.payload := 0
      bb.io.data1.payload := 1
      bb.io.data0.valid := True
      bb.io.data1.valid := False
    })
    .printPruned()
    .printPrunedIo()
    .printUnused()
}
```

nanoric, 2023-08-28:
Thank you so much for your pacient answer. you show me a method that is convient enough to do what I want and let me knowing that is was just a limit of myself makes me need such a "convenient auto connection". I will close this.

closed 2023-08-28 by nanoric