SpinalHDL/SpinalHDL #1185: Are there Some methods to get the specified type in SpinalHDL

xiaochuang-lxc, 2023-08-26:
@Dolu1990 
I want to do some specified type signal connection check in SpinalHDL. For example ,a defined Bundle A, how to get all A instances in design to do connection check by using LatencyAnalysis？
And how to get all ClockDomain used in a specified Component?

xiaochuang-lxc, 2023-08-26:
I test by using reflection but failed

andreasWallner, 2023-08-26:
It makes sense that reflection did not help you much - you'd only be able get information about the scala code and would have to infer HW information from there (which means you pretty much have to "execute" it). The information that you want is available in the SpinalHDL AST.

You can get all clocks used in a module using `spinal.lib.tools.ModuleAnalyzer.getClocks`.

For finding all instances of a specific Bundle you'll have to iterate over the statements in the internal netlist.  Have a look at the `ModuleAnalyzer`, maybe one of the functions there helps you, otherwise you'll have to crawl the AST yourself. 
This could look something like that (missing deduplication, it's also not complete, etc.)
```scala
object Test extends App {
  SpinalVerilog{
    val c = new Component{
      val io = new Bundle {
        val x = slave port Stream(Bits(5 bit))
      }
      io.x.ready := False
      val y = Flow(Bits(5 bit))
      y.payload := io.x.payload
      y.valid := io.x.valid
    }
    DumpAST(c)
    c.walkComponents{ c =>
      c.dslBody.foreachStatements { s => s match {
        case bt: BaseType if bt.parent.isInstanceOf[Stream[_]] => println("stream: ", bt.parent)
        case bt: BaseType if bt.parent.isInstanceOf[Flow[_]] => println("flow: ", bt.parent)
      }}
    }
    c
  }
}
```
If you go that route and want to look at the AST, I have a small utility that I'm still polishing a bit to push to lib that prints the AST, for now you can find it here: https://github.com/andreasWallner/spinalStuff/blob/7ba86e97255cfe39e23443eb32b8068e371ea370/src/main/scala/andreasWallner/Utils.scala#L100

andreasWallner, 2023-10-26:
@xiaochuang-lxc did my response help, can we close this?

xiaochuang-lxc, 2023-10-26:
Yes,I'm so sorry that forget this issue.Thanks !

closed 2023-10-26 by xiaochuang-lxc