SpinalHDL/SpinalHDL #1188: spinal.tester.scalatest.SdramXdrDdr2SpinalSim: Error "CLOCK CROSSING VIOLATION" detected in phase PhaseCheckCrossClock

LurenAA, 2023-08-29:
The path of this file is “tester/src/test/scala/spinal/tester/scalatest/SdramXdrSdrSpinalSim.scala”，When running `sbt "tester/test:runMain spinal.tester.scalatest.SdramXdrDdr2SpinalSim"`, the following error is reported. The same issue also arises in SdramXdrDdr3SpinalSim and SdramXdrSdrSpinalSim.

Thank you.

********************************************************************************

[error] Exception in thread "main" spinal.core.SpinalExit: 
[error]  Error detected in phase PhaseCheckCrossClock
[error] ********************************************************************************
[error] ********************************************************************************
[error] CLOCK CROSSING VIOLATION :
[error] - Source            : null spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply$mcV$sp(RtlPhy.scala:63)
[error] - Source clock      : (toplevel/phy/io_write_clk : in Bool)
[error] - Destination       : (toplevel/ctrl/core/backend/rspPipeline_debugData :  Bits[64 bits]) jdk.internal.reflect.GeneratedConstructorAccessor6.newInstance(Unknown Source)
[error] - Destination clock : (clk :  Bool)
[error] - Source declaration :
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply$mcV$sp(RtlPhy.scala:63)
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply(RtlPhy.scala:63)
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply(RtlPhy.scala:63)
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy.<init>(RtlPhy.scala:62)
[error]     spinal.tester.scalatest.SdramXdrCtrlPlusRtlPhy.<init>(SdramXdrSdrSpinalSim.scala:30)
[error]     spinal.tester.scalatest.SdramXdrTesterHelpers$$anonfun$complied$1.apply(SdramXdrSdrSpinalSim.scala:548)
[error]     spinal.tester.scalatest.SdramXdrTesterHelpers$$anonfun$complied$1.apply(SdramXdrSdrSpinalSim.scala:534)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] - Destination declaration :
[error]     jdk.internal.reflect.GeneratedConstructorAccessor6.newInstance(Unknown Source)
[error]     java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error]     java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:499)
[error]     java.base/java.lang.reflect.ReflectAccess.newInstance(ReflectAccess.java:128)
[error]     java.base/jdk.internal.reflect.ReflectionFactory.newInstance(ReflectionFactory.java:347)
[error]     java.base/java.lang.Class.newInstance(Class.java:645)
[error]     spinal.lib.memory.sdram.xdr.Backend$$anon$5.<init>(Backend.scala:142)
[error]     spinal.lib.memory.sdram.xdr.Backend.<init>(Backend.scala:111)
[error]     spinal.lib.memory.sdram.xdr.Core.<init>(Core.scala:37)
[error]     spinal.lib.memory.sdram.xdr.CtrlWithoutPhy.<init>(CtrlWithPhy.scala:78)
[error]     spinal.tester.scalatest.SdrXdrCtrlPlusPhy.<init>(SdramXdrSdrSpinalSim.scala:21)
[error]     spinal.tester.scalatest.SdramXdrCtrlPlusRtlPhy.<init>(SdramXdrSdrSpinalSim.scala:29)
[error]     spinal.tester.scalatest.SdramXdrTesterHelpers$$anonfun$complied$1.apply(SdramXdrSdrSpinalSim.scala:548)
[error]     spinal.tester.scalatest.SdramXdrTesterHelpers$$anonfun$complied$1.apply(SdramXdrSdrSpinalSim.scala:534)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] - Connection path :
[error]       >>> toplevel/phy/ram : Mem[16777216*64 bits] at spinal.lib.memory.sdram.xdr.phy.RtlPhy.<init>(RtlPhy.scala:60) >>>
[error]       >>> (toplevel/phy/readed :  Bits[64 bits]) at spinal.lib.memory.sdram.xdr.phy.RtlPhy.<init>(RtlPhy.scala:140) >>>
[error]       >>> (toplevel/phy/io_ctrl_phases_0_DQr_1 : out Bits[16 bits]) at spinal.lib.memory.sdram.xdr.SdramXdrPhyCtrlPhase$$anonfun$5.apply(Xdr.scala:100) >>>
[error]       >>> (toplevel/ctrl/io_phy_phases_0_DQr_1 : in Bits[16 bits]) at spinal.lib.memory.sdram.xdr.SdramXdrPhyCtrlPhase$$anonfun$5.apply(Xdr.scala:100) >>>
[error]       >>> (toplevel/ctrl/core/io_phy_phases_0_DQr_1 : in Bits[16 bits]) at spinal.lib.memory.sdram.xdr.SdramXdrPhyCtrlPhase$$anonfun$5.apply(Xdr.scala:100) >>>
[error]       >>> (toplevel/ctrl/core/backend/io_phy_phases_0_DQr_1 : in Bits[16 bits]) at spinal.lib.memory.sdram.xdr.SdramXdrPhyCtrlPhase$$anonfun$5.apply(Xdr.scala:100) >>>
[error]       >>> (toplevel/ctrl/core/backend/rspPipeline_output_payload_fragment_data :  Bits[64 bits]) at spinal.lib.memory.sdram.xdr.Backend$PipelineRsp.<init>(Backend.scala:106) >>>
[error]       >>> (toplevel/ctrl/core/backend/rspPipeline_debugData :  Bits[64 bits]) at jdk.internal.reflect.GeneratedConstructorAccessor6.newInstance(Unknown Source) >>>
[error]              
[error] ********************************************************************************
[error] ********************************************************************************
[error] CLOCK CROSSING VIOLATION :
[error] - Source            : null spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply$mcV$sp(RtlPhy.scala:63)
[error] - Source clock      : (toplevel/phy/io_write_clk : in Bool)
[error] - Destination       : (toplevel/ctrl/core/backend/rspPop_payload_fragment_data :  Bits[64 bits]) spinal.lib.memory.sdram.xdr.Backend$PipelineRsp.<init>(Backend.scala:106)
[error] - Destination clock : (clk :  Bool)
[error] - Source declaration :
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply$mcV$sp(RtlPhy.scala:63)
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply(RtlPhy.scala:63)
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy$$anonfun$1.apply(RtlPhy.scala:63)
[error]     spinal.lib.memory.sdram.xdr.phy.RtlPhy.<init>(RtlPhy.scala:62)
[error]     spinal.tester.scalatest.SdramXdrCtrlPlusRtlPhy.<init>(SdramXdrSdrSpinalSim.scala:30)
[error]     spinal.tester.scalatest.SdramXdrTesterHelpers$$anonfun$complied$1.apply(SdramXdrSdrSpinalSim.scala:548)
[error]     spinal.tester.scalatest.SdramXdrTesterHelpers$$anonfun$complied$1.apply(SdramXdrSdrSpinalSim.scala:534)
[error]     spinal.sim.JvmThread.run(SimManager.scala:51)
[error] - Destination declaration :
[error]     spinal.lib.memory.sdram.xdr.Backend$PipelineRsp.<init>(Backend.scala:106)
[error]     spinal.lib.memory.sdram.xdr.Backend$$anon$5$$anonfun$36$$anonfun$apply$6.apply(Backend.scala:131)
[error]     spinal.lib.memory.sdram.xdr.Backend$$anon$5$$anonfun$36$$anonfun$apply$6.apply(Backend.scala:131)
[... 39 lines cut ...]
[error]         at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2916)
[error]         at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2852)
[error]         at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:71)
[error]         at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2852)
[error]         at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2847)
[error]         at spinal.core.Spinal$.apply(Spinal.scala:412)
[error]         at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:178)
[error]         at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:927)
[error]         at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:897)
[error]         at spinal.tester.scalatest.SdramXdrTesterHelpers$.complied(SdramXdrSdrSpinalSim.scala:534)
[error]         at spinal.tester.scalatest.SdramXdrDdr2SpinalSim$.delayedEndpoint$spinal$tester$scalatest$SdramXdrDdr2SpinalSim$1(SdramXdrSdrSpinalSim.scala:623)
[error]         at spinal.tester.scalatest.SdramXdrDdr2SpinalSim$delayedInit$body.apply(SdramXdrSdrSpinalSim.scala:601)
[error]         at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.App$$anonfun$main$1.apply(App.scala:76)
[error]         at scala.collection.immutable.List.foreach(List.scala:392)
[error]         at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error]         at scala.App$class.main(App.scala:76)
[error]         at spinal.tester.scalatest.SdramXdrDdr2SpinalSim$.main(SdramXdrSdrSpinalSim.scala:601)
[error]         at spinal.tester.scalatest.SdramXdrDdr2SpinalSim.main(SdramXdrSdrSpinalSim.scala)
[error] Nonzero exit code returned from runner: 1
[error] (tester / Test / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 3 s, completed Aug 29, 2023, 3:00:25 AM

Readon, 2023-08-29:
I traced it down, and find that in [RtlPhy](https://github.com/SpinalHDL/SpinalHDL/blob/eac49b950eb8a8da49a49a587361f958ecf8eb66/lib/src/main/scala/spinal/lib/memory/sdram/xdr/phy/RtlPhy.scala#L62) the cmd is written at io.write.clk domain.
However, while read from the ram by readAsync function the [readed](https://github.com/SpinalHDL/SpinalHDL/blob/eac49b950eb8a8da49a49a587361f958ecf8eb66/lib/src/main/scala/spinal/lib/memory/sdram/xdr/phy/RtlPhy.scala#L142) signal is still in io.write.clk domain but clk although the code is using current clock domain(clk).

Readon, 2023-08-29:
As I understand in Connection path the signals should be in `clk` domain.

It might also caused by ram's clock domain which has been set as io.write.clk, so that output signal from it has been set to the same clock. @Dolu1990

Readon, 2023-08-29:
Also if I comment out the ram.write block at line 62 in RtlPhy.scala, there would be no error prompted while elaboration.

Dolu1990, 2023-08-30:
This error appear since i improved the cross clock domain detection across memory ports.
Fixed now ^^

commit 2a2b4346 (2023-08-30): fix #1188 RtlPhy crossclock
closed 2023-08-30 by Dolu1990