SpinalHDL/SpinalHDL #1193: facility to enhance pcie dma.

chenbo-again, 2023-09-04:
Pcie is useful in various domain in fpga. However, there is not any facility to use pcie dma in SpinalHDL.

I'm currently working on it.  Firstly, I will focus on ultrascale device, I will mainly borrow from riffa and corundum.

riffa: https://github.com/KastnerRG/riffa

corundum: https://github.com/corundum/corundum

likewise, 2023-09-04:
Can you explain what this issue is about? Most FPGAs only have the lowest three PCIe layers (up to including the TLP transaction layer) as a hard-IP block, everything else is soft-IP, including the MMIO bridge and SGDMA (such as AMD PG195).

Would you like to implement a full PCIe BAR MMIO bridge and SGDMA solution, or interface with an existing one?

chenbo-again, 2023-09-04:
Firstly, I will focus on sgDma and MMIO based on transaction layer(in detail, it's PG195 UltraScale+ Devices Integrated Block for PCI Express)

likewise, 2023-09-04:
Are you interfacing against PG195 or replacing PG195 with a SpinalHDL implementation?

I am asking because I have co-developed that block. I could help on both approaches.

chenbo-again, 2023-09-04:
Yes, currently, I am working for this.

likewise, 2023-09-04:
Nice work.

My question is:  Are you...
1) ...replacing PG195?
XOR
2) ...interfacing with PG195?

Thanks

chenbo-again, 2023-09-04:
I will develop a component to replace it.

Dolu1990, 2023-09-04:
Hi ^^
What memory bus are you targeting ?

chenbo-again, 2023-09-15:
sorry for my late.

> Hi ^^ What memory bus are you targeting ?

Currently, they are axis and axi,  thank you for updating dmasg, I will try to use it as sg engine.

Dolu1990, 2023-09-18:
dmasg is very very generalist. When i designed it had the hope that it would be usefull for peripherals, but overall, now, i have the feeling that in most case, people will just be better off implementing dedicated DMA in their peripheral to cover their exact needs.

One extrem example of that is the USB OHCI :)

I mean, hard to tell for the PCIE usecase. i don't know it enough.

chenbo-again, 2023-09-29:
> dmasg is very very generalist. When i designed it had the hope that it would be usefull for peripherals, but overall, now, i have the feeling that in most case, people will just be better off [...]

I think it's a good way to implement pcie dma with sgDma, It has a lot in common with [riffa](https://github.com/KastnerRG/riffa), I will go deeper in riffa to find a appropriate way to implement it.

do  you have any other idea, happy to listen. :D

leon-brightai, 2023-09-29:
Look at the Corundum project on GitHub. It has PCIe SGDMA for Intel and Xilinx.

But from RIFFA you can also learn a lot.

Do not underestimate the amount of details though. I have helped writing a PCIe SGDMA 13 years ago, that was a multi-men year effort including the driver support for all the operating systems.

chenbo-again, 2023-09-30:
> Look at the Corundum project on GitHub. It has PCIe SGDMA for Intel and Xilinx.
>
> But from RIFFA you can also learn a lot.
>
> Do not underestimate the amount of details though. I have helped writing a [...]

hello
agree with you, however, currently it aims to provide a easy way to create a demo level pcie dma, and I will keep track in this issue until it really works.

liubenyuan, 2023-11-12:
Regarding #1213 Hi, I think there should be some document on it or test on it using an Altera/Xilinx development board. @likewise @chenbo-again

I have a vcu118 and adm-pcie-9v3, anything I can do to test it? Is there a detailed procedure in generating a PCIe test bitstream?

chenbo-again, 2023-11-25:
> Regarding #1213 Hi, I think there should be some document on it or test on it using an Altera/Xilinx development board. @likewise @chenbo-again
> 
> I have a vcu118 and adm-pcie-9v3, anything I can do to test it? Is there a detailed procedure in generating a PCIe test bitstream?

For now, there is not a specific board support, but I recently got a kcu116, I'm working for supporting it currently. and when it is done, more board supporting is considered.

andreasWallner, 2023-11-28:
Just as a sidenote: I currently don't know much about PCIe, but I got a nitefury recently - so I can offer to test on an Artix if needed/helpful (I might need a little bit of help though ;-) )

PR #1213 (2023-10-14): Pcie

Closes #1193

# Context, Motivation & Description
There is no pcie facilities in spinalhdl.
I create pcie interface(the same as verilog-pcie, it is named [FPGA-independent PCIe](https://github.com/alexforencich/verilog-pcie#fpga-independent-pcie)) and ultrascale -> pcie interface bridge, then I use pcie interface to enhance dmasg, but it is just working in progress.

# Impact on code generation
no, because it is a lib component

# Checklist

- [ ] Unit tests were added
partially, some of them are already tested with test utilities in verilog-pcie.(because of license, I will not add them) [... cut]