SpinalHDL/SpinalHDL #1194: SpinalHDL wth Vivado Issue: Generation of vivdao script failed

qianyich, 2023-09-05:
Hi Community,

I am trying to get Vivdao up and running with SpinalHDL on my machine. I took the SpinalTemplateSbt project and put the Vivado path into my environment path, but I am having some errors when doing `runMain projectname.MyTopLevelSim`

error log as follows

[info] welcome to sbt 1.6.0 (Private Build Java 1.8.0_362)
[info] loading settings for project spinaltemplatesbt-build from plugins.sbt ...
[info] loading project definition from /home/qianyich/SpinalHDL/SpinalTemplateSbt/project
[info] loading settings for project projectname from build.sbt ...
[info] set current project to projectname (in build file:/home/qianyich/SpinalHDL/SpinalTemplateSbt/)
[info]
[info] Here are some highlights of this release:
[info]   - Improved JDK 17 support
[info]   - Improved Build Server Protocol (BSP) support
[info]   - Tab completion of global keys
[info] See https://eed3si9n.com/sbt-1.6.0 for full release notes.
[info] Hide the banner for this release by running `skipBanner`.
[info] sbt server started at local:///home/qianyich/.sbt/1.0/server/7daad2e576856e6131f6/sock
[info] started sbt server
sbt:projectname> runMain projectname.MyTopLevelSim
[info] compiling 1 Scala source to /home/qianyich/SpinalHDL/SpinalTemplateSbt/target/scala-2.12/classes ...
[info] running (fork) projectname.MyTopLevelSim
[info] [Runtime] SpinalHDL v1.9.3    git head : 029104c77a54c53f1edda327a3bea333f7d65fd9
[info] [Runtime] JVM max memory : 21156.0MiB
[info] [Runtime] Current date : 2023.09.04 20:28:39
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 0.249 : Checks and transforms
[info] [Progress] at 0.347 : Generate Verilog
[info] [Done] at 0.432
[info] [Progress] Simulation workspace in /home/qianyich/SpinalHDL/SpinalTemplateSbt/./simWorkspace/MyTopLevel
[info] [Progress] XSIM compilation started
[info] ****** Vivado v2021.2 (64-bit)
[info]   **** SW Build 3367213 on Tue Oct 19 02:47:39 MDT 2021
[info]   **** IP Build 3369179 on Thu Oct 21 08:25:16 MDT 2021
[error] Exception in thread "main" java.lang.Exception: Generation of vivado script failed
[info]     ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.
[error]         at spinal.sim.XSimBackend.doCmd(XSimBackend.scala:104)
[error]         at spinal.sim.XSimBackend.generateScript(XSimBackend.scala:220)
[error]         at spinal.sim.XSimBackend.<init>(XSimBackend.scala:333)
[error]         at spinal.core.sim.SpinalXSimBackend$Backend.<init>(SimBootstraps.scala:446)
[info] source /home/qianyich/SpinalHDL/SpinalTemplateSbt/./simWorkspace/MyTopLevel/xsim/spinal_xsim.tcl
[error]         at spinal.core.sim.SpinalXSimBackend$.apply(SimBootstraps.scala:488)
[info] # proc findFiles { basedir pattern } {
[error]         at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:1113)
[info] #
[error]         at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:934)
[info] #    # Fix the directory name, this ensures the directory name is in the
[error]         at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:897)
[info] #    # native format for the platform and contains a final directory seperator
[error]         at projectname.MyTopLevelSim$.delayedEndpoint$projectname$MyTopLevelSim$1(MyTopLevelSim.scala:8)
[info] #    set basedir [string trimright [file join [file normalize $basedir] { }]]
[error]         at projectname.MyTopLevelSim$delayedInit$body.apply(MyTopLevelSim.scala:6)
[info] #    set fileList {}
[error]         at scala.Function0.apply$mcV$sp(Function0.scala:39)
[info] #
[error]         at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[info] #    # Look in the current directory for matching files, -type {f r}
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[info] #    # means ony readable normal files are looked at, -nocomplain stops
[error]         at scala.App.$anonfun$main$1$adapted(App.scala:80)
[info] #    # an error being thrown if the returned list is empty
[error]         at scala.collection.immutable.List.foreach(List.scala:431)
[info] #    foreach fileName [glob -nocomplain -type {f r} -path $basedir $pattern] {
[error]         at scala.App.main(App.scala:80)
[info] #        lappend fileList $fileName
[error]         at scala.App.main$(App.scala:78)
[info] #    }
[error]         at projectname.MyTopLevelSim$.main(MyTopLevelSim.scala:6)
[info] #
[error]         at projectname.MyTopLevelSim.main(MyTopLevelSim.scala)
[info] #    # Now look for any sub direcories in the current directory
[info] #    foreach dirName [glob -nocomplain -type {d  r} -path $basedir *] {
[info] #        # Recusively call the routine on the sub directory and append any
[info] #        # new files to the results
[info] #        set subDirList [findFiles $dirName $pattern]
[info] #        if { [llength $subDirList] > 0 } {
[info] #            foreach subDirFile $subDirList {
[info] #                lappend fileList $subDirFile
[info] #            }
[info] #        }
[info] #    }
[info] #    return $fileList
[info] # }
[info] # create_project -force MyTopLevel_xsim -part xc7vx485tffg1157-1
[info] WARNING: [Device 21-436] No parts matched 'xc7vx485tffg1157-1'
[info] ERROR: [Coretcl 2-106] Specified part could not be found.
[info] INFO: [Common 17-206] Exiting Vivado at Mon Sep  4 20:28:46 2023...
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 12 s, completed Sep 4, 2023 8:28:46 PM
sbt:projectname>

sim code as follows:
```
package projectname

import spinal.core._
import spinal.core.sim._

object MyTopLevelSim extends App {

  SimConfig.withXSim.withWave.compile(new MyTopLevel).doSim { dut =>
    // Fork a process to generate the reset and the clock on the dut
    dut.clockDomain.forkStimulus(period = 10)

    var modelState = 0
    for (idx <- 0 to 99) {
      // Drive the dut inputs with random values
      dut.io.cond0.randomize()
      dut.io.cond1.randomize()

      // Wait a rising edge on the clock
      dut.clockDomain.waitRisingEdge()

      // Check that the dut values match with the reference model ones
      val modelFlag = modelState == 0 || dut.io.cond1.toBoolean
      assert(dut.io.state.toInt == modelState)
      assert(dut.io.flag.toBoolean == modelFlag)

      // Update the reference model value
      if (dut.io.cond0.toBoolean) {
        modelState = (modelState + 1) & 0xff
      }
    }
  }
}
```

I am choosing this template project because it does not involve any xilinx ips.

Could anyone point out where is the problem?

Thanks!

kleinai, 2023-09-05:
Could it be this? Do you have Virtex 7 support installed?
>[info] WARNING: [Device 21-436] No parts matched 'xc7vx485tffg1157-1'
[info] ERROR: [Coretcl 2-106] Specified part could not be found.

You can use `withXilinxDevice` to set your specific part.

qianyich, 2023-09-05:
[image]

Virtex 7 board files have been installed as shown in the pic. I do not have the board connected , I do not think this is an issue though.

qianyich, 2023-09-05:
> You can use `withXilinxDevice` to set your specific part.
This fixes the issue. Thank you so much!

closed 2023-09-05 by kleinai