SpinalHDL/SpinalHDL #1195: Why does the switch generate verilog code for if else?

13230668653, 2023-09-05:
io.data := SpinalMap(io.funct3, B"3'b0" -> B"2'b1",
        B"3'b1" -> B"2'b10",
        B"3'b10" -> B"2'b11",
        B"3'b11" -> B"2'b00",
        default -> B"2'b00")

    val table:Seq[(Any, Bits)] = Seq(B"3'b0" -> B"2'b1", B"3'b1" -> B"2'b10", B"3'b10" -> B"2'b11", B"3'b11" -> B"2'b00", spinal.core.default -> B"2'b00")
    io.data := SpinalMap.list(io.funct3, table)

These two spinalHDL codes generate different verilog codes。
first verilog code

```
  always @(*) begin
    case(io_funct3)
      3'b000 : begin
        _zz_io_data = 2'b01;
      end
      3'b001 : begin
        _zz_io_data = 2'b10;
      end
      3'b010 : begin
        _zz_io_data = 2'b11;
      end
      3'b011 : begin
        _zz_io_data = 2'b00;
      end
      default : begin
        _zz_io_data = 2'b00;
      end
    endcase
  end

  assign io_data = _zz_io_data;
```

second verilog code

```
  always @(*) begin
    if((io_funct3 == table_0_0)) begin
        _zz_io_data = table_0_1;
    end else if((io_funct3 == table_1_0)) begin
        _zz_io_data = table_1_1;
    end else if((io_funct3 == table_2_0)) begin
        _zz_io_data = table_2_1;
    end else if((io_funct3 == table_3_0)) begin
        _zz_io_data = table_3_1;
    end else begin
        _zz_io_data = table_4_1;
    end
  end

  assign io_data = _zz_io_data;
```

Then I tried to implement this function myself using switches in the code, but found that. How to use the first method to pass parameters without using a set type to generate a verilog is a case statement. How can I use set types to generate verilogs that are nested if else. I hope to generate case statements using collection types for easy extension. How should I use it?

Dolu1990, 2023-09-12:
Hi,

The reason why, is that SpinalHDL is trying to preserve stuff.

```
    val table: Seq[(Any, Bits)] = Seq(B"3'b0" -> B"2'b1", B"3'b1" -> B"2'b10", B"3'b10" -> B"2'b11", B"3'b11" -> B"2'b00", default -> B"2'b00")
```

Create a list of named hardware signals

``` 
  assign table_0_0 = 3'b000;
  assign table_0_1 = 2'b01;
  assign table_1_0 = 3'b001;
  assign table_1_1 = 2'b10;
  assign table_2_0 = 3'b010;
  assign table_2_1 = 2'b11;
  assign table_3_0 = 3'b011;
  assign table_3_1 = 2'b00;
  assign table_4_1 = 2'b00;
```

Then SpinalHDL do a switch case on them as key, but thing is that verilog doesn't support using signals in cases, so SpinalHDL fallback to if / else

Way to avoid that is to ensure that the table doesn't generate named signal, so SpinalHDL is alowed to symplify them to their raw literal : 

```
@dontName val table: Seq[(Any, Bits)] = Seq(B"3'b0" -> B"2'b1", B"3'b1" -> B"2'b10", B"3'b10" -> B"2'b11", B"3'b11" -> B"2'b00", default -> B"2'b00")
or 
def table: Seq[(Any, Bits)] = Seq(B"3'b0" -> B"2'b1", B"3'b1" -> B"2'b10", B"3'b10" -> B"2'b11", B"3'b11" -> B"2'b00", default -> B"2'b00")
```

13230668653, 2023-09-12:
Thank you for solving my problem perfectly.

andreasWallner, 2023-09-19:
Assuming from you last comment that this is solved, if not please reopen.

closed 2023-09-19 by andreasWallner