SpinalHDL/SpinalHDL #1198: when I use withXSim there happend A fatal error in java runtime

izumkineno, 2023-09-07:
outlog:

>  "E:\Program\IntelliJ IDEA 2023.2\java_jb\bin\java.exe" "-javaagent:E:\Program\IntelliJ IDEA 2023.2\lib\idea_rt.jar=8529:E:\Program\IntelliJ IDEA 2023.2\bin" -Dfile.encoding=UTF-8 -classpath E:\code\SpinalHDL\SpinalTemplateSbt_\target\scala-2.12\classes;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\oshi\oshi-core\6.4.0\oshi-core-6.4.0.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\scopt\scopt_2.12\4.1.0\scopt_2.12-4.1.0.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-core_2.12\1.9.3\spinalhdl-core_2.12-1.9.3.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-idsl-payload_2.12\1.9.3\spinalhdl-idsl-payload_2.12-1.9.3.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-idsl-plugin_2.12\1.9.3\spinalhdl-idsl-plugin_2.12-1.9.3.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-lib_2.12\1.9.3\spinalhdl-lib_2.12-1.9.3.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\github\spinalhdl\spinalhdl-sim_2.12\1.9.3\spinalhdl-sim_2.12-1.9.3.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\com\lihaoyi\sourcecode_2.12\0.3.0\sourcecode_2.12-0.3.0.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\commons-io\commons-io\2.11.0\commons-io-2.11.0.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\net\java\dev\jna\jna-platform\5.12.1\jna-platform-5.12.1.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\net\java\dev\jna\jna\5.12.1\jna-5.12.1.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\net\openhft\affinity\3.23.2\affinity-3.23.2.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\modules\scala-xml_2.12\1.0.6\scala-xml_2.12-1.0.6.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\scala-compiler\2.12.16\scala-compiler-2.12.16.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\scala-library\2.12.16\scala-library-2.12.16.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scala-lang\scala-reflect\2.12.16\scala-reflect-2.12.16.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\scalactic\scalactic_2.12\3.2.10\scalactic_2.12-3.2.10.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\slf4j\slf4j-api\2.0.5\slf4j-api-2.0.5.jar;C:\Users\CEO\AppData\Local\Coursier\cache\v1\https\repo1.maven.org\maven2\org\slf4j\slf4j-simple\2.0.5\slf4j-simple-2.0.5.jar projectname.MyTopLevelSim
> 
[hs_err_pid8976.log](https://github.com/SpinalHDL/SpinalHDL/files/12545503/hs_err_pid8976.log)
[Runtime] SpinalHDL v1.9.3    git head : 029104c77a54c53f1edda327a3bea333f7d65fd9
> [Runtime] JVM max memory : 4050.0MiB
> [Runtime] Current date : 2023.09.07 14:22:22
> [Progress] at 0.000 : Elaborate components
> [Progress] at 0.339 : Checks and transforms
> [Progress] at 0.412 : Generate Verilog
> [Done] at 0.513
> [Progress] Simulation workspace in E:\code\SpinalHDL\SpinalTemplateSbt_\.\simWorkspace\MyTopLevel
> [Progress] XSIM compilation started
> \#
> \# A fatal error has been detected by the Java Runtime Environment:
> \#
> \#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x00007ff8a47a6326, pid=8976, tid=30524
> \#
> \# JRE version: OpenJDK Runtime Environment JBR-17.0.8+7-1000.8-jcef (17.0.8+7) (build 17.0.8+7-b1000.8)
> \# Java VM: OpenJDK 64-Bit Server VM JBR-17.0.8+7-1000.8-jcef (17.0.8+7-b1000.8, mixed mode, tiered, compressed oops, compressed class ptrs, g1 gc, windows-amd64)
> \# Problematic frame:
> \# C  [libstdc++-6.dll+0x86326]
> \#
> \# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
> \#
> JNI global refs:
> JNI global refs: 18, weak refs: 45
> 
> JNI global refs memory usage: 843, weak refs: 841
> 
> OOME stack traces (most recent first):
> Classloader memory used:
> Loader jdk.internal.loader.ClassLoaders$AppClassLoader                                 : 1699K
> Loader bootstrap                                                                       : 1691K
> Loader jdk.internal.reflect.DelegatingClassLoader                                      : 4636B
> Loader jdk.internal.loader.ClassLoaders$PlatformClassLoader                            : 3713B
> 
> \# An error report file with more information is saved as:
> \# E:\code\SpinalHDL\SpinalTemplateSbt_\hs_err_pid8976.log
> \#
> \# If you would like to submit a bug report, please visit:
> \#   https://youtrack.jetbrains.com/issues/JBR
> \# The crash happened outside the Java Virtual Machine in native code.
> \# See problematic frame for where to report the bug.
> \#
> 

my code:

```
package projectname

import spinal.core._
import spinal.core.sim._

object MyTopLevelSim extends App {
  sim.SimConfig.withXilinxDevice("xa7s6cpga196-2I").withXSim.withWave.compile(MyTopLevel()).doSim { dut =>
    // Fork a process to generate the reset and the clock on the dut
    dut.clockDomain.forkStimulus(period = 10)

    var modelState = 0
    for (idx <- 0 to 99) {
      // Drive the dut inputs with random values
      dut.io.cond0.randomize()
      dut.io.cond1.randomize()

      // Wait a rising edge on the clock
      dut.clockDomain.waitRisingEdge()

      // Check that the dut values match with the reference model ones
      val modelFlag = modelState == 0 || dut.io.cond1.toBoolean
      assert(dut.io.state.toInt == modelState)
      assert(dut.io.flag.toBoolean == modelFlag)

      // Update the reference model value
      if (dut.io.cond0.toBoolean) {
        modelState = (modelState + 1) & 0xff
      }
    }
  }
}
```

it is runing on IDEA windows 10, this is error log

[hs_err_pid8976.log](https://github.com/SpinalHDL/SpinalHDL/files/12545510/hs_err_pid8976.log)

izumkineno, 2023-09-07:
vivado version is 2023.1

Dolu1990, 2023-09-19:
i didn't used the xsim so far, but the issue is possibly related to versions

Could you try with SpinalHDL 1.9.1 1.9.2  and maybe a "good" old Artix 7 FPGA device ?