SpinalHDL/SpinalHDL #1199: `slave()` seem's doesn't work in version 1.9.3

hy0000, 2023-09-07:
I try the 1.9.3 of spinal, then meet the `HIERARCHY VIOLATION` error when writing my `Bundle`  with master and slave.
It seems that the `slave()` method have not effect on my `Bundle`, The direction of io has not changed after calling `asSlave()`
Here is the code to reproduce：
```
case class P() extends Bundle with IMasterSlave {
  val a = Bool()
  override def asMaster(): Unit = {
    in(a)
  }
}

case class MyTopLevel() extends Component {
  val p = slave(P())
  p.a := True
}
```
When I change version to 1.9.0, the code runs correctly

Dolu1990, 2023-09-07:
Hi,

What is the error reported in the terminal ? (copy past)

It maybe due to some SBT issue which doesn't load the compiler plugin.
Did you tried a sbt clean compile ?

hy0000, 2023-09-07:
```
[Runtime] SpinalHDL v1.9.3    git head : 029104c77a54c53f1edda327a3bea333f7d65fd9
[Runtime] JVM max memory : 1744.0MiB
[Runtime] Current date : 2023.09.07 18:40:41
[Progress] at 0.000 : Elaborate components
[Progress] at 1.276 : Checks and transforms

**********************************************************************************************
[Warning] Elaboration failed (2 errors).
          Spinal will restart with scala trace to help you to find the problem.
**********************************************************************************************

[Progress] at 1.360 : Elaborate components
[Progress] at 1.377 : Checks and transforms
Exception in thread "main" spinal.core.SpinalExit: 

 Error detected in phase PhaseCheckHiearchy
********************************************************************************
********************************************************************************

HIERARCHY VIOLATION : (toplevel/??? : in Bool) is driven by Bool(x), but isn't accessible in the toplevel component.
    projectname.MyTopLevel.<init>(MyTopLevel.scala:15)
    projectname.MyTopLevelVerilog$.$anonfun$new$1(MyTopLevel.scala:19)
    spinal.sim.JvmThread.run(SimManager.scala:51)

********************************************************************************
********************************************************************************

Design's errors are listed above.
SpinalHDL compiler exit stack : 

	at spinal.core.SpinalExit$.apply(Misc.scala:441)
	at spinal.core.SpinalError$.apply(Misc.scala:496)
	at spinal.core.internals.PhaseContext.checkPendingErrors(Phase.scala:177)
	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:193)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$12(Phase.scala:2976)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$12$adapted(Phase.scala:2974)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$7(Phase.scala:2974)
	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:71)
	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2910)
	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2905)
	at spinal.core.Spinal$.apply(Spinal.scala:412)
	at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:178)
	at projectname.MyTopLevelVerilog$.delayedEndpoint$projectname$MyTopLevelVerilog$1(MyTopLevel.scala:19)
	at projectname.MyTopLevelVerilog$delayedInit$body.apply(MyTopLevel.scala:18)
	at scala.Function0.apply$mcV$sp(Function0.scala:39)
	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
	at scala.App.$anonfun$main$1$adapted(App.scala:80)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at scala.App.main(App.scala:80)
	at scala.App.main$(App.scala:78)
	at projectname.MyTopLevelVerilog$.main(MyTopLevel.scala:18)
	at projectname.MyTopLevelVerilog.main(MyTopLevel.scala)

Process finished with exit code 1
```

hy0000, 2023-09-07:
when i recompile, there is an error in my sbtshell:
```
[IJ]clean
[success] Total time: 0 s, completed Sep 7, 2023 6:45:08 PM
[IJ]compile
[info] compiling 2 Scala sources to /home/hyl/Code/MySpinalTemplete-master/target/scala-2.12/classes ...
[info] Error: unable to load class: spinal.idslplugin.IdslPlugin
[info] Ignoring duplicate plugin idsl-plugin (spinal.idslplugin.IdslPlugin)
[info] done compiling
[success] Total time: 10 s, completed Sep 7, 2023 6:45:22 PM
```

Dolu1990, 2023-09-07:
Can you share your build.sbt ?

hy0000, 2023-09-07:
sure, I have pushed all the codes at https://github.com/hy0000/MySpinalTemplete

Dolu1990, 2023-09-07:
That's funny, the thing which make it fail is the .map(_.withSources()) on spinalIdslPlugin XD

Here is a workaround : 

```scala
ThisBuild / version := "1.0"
ThisBuild / scalaVersion := "2.12.16"
ThisBuild / organization := "org.example"

val spinalVersion = "1.9.3"
val spinalCore = "com.github.spinalhdl" %% "spinalhdl-core" % spinalVersion
val spinalLib = "com.github.spinalhdl" %% "spinalhdl-lib" % spinalVersion
val spinalIdslPlugin = compilerPlugin("com.github.spinalhdl" %% "spinalhdl-idsl-plugin" % spinalVersion)
val scalactic = "org.scalactic" %% "scalactic" % "3.2.16"
val scalatest = "org.scalatest" %% "scalatest" % "3.2.16" % "test"

lazy val projectname = (project in file("."))
  .settings(
    Compile / scalaSource := baseDirectory.value / "hw" / "spinal",
    libraryDependencies ++= Seq(spinalIdslPlugin),
    libraryDependencies ++= Seq(spinalCore, spinalLib, scalactic, scalatest).map(_.withSources())
  )

fork := true
```

Absolutly no idea why it happens XD
Maybve withSources isn't allowed on scala compiler plugins.

hy0000, 2023-09-07:
Great, I didn’t expect this to be the problem here
thank you very much!

closed 2023-09-07 by hy0000