SpinalHDL/SpinalHDL #1200: Exception on Component Init

jjjt-git, 2023-09-08:
When building my component, I get following Exception (after runMain):
`Exception in Thread "main" java.lang.NullPointerException: cannot invoke "spinal.core.GlobalData.toplevel()" because the return value of "spinal.core.Component.globalData()" is null`
This happens in Component.scala:72
In my custom files, the BT always points to the first line of the class definition (where the `class` and `extends` keywords are)

It might be, that I did not correctly use something (I am very new to Spinal)

Generation class:
```scala
object GenCache extends App {
	val config = SpinalConfig(
		targetDirectory = "gen",
		defaultConfigForClockDomains = ClockDomainConfig(
			resetActiveLevel = HIGH
		)
	)

	val dev = new TLM(TLMConfig(
                ...
		new Mod(ModConfig(...)),
                ...
	))

	config.generateVerilog(dev)
}
```

The module is a simple class structured like this:
```scala
case class ModIO(...) extends Bundle {...}
class Mod(...) {
	def getIO() : ModIO = ModIO(...)

	def build(io : ModIO) : Area = new Area {
		...
	}
}
```

TLM extends Component and instaniates the logic like this:
```scala
val mod = new Component {
	val io = modGen.getIO()
	val logic = modGen.build(io)
}
```
TLM does not have a io-Bundle but a single bus, where it receives commands defined like this:
`val bus = slave(BusObject())`
BusObject is quite similar to the simple DBusSimpleBus from VexRiscv

Thanks for helping (and reading this wall), if I missed any information, don't hesitate to ask.

Dolu1990, 2023-09-08:
Hi,

Component / hardware should never be instanciated outside the generateVerilog body

So in your case, def def instead of val dev would work i think

jjjt-git, 2023-09-09:
Thanks for the fast response. now it works.

closed 2023-09-09 by jjjt-git