SpinalHDL/SpinalHDL #1203: simulation RTL path from tmp

raochl, 2023-09-18:
Hi, 

I want to know why the simulation read in the rtl generated from the `tmp` directory not from the rtl directory of the simulation project directory as follows:

[image]

When I change a code from module A and simulate it, It will generate a new rtl in the `tmp` directory and cover the previous rtl files. 
If I want to check the difference between modified and before modified from the waveform, since the second simulation regenerates rtl and replaces the previous one, I cannot read the correct rtl file of the first waveform.

andreasWallner, 2023-10-26:
@raochl Did @Dolu1990's reponse help? If so, can we close this?

raochl, 2023-10-28:
Yes, there is an option to change the simulation workspace. But it is still related to the same Verilog in tmp directory. If I want to rebuild/simulate the same module with a different test case or different testbench. The waveform may change and be replaced with the newer waveform. This makes us unable to check the difference between the waveforms (related to the different test cases).

closed 2023-10-28 by raochl