SpinalHDL/SpinalHDL #1204: PackedBundle BUG?

dreamflyings, 2023-09-18:
@Dolu1990 Hello~   
Is there a problem with the following code line logic?
[bug line](https://github.com/SpinalHDL/SpinalHDL/blob/2fdc3aadfc859bbfcf0fb559658606f3dbe4f705/lib/src/main/scala/spinal/lib/PackedBundle.scala#L60) 
Should this line lastPos = r.high be changed to lastPos = r.high + 1?

Dolu1990, 2023-09-18:
Hi,

Looking at "downto (lastPos)"  =>  yes, likely, but i don't know much about PackedBundle at all XD

andreasWallner, 2023-10-26:
@dokleina could you have a look at this?

dokleina, 2023-11-01:
Hey all,

I agree that this is broken. The `lastPos` probably should have been named `nextPos` instead with a `+1` to the value.

I've got a branch with a potential fix, but it'll be a bit until I get complete test coverage.

PR #1271 (2024-01-12): PackedBundle Bug Fix

Closes #1204

# Context, Motivation & Description
Fixed PackedBundle bit overlap when using multiple `Data`s without using an explicit specifier (e.g., `packTo` or `packFrom`).

Other changes: 

- Cleaned up unused debug statements from earlier commits.
- Removed `withWave` from sims.

# Impact on code generation
None

# Checklist

- [x] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`?
  - on [RTD](https://github.com/SpinalHDL/SpinalDoc-RTD)?
  - thanks to a [new tracking issue on RTD](https://github.com/SpinalHDL/S [... cut]
closed 2024-01-12 by Dolu1990