SpinalHDL/SpinalHDL #1206: Something wrong with "Reg"

qjl-ukp, 2023-09-21:
Just beginning to learn Spinal/Scala , so this could be very stupid. 
a line of 
`val register = Reg(UInt(width bits)) init(0)`
reported error:  can not resolve overloaded method 'Reg'

I have 
`import spinal.core._`
at the beginning.
Scala ver. 2.12.15,  Spinal ver. 1.9.3

likewise, 2023-09-21:
The "issues" section on GitHub is not intended to ask questions when learning SpinalHDL. It is really to submit bug and feature requests.

Please try asking at one of the community channels, and post a (link to a) minimal failing example. https://matrix.to/#/#SpinalHDL_SpinalHDL:gitter.im

There are a lot of examples to start learning SpinalHDL with.

Dolu1990, 2023-09-21:
I'm think, maybe the github discutions channels could also be a way ? 
https://github.com/SpinalHDL/SpinalHDL/discussions