SpinalHDL/SpinalHDL #1209: Simulation fails when using Verilator of new version which requires C++14 standard

ZimingYuan, 2023-09-26:
When using Verilator of new version (such as 5.016), simulation fails and complains something like

```
[info] In file included from /usr/local/share/verilator/include/verilated.h:937,
[info]                  from VLinear___024root__DepSet_h572e5853__3.cpp:5:
[info] /usr/local/share/verilator/include/verilated_types.h: In function 'constexpr IData VL_CLOG2_CE_Q(QData)':
[info] /usr/local/share/verilator/include/verilated_types.h:85:1: error: body of 'constexpr' function 'constexpr IData VL_CLOG2_CE_Q(QData)' not a return-statement
[info]    85 | }
[info]       | ^
```

It seems that Verilator requires using C++14 standard to compile the simulation files. However, the compile options are hard coding in `sim/src/main/scala/spinal/sim/VerilatorBackend.scala`. Is it possible to change compile options by user?

andreasWallner, 2023-09-26:
I'm currently updating to check, still one question: do you pass any additional parameters to verilator? I've been using `5.005` which should have the same C++ requirements, which worked.

andreasWallner, 2023-09-26:
Just updated to `5.0017` with which I can reproduce the issue.

PR #1210 (2023-09-27): Update C++ version for verilator (to C++14)
Closes #1209

# Context, Motivation & Description

With the jump to verilator 5, verilator started using newer C++ syntax. For now it seems that compiling against C++14 is sufficient (when not using the recent timing support that required C++20).

It seems we pinned the C++ version to C++11 to support older compilers (see #278).
This PR removes the pinned version and goes back to using the compiler default that should work in most cases, possibly being faster then if set to C++11. It provides an additional environment variable to add arbitrary settin [... cut]
closed 2023-09-27 by Dolu1990