SpinalHDL/SpinalHDL #1211: Use default ClockDomain without reset signal

Alen302, 2023-09-27:
Hi, I encountered a problem about the clock domain when I used the Blackbox to encapsulate the RTL code:
There is no explicit use of registers in my Blackbox code. I only need CLK instead of RESET, so I use mapClockDomain(clock= CLK), But when I used forkStimulus in the simulation, it assumed that I was using the reset signal:
An exception or error caused a run to abort: key not found: (reset :  Bool) 
java.util.NoSuchElementException: key not found: (reset :  Bool)
	at scala.collection.MapLike.default(MapLike.scala:236)
	at scala.collection.MapLike.default$(MapLike.scala:235)
	at scala.collection.AbstractMap.default(Map.scala:65)
	at scala.collection.mutable.HashMap.apply(HashMap.scala:69)
	at spinal.core.sim.package$SimClockDomainPimper.getBool(package.scala:570)
	at spinal.core.sim.package$SimClockDomainPimper.resetSim(package.scala:580)
	at spinal.core.sim.package$SimClockDomainPimper.deassertReset(package.scala:876)
	at spinal.core.sim.package$SimClockDomainPimper.forkStimulus(package.scala:787)

I would like to ask how to make the default clock domain only use CLK in the io list as the clock signal, instead of using the RESET port, My solution for now is to temporarily use the resetKind = BOOT configuration in SpinalSim, But that doesn't seem to be a good solution.

Dolu1990, 2023-09-27:
Maybe like this : 

```scala
  class Toplevel extends Component {
    val counter = Reg(UInt(8 bits))
    counter := counter + 1
  }

  SimConfig.withWave.compile(
    ClockDomain.external("mainCd", withReset = false) on new Toplevel
  ).doSim { dut =>
    dut.clockDomain.forkStimulus(10)
    dut.clockDomain.waitSampling(5)
  }
```

andreasWallner, 2023-10-02:
@Alen302 Does either of our answers work for you? If yes I'd close this.

Alen302, 2023-10-26:
Thanks, This answer is very helpful!

closed 2023-10-26 by Alen302