SpinalHDL/SpinalHDL #1214: RFC: Hardware Sum Types (aka TaggedUnion)

ronan-lashermes, 2023-10-02:
# RFC: Hardware Sum Types (aka TaggedUnion)

This is a description of a pet idea that I did not succeed to implement properly.
At this point, I would like to gather opinions on this topic, and some ideas to solve my questions in the corresponding section below.

## Introduction 

Modern programming trends leverage the type system to ensure program correctness whenever possible.
With SpinalHDL, this trend can be applied to hardware types.

Principle P1 can be stated as:

> The type of a value should designate its purpose in the most specific way possible.

Consider a simple example for a signal that indicates whether a request is a read or a write:

```scala
val read_write = Bool() //incorrect, does not follow P1

when(read_write === True) { // read or write?
    //...
}
```

A better way to define this variable, while satisfying P1, would be:

```scala
object ReadWrite extends SpinalEnum {
  val read, write = newElement()
}

val read_write = ReadWrite()

when(read_write === read) { //unambiguous
    //...
}
```

This isn't about correctness, as both implementations are correct. However, the second one leverages the type system to remove ambiguity, thus reducing the verification effort.

## What Are Sum Types / Tagged Unions?

A product type is a composite type that includes both type A **and** type B. This is the regular SpinalHDL’s Bundle.
A sum type, on the other hand, is a composite type that includes either type A **or** type B.

The most well-known implementation of sum types is probably in Rust:

```rust
enum ReadWriteRequest {
    Read(ReadRequest),
    Write(WriteRequest)
}
```

Our Rust example defines a *ReadWriteRequest* type that can either be a *Read* with type *ReadRequest* or a *Write* with type *WriteRequest*, but not both simultaneously.

A hardware sum type would be equivalent: for the I/O of a Read/Write port, the type would specify that only one of the two is valid at a given time.
For data in registers, the type would specify that only one of the two is valid.

The goal is for the type system to describe the hardware and its associated semantics more precisely.
There are several possible hardware ways to implement a sum type, which we will discuss below.

## Implementation ideas (not working for now)

### Usage

Let’s start with how we would use such a thing in SpinalHDL.
This description will include sleights of hand, to simplify the discussion.

```scala
[... 289 lines cut ...]
        out(selector)
        in(dataInMaster)
        out(dataOutMaster)
        out(dataDirectionLess)
    }

    override def asSlave(): Unit = {
        isMaster = false
        in(selector)
        out(dataInMaster)
        in(dataOutMaster)
        in(dataDirectionLess)
    }
}

object TaggedUnionCraft {
    private def setAsMasterIfPossible(data: Data): Unit = {
        data match {
            case masterSlave: IMasterSlave => masterSlave.asMaster()
            case _ =>
        }
    }
}

```

Dolu1990, 2023-10-06:
Hi,

Tagged union would be usefull, i have to take some time to think about it.

Thanks for the issue :)

ronan-lashermes, 2023-10-13:
Neat ! I will take some time next week to give you a feedback.

Dolu1990, 2023-10-16:
Hi,

Here is one implementation of a tagged union. Not saying it is good, but that's just to give some working code :

```scala

    class TaggedUnion extends Bundle with PostInitCallback{
      var tag : UInt = null
      val union = new Union(selfBuild = false)

      def newElement[T <: Data](t: HardType[T]) = {
        union.newElement(t)
      }

      def build(): Unit = {
        union.build()
        tag = UInt(log2Up(union.uTypes.size) bits)
        valCallbackRec(tag, "tag")
      }

      override def postInitCallback() = {
        build()
        this
      }

      def setTag(e : UnionElement[_ <: Data]) = {
        tag := union.uTypes.indexOf(e)
      }
      
      def sMatch(body : Any => Unit): Unit = {
        switch(tag){
          for((e, i) <- union.uTypes.zipWithIndex) is(i){
            body(e)
          }
        }
      }
    }
```

```scala
    case class MyTaggedUnion() extends TaggedUnion{
      val a = newElement(TypeA())
      val b = newElement(TypeB())
    }

    val mtu = MyTaggedUnion()
    mtu.setTag(mtu.b)
    mtu.union.raw := 0
    mtu.b.m := 12

    val rawrr = UInt(10 bits)
    mtu.sMatch{
      case mtu.a => rawrr := mtu.a.z.resized
      case mtu.b => rawrr := mtu.b.n.resized
    }
```

Dolu1990, 2023-10-16:
Question now is :
- Should union stay as it is
- Or should it be something more like enum, meaning the type is defined a bit like SpinalEnum, where you define a type, that you can then "craft" into the netlist

I think it is an important consideration.

Dolu1990, 2023-10-16:
I pushed 2 new feature which completly overlap the previously added Union class : 
- HardType.union(Data*) : HardType[Bits] => Return a hardtype[Bits] wide enough to hold the bigest of all the type given as arguements
- Bits.union(HardType[T]) : T => return a hardware instance of T which read Bits, and write Bits on assignements.

Here is an example : 

```scala
    case class TypeA() extends Bundle {
      val x, y, z = UInt(8 bits)
    }

    case class TypeB() extends Bundle {
      val l, m, n = UInt(4 bits)
      val rgb = Rgb(2,3,4)
    }

    val MyUnion = HardType.union(TypeA(), TypeB())

    val miaou, wuff : Bits = MyUnion()
    wuff := miaou
    miaou := 0

    val a : TypeA = miaou.union(TypeA())
    val b : TypeB = miaou.union(TypeB())

    b.m(2 downto 1) := U"10"
    val sel = in UInt(2 bits)
    b.rgb.g(1) := False
    b.rgb.b(sel) := True
    a.z(sel, 2 bits) := U"11"

    a.y := U(4)
```

I think i kind of prefer that aproche compared to the previously added Union class, as it seems more flexible more decoupled.

Dolu1990, 2023-10-16:
Also the good thing about this aproache, is that we actualy "synthetise" the type once, instead of reprocessing it at every instance.
The bad thing is that the union is represented by Bits on the scala typing, so you can assign union of different kinds into each other without typing error being catched.

ronan-lashermes, 2023-10-16:
Does it handle directions ?
```
case class TypeA() extends Bundle {
      val x, y, z = in UInt(8 bits)
    }

    case class TypeB() extends Bundle {
      val l, m, n = in UInt(4 bits)
      val rgb = out Rgb(2,3,4)
    }
```

The size of the union should be max(sizes in) = 24 bits + max(sizes out) = size of Rgb.
MultiData may handle this kind of stuff, I am not experienced enough with the language but dealing with I/O bundles seems to be the harder part.

ronan-lashermes, 2023-10-16:
I am not sure that the Bits typing for the union is a problem, detecting sound assignments should be handled at the TaggedUnion level in my opinion.

Dolu1990, 2023-10-16:
I would say, i think it would work just fine with in out, but i also think that that's realy not a place for in out to be used.

Size of miaou is 24 bits, as TypeA is the biggest with 8+8+8 bits

ronan-lashermes, 2023-10-16:
Hmm, all the examples I have in mind to use in/out seem indeed a bit far fetched, so I am not too sure we should support the use case. 

```scala
case class ReadPort() extends Bundle  {
    val r1 = in Bits(8 bits)

}

case class WritePort() extends Bundle {
    val r2 = in Bits(5 bits)
    val r3 = out Bits(3 bits)
}

object ReadWritePort extends TaggedUnion {
    val read = ReadPort()
    val write = WritePort()
}
```

ronan-lashermes, 2023-10-20:
I did test a bit the new Union stuff and it works great. But it is a bit dangerous to use : you can override assignments without noticing. The TaggedUnion is critical for proper use. I will try to propose a full TaggedUnion implementation.