SpinalHDL/SpinalHDL #1216: SBT does not work on  MySpinalProject

PythonLinks, 2023-10-04:
I am able to start sbt on an empty directory, but not on the git clone of MySpinalProject.  For context I am completely new to Java, Scala and SpinalHDL.   

Here is the short error message. 
> error:   bad constant pool index: 

This is my environment. 
`
M1 Mac 
Mac OS X Ventura 13.5.1 (22G90)

java -version
java version "21" 2023-09-19 LTS
Java(TM) SE Runtime Environment (build 21+35-LTS-2513)
Java HotSpot(TM) 64-Bit Server VM (build 21+35-LTS-2513, mixed mode, sharing)

bash-3.2$ scala -version
Scala code runner version 3.3.1 -- Copyright 2002-2023, LAMP/EPFL

bash-3.2$ sbt -version

 bash-3.2$ sbt -version
sbt version in this project: 1.9.6
sbt script version: 1.9.6

Here is the full error log. 

bash-3.2$ pwd
/Users/lozinski/Documents/Products/FORTH/CODE/MySpinalProject
bash-3.2$ sbt
[info] welcome to sbt 1.6.0 (Homebrew Java 21)
error:
  bad constant pool index: 0 at pos: 48445
     while compiling: <no file>
        during phase: globalPhase=<no phase>, enteringPhase=<some phase>
     library version: version 2.12.15
    compiler version: version 2.12.15
  reconstructed args: -classpath /Users/lozinski/.sbt/boot/scala-2.12.15/lib/scala-library.jar -Yrangepos

  last tree to typer: EmptyTree
       tree position: <unknown>
            tree tpe: <notype>
              symbol: null
           call site: <none> in <none>

== Source file context for tree position ==

error:
  bad constant pool index: 0 at pos: 48445
     while compiling: <no file>
        during phase: globalPhase=<no phase>, enteringPhase=<some phase>
     library version: version 2.12.15
    compiler version: version 2.12.15
  reconstructed args: -classpath /Users/lozinski/.sbt/boot/scala-2.12.15/lib/scala-library.jar -Yrangepos

  last tree to typer: EmptyTree
       tree position: <unknown>
            tree tpe: <notype>
              symbol: null
           call site: <none> in <none>

== Source file context for tree position ==

Exception in thread "sbt-parser-init-thread" java.lang.ExceptionInInitializerError
	at sbt.internal.parser.SbtParserInit$$anon$2.run(SbtParser.scala:191)
Caused by: scala.reflect.internal.FatalError: 
  bad constant pool index: 0 at pos: 48445
     while compiling: <no file>
        during phase: globalPhase=<no phase>, enteringPhase=<some phase>
     library version: version 2.12.15
    compiler version: version 2.12.15
[... 353 lines cut ...]
[error] 	at scala.tools.nsc.symtab.SymbolLoaders$SymbolLoader.complete(SymbolLoaders.scala:250)
[error] 	at scala.reflect.internal.Symbols$Symbol.completeInfo(Symbols.scala:1542)
[error] 	at scala.reflect.internal.Symbols$Symbol.info(Symbols.scala:1514)
[error] 	at scala.reflect.internal.Definitions.scala$reflect$internal$Definitions$$enterNewMethod(Definitions.scala:49)
[error] 	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus$lzycompute(Definitions.scala:1134)
[error] 	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus(Definitions.scala:1134)
[error] 	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods$lzycompute(Definitions.scala:1438)
[error] 	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods(Definitions.scala:1420)
[error] 	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode$lzycompute(Definitions.scala:1450)
[error] 	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode(Definitions.scala:1450)
[error] 	at scala.reflect.internal.Definitions$DefinitionsClass.init(Definitions.scala:1506)
[error] 	at scala.tools.nsc.Global$Run.<init>(Global.scala:1213)
[error] 	at sbt.internal.parser.SbtParser$.<init>(SbtParser.scala:141)
[error] 	at sbt.internal.parser.SbtParser$.<clinit>(SbtParser.scala)
[error] 	at sbt.internal.parser.SbtParserInit$$anon$2.run(SbtParser.scala:191)
[error] java.lang.NoClassDefFoundError: Could not initialize class sbt.internal.parser.SbtParser$
[error] Use 'last' for the full log.
[debug] > Exec(loadFailed, None, None)
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
[debug] > Exec(last, None, None)
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
bash-3.2$ 
bash-3.2$ 

`

andreasWallner, 2023-10-04:
Hi,
Could you please check if our template works for you: https://github.com/spinalhdl/spinaltemplatesbt ?

PythonLinks, 2023-10-04:
I am very happy to do whatever it takes to get this most interesting software working. 

Sadly that also did not work.  
Below is the log. 

And thank you for the blindingly fast service. 
It makes SpinalHDL look even more promising. 
Chris
`
bash-3.2$ git clone https://github.com/spinalhdl/spinaltemplatesbt
Cloning into 'spinaltemplatesbt'...
remote: Enumerating objects: 431, done.
remote: Counting objects: 100% (146/146), done.
remote: Compressing objects: 100% (72/72), done.
remote: Total 431 (delta 68), reused 108 (delta 51), pack-reused 285
Receiving objects: 100% (431/431), 56.48 KiB | 1.28 MiB/s, done.
Resolving deltas: 100% (192/192), done.
bash-3.2$ cd spinaltemplatesbt/
bash-3.2$ sbt
error:
  bad constant pool index: 0 at pos: 48445
     while compiling: <no file>
        during phase: globalPhase=<no phase>, enteringPhase=<some phase>
     library version: version 2.12.15
    compiler version: version 2.12.15
  reconstructed args: -classpath /Users/lozinski/.sbt/boot/scala-2.12.15/lib/scala-library.jar -Yrangepos

  last tree to typer: EmptyTree
       tree position: <unknown>
            tree tpe: <notype>
              symbol: null
           call site: <none> in <none>

== Source file context for tree position ==

error:
  bad constant pool index: 0 at pos: 48445
     while compiling: <no file>
        during phase: globalPhase=<no phase>, enteringPhase=<some phase>
     library version: version 2.12.15
    compiler version: version 2.12.15
  reconstructed args: -classpath /Users/lozinski/.sbt/boot/scala-2.12.15/lib/scala-library.jar -Yrangepos

  last tree to typer: EmptyTree
       tree position: <unknown>
            tree tpe: <notype>
              symbol: null
           call site: <none> in <none>

== Source file context for tree position ==

Exception in thread "sbt-parser-init-thread" java.lang.ExceptionInInitializerError
	at sbt.internal.parser.SbtParserInit$$anon$2.run(SbtParser.scala:191)
Caused by: scala.reflect.internal.FatalError: 
  bad constant pool index: 0 at pos: 48445
     while compiling: <no file>
        during phase: globalPhase=<no phase>, enteringPhase=<some phase>
     library version: version 2.12.15
    compiler version: version 2.12.15
  reconstructed args: -classpath /Users/lozinski/.sbt/boot/scala-2.12.15/lib/scala-library.jar -Yrangepos

  last tree to typer: EmptyTree
       tree position: <unknown>
            tree tpe: <notype>
              symbol: null
           call site: <none> in <none>

== Source file context for tree position ==

	at scala.reflect.internal.Reporting.abort(Reporting.scala:69)
[... 166 lines cut ...]
	at scala.tools.nsc.symtab.classfile.ClassfileParser.$anonfun$parse$2(ClassfileParser.scala:161)
	at scala.tools.nsc.symtab.classfile.ClassfileParser.$anonfun$parse$1(ClassfileParser.scala:147)
	at scala.tools.nsc.symtab.classfile.ClassfileParser.parse(ClassfileParser.scala:130)
	at scala.tools.nsc.symtab.SymbolLoaders$ClassfileLoader.doComplete(SymbolLoaders.scala:343)
	at scala.tools.nsc.symtab.SymbolLoaders$SymbolLoader.complete(SymbolLoaders.scala:250)
	at scala.reflect.internal.Symbols$Symbol.completeInfo(Symbols.scala:1542)
	at scala.reflect.internal.Symbols$Symbol.info(Symbols.scala:1514)
	at scala.reflect.internal.Definitions.scala$reflect$internal$Definitions$$enterNewMethod(Definitions.scala:49)
	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus$lzycompute(Definitions.scala:1134)
	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus(Definitions.scala:1134)
	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods$lzycompute(Definitions.scala:1438)
	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods(Definitions.scala:1420)
	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode$lzycompute(Definitions.scala:1450)
	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode(Definitions.scala:1450)
	at scala.reflect.internal.Definitions$DefinitionsClass.init(Definitions.scala:1506)
	at scala.tools.nsc.Global$Run.<init>(Global.scala:1213)
	at sbt.internal.parser.SbtParser$.<init>(SbtParser.scala:141)
	at sbt.internal.parser.SbtParser$.<clinit>(SbtParser.scala)
	at sbt.internal.parser.SbtParserInit$$anon$2.run(SbtParser.scala:191)
[error] java.lang.NoClassDefFoundError: Could not initialize class sbt.internal.parser.SbtParser$
[error] Use 'last' for the full log.
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
Invalid response: ''
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)
`

PythonLinks, 2023-10-04:
So I used corsair to install  scala version 3.3.1
and edited build.sc and build.sbt to refer to the correct scala version.  

I also edited those two files to refer to the same spinal version 1.9.3.  Previously spinalTemplateSbt/build.sc refered to spinalVersion 1.9.0
Still the same problem. 

Thank you so much for all of the help.   

Should I revert to older versions of everything?  

J1Sc uses

scalaVersion := "2.11.8"

val spinalVersion = "1.4.2"
 
 I had difficulty finding an older version of Java. 
 So I thought I should first get just spinalHDL working on newer versions of the tools, and then port J1Sc to the newer versions. 
 
I added lots of `<br>` statements to the text below to make it more readable. 

<details><summary>
Here are the details of the error message. 
</summary>
` 
 bash-3.2$ pwd<br>
/Users/lozinski/Documents/Products/FORTH/CODE/spinaltemplatesbt<br>
bash-3.2$ <br>
bash-3.2$ <br>
bash-3.2$ java -version<br>
java version "21" 2023-09-19 LTS
Java(TM) SE Runtime Environment (build 21+35-LTS-2513)
Java HotSpot(TM) 64-Bit Server VM (build 21+35-LTS-2513, mixed mode, sharing)
bash-3.2$ <br>
bash-3.2$ <br>
bash-3.2$ scala -version<br>
Scala code runner version 3.3.1 -- Copyright 2002-2023, LAMP/EPFL
bash-3.2$ <br>
bash-3.2$ <br>
bash-3.2$ grep spinalVersion *<br>
build.sbt:val spinalVersion = "1.9.3"
build.sbt:val spinalCore = "com.github.spinalhdl" %% "spinalhdl-core" % spinalVersion
build.sbt:val spinalLib = "com.github.spinalhdl" %% "spinalhdl-lib" % spinalVersion
build.sbt:val spinalIdslPlugin = compilerPlugin("com.github.spinalhdl" %% "spinalhdl-idsl-plugin" % spinalVersion)
build.sc:val spinalVersion = "1.9.3"
build.sc:    ivy"com.github.spinalhdl::spinalhdl-core:$spinalVersion",
build.sc:    ivy"com.github.spinalhdl::spinalhdl-lib:$spinalVersion"
build.sc:  def scalacPluginIvyDeps = Agg(ivy"com.github.spinalhdl::spinalhdl-idsl-plugin:$spinalVersion")
grep: hw: Is a directory
grep: project: Is a directory
grep: target: Is a directory<br>
bash-3.2$ <br>
bash-3.2$ <br>
bash-3.2$ sbt<br>
error:
  bad constant pool index: 0 at pos: 48445
     while compiling: <no file>
        during phase: globalPhase=<no phase>, enteringPhase=<some phase>
     library version: version 2.12.15
    compiler version: version 2.12.15
  reconstructed args: -classpath /Users/lozinski/.sbt/boot/scala-2.12.15/lib/scala-library.jar -Yrangepos
<br><br>
  last tree to typer: EmptyTree
       tree position: <unknown>
            tree tpe: <notype>
              symbol: null
           call site: <none> in <none>
<br>
== Source file context for tree position ==
<br>
[... 202 lines cut ...]
	at scala.tools.nsc.symtab.classfile.ClassfileParser.$anonfun$parse$2(ClassfileParser.scala:161)
	at scala.tools.nsc.symtab.classfile.ClassfileParser.$anonfun$parse$1(ClassfileParser.scala:147)
	at scala.tools.nsc.symtab.classfile.ClassfileParser.parse(ClassfileParser.scala:130)
	at scala.tools.nsc.symtab.SymbolLoaders$ClassfileLoader.doComplete(SymbolLoaders.scala:343)
	at scala.tools.nsc.symtab.SymbolLoaders$SymbolLoader.complete(SymbolLoaders.scala:250)
	at scala.reflect.internal.Symbols$Symbol.completeInfo(Symbols.scala:1542)
	at scala.reflect.internal.Symbols$Symbol.info(Symbols.scala:1514)
	at scala.reflect.internal.Definitions.scala$reflect$internal$Definitions$$enterNewMethod(Definitions.scala:49)
	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus$lzycompute(Definitions.scala:1134)
	at scala.reflect.internal.Definitions$DefinitionsClass.String_$plus(Definitions.scala:1134)
	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods$lzycompute(Definitions.scala:1438)
	at scala.reflect.internal.Definitions$DefinitionsClass.syntheticCoreMethods(Definitions.scala:1420)
	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode$lzycompute(Definitions.scala:1450)
	at scala.reflect.internal.Definitions$DefinitionsClass.symbolsNotPresentInBytecode(Definitions.scala:1450)
	at scala.reflect.internal.Definitions$DefinitionsClass.init(Definitions.scala:1506)
	at scala.tools.nsc.Global$Run.<init>(Global.scala:1213)
	at sbt.internal.parser.SbtParser$.<init>(SbtParser.scala:141)
	at sbt.internal.parser.SbtParser$.<clinit>(SbtParser.scala)
	at sbt.internal.parser.SbtParserInit$$anon$2.run(SbtParser.scala:191)
[error] java.lang.NoClassDefFoundError: Could not initialize class sbt.internal.parser.SbtParser$
[error] Use 'last' for the full log.
[warn] Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? (default: r)

`
</details>

andreasWallner, 2023-10-04:
After a bit more looking: Java 21 is definitely not compatible with Scala 2.12.15 (see https://docs.scala-lang.org/overviews/jdk-compatibility/overview.html#version-compatibility-table)

Something in your environment is still making a problem, I'd first try to get the SpinalTemplateSbt to run so that we don't have to try to update 7 year old tools at the same time we are trying to fix the environment.
SpinalTemplateSbt uses 2.12.16 by default, which also does not support Java 21 (17 is still the recommended LTS version, https://whichjdk.com/, I can't even install it to try it).

So: could you please try to bump (only) the scala version in the template to 2.12.18 (in `build.sbt`: `ThisBuild / scalaVersion := "2.12.18"`, you can ignore `build.sc` that is for another tool) and try to run sbt then? 

If you are tied to Java 21: https://formulae.brew.sh/formula/openjdk@17

PythonLinks, 2023-10-05:
Thank you.  sbt now works in 

./spinalTemplatesbt,

and in 

./MySpinalProject, 

but not yet in 

./J1Sc.  

Not your problem.  

System Administration is hard to do, because I do not really know what Mac OS X does with Java and what brew does.  Also there are multiple ways of installing things, and after all of that does not quite work, it is not quite clear how to uninstall them.  And be careful not to delete the version of Java used by the Mac OS X itself!

I used these commands. 
`
 brew uninstall sbt
 brew uninstall scala
 
How to uninstall Java. 
`
https://www.java.com/en/download/help/mac_uninstall_java.html

To manage multiple versions of Java, it is also essential to have jenv installed.  
`
brew install jenv

It added these lines to my .bash_profile

export PATH="$HOME/.jenv/bin:$PATH"
eval "$(jenv init -)"
 `

Then we have to install Java 1.7, confusingly also called Java 17. 
`
brew install openjdk@17

And then add this to your path. 
export PATH="/opt/homebrew/opt/openjdk@17/bin:$PATH"
`
This all sounds very straight forward.  I am not quite sure that i documented it correctly, and I do not have another mac to test out the installation procedure,  but it is working.  The key point is stick to this path, and do not try to install any of this any other way, and you should be fine.  

As for me, next I will try to get a SpinalHDL blinky working, and once that is okay, I should be able to port the J1Sc over to the current release of SpinalHDL. 

And now I am off for a bicycle ride, on my beautiful brand new Romet treking bicycle!

andreasWallner, 2023-10-05:
Great to hear!

When you update J1sc then I'd start with the sbt version (in `project/build.properties`), easy to overlook^^.

Also: for questions you can also join us in the gitter channel https://gitter.im/SpinalHDL/SpinalHDL

closed 2023-10-05 by andreasWallner

Dolu1990, 2023-10-06:
@andreasWallner I'm thinking that maybe we should move everything to 2.12.18 ? 
I kinda never realy went to the 2.12 lands XD
Did you had some experience with it ? (incompatibilties and stuff like it XD)

andreasWallner, 2023-10-06:
@Dolu1990 
I'm currently using 2.13.6 - updating the minor versions does make sense though (even though one can use our built 2.12 lib for 2.12.x, tried it yesterday with 2.12.18).
I'll open a PR

Dolu1990, 2023-10-09:
Ahhh 2.13 is realy the worst with the Seq stuff XD