SpinalHDL/SpinalHDL #1217: Better documentation for library components

marklinmax, 2023-10-06:
The documentation seems to be way behind the SpinalHDL lib repository.

As an example, the lib has AXI4 Crossbar, which is never mentioned in the documentation. A lot of the com modules are not documented either.

Could there be an effort from those who wrote / understand those modules to improve / add the documentation on them?

PythonLinks, 2023-10-06:
My theory is that if you find something different between the code and the docs, edit the docs, and issue a pull request.   "Many hands make light work."

andreasWallner, 2023-10-06:
@PythonLinks One quick hint here: If you plan to document an undocumented module, please say something first, some parts of `lib` are not documented by choice, e.g. because we didn't want to commit to the API.

Dolu1990, 2023-10-09:
The main issue i have with documenting things, is that i'm mostly overloaded.

Between the email, bug tracking, and other SpinalHDL related tasks, i do not have the will power to expend the documentation. Keeping the motivation to do things on long term projects (as SpinalHDL) is a marathon, and i'm trying to not push myself too much, keep the fun, to avoid the risk of getting burned out XD

So, help is welcome