SpinalHDL/SpinalHDL #1224: RFC: repeat data specified times.

Readon, 2023-10-17:
There are cases that need to repeat Bits n times to construct a new signal with n times width.
What I do now is to use:
``` scala
val new = Vec((0 until times).map(_ => old)).asBits
```
This is really heavy and not good for understand, especially when the expression of old is long.

Define a method repeat can help with this.
``` scala
val new = old.repeat(times)
```

andreasWallner, 2023-10-20:
Stumbled over this: https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/Data.scala#L777
`#*` is the operator that does what you need, it's on my list to document it...

closed 2023-10-20 by Readon

andreasWallner, 2023-10-20:
Just added PR for docs: https://github.com/SpinalHDL/SpinalDoc-RTD/pull/223