SpinalHDL/SpinalHDL #1225: The signals in the toplevel port of a Verilog module should be explicitly given a net-type.

RiceShelley, 2023-10-20:
Currently Spinal generates top level Verilog port descriptions similar to the following example. 

```
module Top (
  output              io_sdClk,
  input               io_sdCmd_read,
  output              io_sdCmd_write,
  output              io_sdCmd_writeEnable,
  input      [3:0]    io_sdData_read,
  output reg [3:0]    io_sdData_write,
  output reg [3:0]    io_sdData_writeEnable,
  output              io_led,
  output              io_led_red,
  output reg          io_led_green,
  input               resetn,
  input               clkDom33Mhz_clk,
  output              vio_uart_txd,
  input               vio_uart_rxd,
  input               clkDom33Mhz_resetn,
  input               clk
);
```

Not specifying the net type for signals commonly leads to synthesis flows failing or giving warnings.
Example:

```ERROR: net type must be explicitly specified for 'io_sdClk' when default_nettype is none (VERI-1906)```

In addition to the above since the net type is not specified or is "none" it is susceptible to being changed if the user has set `default_nettype elsewhere in the project. 

I am proposing that the net type in the generated Verilog be explicitly specified. 
Example:
 ```
module Top (
  output wire             io_sdClk,
  input  wire             io_sdCmd_read,
  output wire             io_sdCmd_write,
  output wire             io_sdCmd_writeEnable,
  input  wire [3:0]       io_sdData_read,
  output reg  [3:0]       io_sdData_write,
  output reg  [3:0]       io_sdData_writeEnable,
  output wire             io_led,
  output wire             io_led_red,
  output reg              io_led_green,
  input  wire             resetn,
  input  wire             clkDom33Mhz_clk,
  output wire             vio_uart_txd,
  input  wire             vio_uart_rxd,
  input  wire             clkDom33Mhz_resetn,
  input  wire             clk
);
```
Explicitly specifying the net type would prevent synthesis flow errors with some vendor toolchains, and ensure that the proper nettype is always used.

If this seems reasonable I would like to try a PR.

Dolu1990, 2023-10-20:
Seems good for me, PR welcome :)

andreasWallner, 2023-10-21:
I think that we also have to add the nettype to all signals in the module, not only in the interface?

Dolu1990, 2023-10-23:
I guess so yes, but i don't know verilog well at all.

RiceShelley, 2023-10-25:
It's my understanding that currently the nettype is set for all signals inside the module. @andreasWallner

PR #1228 (2023-10-26): Added explicit net types to verilog ports

Closes #1225 

# Context, Motivation & Description

Adds explicit net types to the top level ports of Verilog modules. Specifying the net type will keep some vendor synthesis flows from failing or giving warnings.

Example:
```ERROR: net type must be explicitly specified for 'io_sdClk' when default_nettype is none (VERI-1906)```

In addition to the above since the net type is not specified or is "none" it is susceptible to being changed if the user has set `default_nettype elsewhere in the project.

Notes:
Did not add unit test because the change is s [... cut]
closed 2023-10-26 by andreasWallner