SpinalHDL/SpinalHDL #1226: Expected HW when doing `/=` assign to Reg

andreasWallner, 2023-10-21:
Came up today in Gitter: When using `/=` on a Reg the outcome "feels" weird. From what I understand `/=` should kind of mimic a VHDL variable, which does not have a similar concept. But in SpinalHDL we can currently mix those things:

```scala
object TestAssign extends App {
  // the example from the docs, just with Reg
  SpinalVerilog(new Component {
    var x = Reg(UInt(4 bits))
    val y, z = UInt(4 bits)
    x := 0
    y := x
    x \= x + 1
    z := x
  })
}
```
In the resulting HW `x` will stay `0` at all times, `z` will be `1` - at face value, ignoring that I know it should mimic VHDL (new user) this feels weird since we did do an assignment to `x` with `x + 1`. Do we expect this behavior?

I'm not sure it's worth it to "fix" this, but I think it would make sense to catch it by checking whether there are any `DataAssignmentStatements` to targets with a `VarAssignmentTag` on them.

PS: the generated HW is
```
  reg        [3:0]    x_1;
  reg        [3:0]    x;
  wire       [3:0]    y;
  wire       [3:0]    z;

  always @(*) begin
    x_1 = x;
    x_1 = (x + 4'b0001);
  end

  assign y = x;
  assign z = x_1;
  always @(posedge clk) begin
    x <= 4'b0000;
  end
```

Dolu1990, 2023-10-23:
Hi,

It is the expected behaviour.
The intented behaviour is to not use \= on registers, but only for combinatorial logic.

Maybe we should just throw an error then \= is used on a register ?

andreasWallner, 2023-10-24:
I this it would make sense to produce an error in that case - just to make it very clear that it's not supported.

closed 2023-10-24 by Dolu1990