SpinalHDL/SpinalHDL #1227: the issues of tester's sim

Amegodvivado, 2023-10-25:
I want to sim the SpinalSimSpiXdrMaster.scala,but some problem:

`    test(prefix + "compile") {
      doCompile()
    }

    for (cpol <- List(false, true); cpha <- List(false, true); repeat <- 0 to 0) {
      val name = s"$prefix test cpol=$cpol cpha=$cpha $repeat"
      test(name) {
        if(compiled == null){
          doCompile()
        }`
the "test" always be error(highlight),
and  in SpinalFormalFunduite.scala,
`import org.scalatest.funsuite.AnyFunSuite
import org.scalatest.Tag
import spinal.core._`,scalatest and AnyFunsuite is red higlight.

may be sbt 'probelm or scala'version?

andreasWallner, 2023-10-26:
Hard to say you don't say whether you are trying to run a test in the SpinalHDL project or what you are doing.
If you are trying to run a simulation in the SpinalHDL project then it's most likely an issue with your setup. If it's in you project then I'd guess that you are missing some dependencies in your `build.sbt`

Amegodvivado, 2023-10-27:
The test is in file of tester,I want to konw how to use the lib SpiXdrMasterctrl.scala,so the test project of SpinalSimSpiXdrMaster.scala my explain it. Anyone sim it before?

andreasWallner, 2023-10-27:
The code that you are trying to run uses scalatest, but your build.sbt does not include it.
You'll have to add `"org.scalatest" %% "scalatest" % "3.1.1" % "test",` to the libraryDependencies to make this work.

Amegodvivado, 2023-10-27:
Thank you for your answer，The scalatest has be added ,by the way,how to add  the file-SpinalHdl/tester,the SpinalSimSpiXdrMaster.scala   lib in it.When i copy this file to my project ,it always be red of highlight, because of above highlight though add the scalatest.

andreasWallner, 2023-10-27:
If I understand correctly then you still get an error even though you added scalatest?
Looking again at your `build.sbt` that makes sense: The `"test"` at the end of the dependency limits it to test builds. At the same time you configure an unusual directory layout, have a look [here](https://www.scala-sbt.org/1.x/docs/Directories.html) for the default.
I'd remove the `Compile / scalaSource` line from the file, move the normal HDL sources to a folder `src/main/scala/<packageName>` and the tests to `src/test/scala/<packageName>` (similar to how [VexRiscv does it](https://github.com/SpinalHDL/VexRiscv/tree/master/src)).

If you don't want to separate HDL/tests into separate folders and keep your folder structure then you could remove the `"test"` at the end of the scalatest dependencies, that limits the dependency to test builds only.

Readon, 2023-11-07:
you can import spinal-tester as below:
``` scala
val spinalTester = "com.github.spinalhdl" %% "spinalhdl-tester" % spinalVersion % "test"
libraryDependencies ++= Seq(spinalCore, spinalLib, spinalIdslPlugin, spinalTester)
```
directly use scalatest, especially using 3.2.5 would cause some problem on dependency.
Notice that, until spianlhdl 1.8.0 the version of scalatest is updated. And spinal-tester release with SpinalAnyFunSuite only after 1.9.4, if I remmeber right.

Readon, 2023-11-07:
I suggest you use scalatest 3.2.14, the same as Spinalhdl itself.