SpinalHDL/SpinalHDL #1229: tester version conflict in scala 2.12 but not 2.11.

Readon, 2023-10-30:
I am trying on update the SpinalWorkshop and SpinalTemplateSbt to depends on spinal.tester only but import scalatest directly.
So I add below line into build.sbt
```
val spinalTester = "com.github.spinalhdl" %% "spinalhdl-tester" % spinalVersion % "test"
```
it works when I use scala 2.11 as default scala version.

However, when I switched scalaVersion to 2.12, the `sbt test` would report error that 
```
(update) found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[error]
[error]         * org.scala-lang.modules:scala-xml_2.12:2.1.0 (early-semver) is selected over 1.0.6
[error]             +- org.scalatest:scalatest-core_2.12:3.2.14           (depends on 2.1.0)
[error]             +- org.scala-lang:scala-compiler:2.12.16              (depends on 1.0.6)
```
Where the "org.scala:scalatest-core_**2.12**:3.2.14" package is required by spinalhdl-tester for testing.
It means that package "org.scala:scalatest-core_**2.11**:3.2.14" do not have this constraints.

I have also checked that it seems that we need to depend on old package of scalatest to support scala 2.12.

Readon, 2023-10-30:
@andreasWallner @Dolu1990 have you met this while use scala 2.12?

Readon, 2023-10-30:
or even more for scala 2.13?

andreasWallner, 2023-10-30:
Didn't see that before but it might be caused by the old scala 2.12 that we are depending on - did you try switching to the latest 2.12.18?

andreasWallner, 2023-10-30:
We did change the version of few days ago, not released yet though

Dolu1990, 2023-10-30:
> to depends on spinal.tester

But  spinal.tester isn't released, right ?
Did you localy released SpinalHDL  tester, or integrated SpinalHDL code as a sbt module ?

I had much pain in general with the second option.

Readon, 2023-10-30:
> > to depends on spinal.tester
> 
> But spinal.tester isn't released, right ? Did you localy released SpinalHDL tester, or integrated SpinalHDL code as a sbt module ?
> 
> I had much pain in general with the second option.

it is released since 1.8.0. I suppose.

Dolu1990, 2023-10-31:
Ahhh right XD

I think we can release a new SpinalHDL version. Seems good to you ?

Readon, 2023-10-31:
> Ahhh right XD
> 
> I think we can release a new SpinalHDL version. Seems good to you ?

It's ok, in new release the project can depends on spinal.tester only. so that no extra scalatest import is required.

Readon, 2023-11-02:
i noticed that and just updated the msys-installer.

Readon, 2023-11-04:
It seems caused by scala 2.12.13, after upgrade to 2.12.18, it works. So the lib version below 1.9.3 would cause this error.

closed 2023-11-04 by Readon