SpinalHDL/SpinalHDL #1233: how to drive inout IO ?for example i2c

Mido-sun, 2023-11-02:
val sda =inout(Analog(Bool())) 

when tx data , sda := xxxx 
when the ack back ,   I need to judge "when(sda===False )"

can I just drive inout like that , the sim seem what  I want

andreasWallner, 2023-11-02:
I'd say that the "normal" way it to have a `TriState` signal in your toplevel component, which you drive accordingly from the testbench. (speaking for SpinalSim, other test environments can - depending on the simulator used) drive `Z`.

Then use `InOutWrapper` during generation of you hardware to automatically convert `TriState` signals to inout analog.

PS: I have a small utility [here](https://github.com/andreasWallner/spinalStuff/blob/master/src/main/scala/andreasWallner/sim/package.scala#L123) to simulate pullup and verify that there are no driver collisions, etc. - haven't upstreamed it yet since the multi-bit version doesn't work yet, but maybe I should anyways... (just the bool version)

Mido-sun, 2023-11-03:
I will try  thanks