SpinalHDL/SpinalHDL #1234: XSim on Windows : call vivado without bash

oletf, 2023-11-03:
Greetings, 

I'm currently trying to get into SpinalHDL,
and to use XSim simulation on Windows.

In the test project's `Config.scala` I changed the `def sim = SimConfig.withConfig(spinal).withFstWave`  to `SimConfig.withConfig(spinal).withXSim.withWave.withXilinxDevice("xc7a35tftg256-2")`.

It complains [at this point](https://github.com/SpinalHDL/SpinalHDL/blob/7314a3d140b89c5ddc1f8d46274c28c36b98af46/sim/src/main/scala/spinal/sim/XSimBackend.scala#L212-L216) because I don't have any `sh` in my Path.

I was wondering if it would be possible to make the XSimBackend call the `vivado` command without calling `sh -c [...]` (or at least add a setting to opt-out), 
although I have no idea if it would cause more problems afterwards in the case where other shell based calls or tools are needed in the current state of the simulation backend, 
so feel free to close if it would need too many changes.

When launching simulation from within Windows GUI, Vivado seems to copy an `xsimk.exe` in the project's sim subdir,
and then calls it with `xsim.dir/testbench_archname/xsimk.exe -simmode gui -wdb testbench_archname.wdb -simrunnum 0 -socket 57473`.
 
Thanks forward

Readon, 2023-11-03:
May be you can use sbt in msys2 environment. And share Vivado related environment variables from windows.

Readon, 2023-11-03:
This might also be solved through the scala's process helper. It can be used as below:
``` scala
import scala.sys.process._

object Test extends App {
    val cmd = "ls"
    val exitCode = cmd.!
}
```

oletf, 2023-11-06:
Thanks a lot for the hint,
indeed going through msys2 seems to get the process further !

I still get a problem which seems to be related to Vivado xsi interface path... here's my log:
```
sbt:projectname> runMain projectname.MyTopLevelSim
[info] running (fork) projectname.MyTopLevelSim
[info] [Runtime] SpinalHDL v1.9.4    git head : 270018552577f3bb8e5339ee2583c9c22d324215
[info] [Runtime] JVM max memory : 8152,0MiB
[info] [Runtime] Current date : 2023.11.06 12:57:20
[info] [Progress] at 0,000 : Elaborate components
[info] [Progress] at 0,383 : Checks and transforms
[info] [Progress] at 0,475 : Generate Verilog
[info] [Done] at 0,568
[info] [Progress] Simulation workspace in C:\Users\user\Documents\SpinalHDLProjects\ProjectTest\.\simWorkspace\MyTopLevel
[info] [Progress] XSIM compilation started
[error] Could not load XSI simulation shared library (xsim.dir/MyTopLevel_behav/xsimk.dll): Failed to load shared library "xsim.dir/MyTopLevel_behav/xsimk.dll". Specified module could not be found.
[error] Exception in thread "main" java.lang.Exception: ISim engine error: Failed to Load up XSI.
[error]         at spinal.sim.xsi.JNIXSIIfaceJNI.new_XSIIface(Native Method)
[error]         at spinal.sim.xsi.XSIIface.<init>(XSIIface.java:40)
[error]         at spinal.sim.XSimBackend.getInterface(XSimBackend.scala:327)
[error]         at spinal.sim.SimXSim.<init>(SimXSim.scala:9)
[error]         at spinal.core.sim.SpinalSimConfig$$anon$8.newSimRaw(SimBootstraps.scala:1118)
[error]         at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:587)
[error]         at spinal.core.sim.SimCompiled.doSim(SimBootstraps.scala:557)
[error]         at projectname.MyTopLevelSim$.delayedEndpoint$projectname$MyTopLevelSim$1(MyTopLevelSim.scala:7)
[error]         at projectname.MyTopLevelSim$delayedInit$body.apply(MyTopLevelSim.scala:6)
[error]         at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error]         at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[error]         at scala.App.$anonfun$main$1$adapted(App.scala:80)
[error]         at scala.collection.immutable.List.foreach(List.scala:431)
[error]         at scala.App.main(App.scala:80)
[error]         at scala.App.main$(App.scala:78)
[error]         at projectname.MyTopLevelSim$.main(MyTopLevelSim.scala:6)
[error]         at projectname.MyTopLevelSim.main(MyTopLevelSim.scala)
```

The problem might be caused by the fact that the `xsim` folder is created in `simWorkspace/MyTopLevel/` where the `xsim.dir` folder with `xsimk.dll` that the backend tries to load is created by Vivado in the root of the project.
I wasn't able to figure out if it's a part of Vivado not getting the correct cwd at some point, or an absolute path setting somewhere in some scala/tcl call.

Here's the tree of my project folder after calling `runMain projectname.MyTopLevelSim` from sbt in msys2 shell:
```
.
├── build.sbt
├── hw
├── project
├── README.md
├── simWorkspace
│   └── MyTopLevel
│       ├── rtl
│       └── xsim
├── target
├── tmp
└── xsim.dir
    ├── MyTopLevel_behav
    │   └── xsimk.dll
```

I can provide more info if it can help figure out the problem, but I'm not used enough to both Scala and the intricacies of Vivado tcl scripting yet, so I don't really know where to look.

Thanks forward

oletf, 2023-11-06:
I tried to switch [msys2 environment](https://www.msys2.org/docs/environments/) from MINGW64 to UCRT64, 
the error in the log happens in a different place, 
but cannot really figure out if it's better or not : 
```
[error] Exception in thread "main" java.lang.UnsatisfiedLinkError: C:\Users\user\Documents\SpinalHDLProjects\ProjectTest\simWorkspace\MyTopLevel\xsim\spinal_xsim.dll: a dynamic link library (dll) initialization routine failed
[error]         at java.base/jdk.internal.loader.NativeLibraries.load(Native Method)
[error]         at java.base/jdk.internal.loader.NativeLibraries$NativeLibraryImpl.open(NativeLibraries.java:388)
[error]         at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:232)
[error]         at java.base/jdk.internal.loader.NativeLibraries.loadLibrary(NativeLibraries.java:174)
[error]         at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2389)
[error]         at java.base/java.lang.Runtime.load0(Runtime.java:755)
[error]         at java.base/java.lang.System.load(System.java:1953)
[error]         at spinal.sim.XSimBackend.compileDriver(XSimBackend.scala:323)
[error]         at spinal.sim.XSimBackend.<init>(XSimBackend.scala:335)
[error]         at spinal.core.sim.SpinalXSimBackend$Backend.<init>(SimBootstraps.scala:446)
[error]         at spinal.core.sim.SpinalXSimBackend$.apply(SimBootstraps.scala:488)
[error]         at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:1115)
[error]         at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:934)
[error]         at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:897)
[error]         at projectname.MyTopLevelSim$.delayedEndpoint$projectname$MyTopLevelSim$1(MyTopLevelSim.scala:7)
[error]         at projectname.MyTopLevelSim$delayedInit$body.apply(MyTopLevelSim.scala:6)
[error]         at scala.Function0.apply$mcV$sp(Function0.scala:39)
[error]         at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[error]         at scala.App.$anonfun$main$1$adapted(App.scala:80)
[error]         at scala.collection.immutable.List.foreach(List.scala:431)
[error]         at scala.App.main(App.scala:80)
[error]         at scala.App.main$(App.scala:78)
[error]         at projectname.MyTopLevelSim$.main(MyTopLevelSim.scala:6)
[error]         at projectname.MyTopLevelSim.main(MyTopLevelSim.scala)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
```

oletf, 2023-11-06:
retrying gives me this error : 
```
[info] [Progress] Simulation workspace in C:\Users\user\Documents\SpinalHDLProjects\ProjectTest\.\simWorkspace\MyTopLevel
[info] [Progress] XSIM compilation started
[error] Nonzero exit code returned from runner: -1073741819`
```

Converting error code `-1073741819` back to hex gives `C0000005` which by googling might be referring to [this error](https://stackoverflow.com/questions/17168982/exception-error-c0000005-in-vc)

oletf, 2023-11-06:
The only way I managed to reach the `UnsatisfiedLinkError` was [when I tried UCRT64](https://github.com/SpinalHDL/SpinalHDL/issues/1234#issuecomment-1794900359), 
and didn't get any other error than the one previously shared [with MINGW64 env](https://github.com/SpinalHDL/SpinalHDL/issues/1234#issuecomment-1794756272).

I delete all the simWorkspace as well as the `xsim.dir` created in the root of my project directory between each run  if that's what you call cache folders, I might be missing others.

So this last error feels weird to me, because the file it's saying cannot be found does actually exist after the error is logged
```
[error] Could not load XSI simulation shared library (xsim.dir/MyTopLevel_behav/xsimk.dll): Failed to load shared library "xsim.dir/MyTopLevel_behav/xsimk.dll". Specified module could not be found.
```

oletf, 2023-11-06:
OK I found a dumb error on my side :
the dll which JNI couldn't find wasn't `xsimk.dll` itself but one of its Vivado dll dependencies whose folder wasn't even in my Path : I've run `ldd` on it and it gave me the obvious answer...

notably the solution was to add `D:\Xilinx\Vivado\2023.1\lib\win64.o` to my Path envvar...

Thanks for the support, and sorry for the time I took to realize this !

oletf, 2023-11-06:
Still not sure everything goes as it should,
no idea why the `xsim.dir` ends up in my project root directory instead of in the `simWorkspace/MyTopLevel/xsim`...

If I understand correctly, it's supposed to be copied there, and above are mentioned `compile` and `elaborate` which I cannot seem to find anywhere...

oletf, 2023-11-08:
I'm willing to try to debug the problem on my side,
mainly planning to try to figure out at which step the process breaks in Vivado : either if it's :
- somewhere in [`XSimBackend.scala`](https://github.com/SpinalHDL/SpinalHDL/blob/v1.9.4/sim/src/main/scala/spinal/sim/XSimBackend.scala)
- a Vivado tcl specific problem
- or a Vivado relative path being lost between the msys2 shell and the cmd calls

I'm just missing the proper way to debug this part of the SpinalHDL lib,
since the example Project gets the latest version from the Maven repo, in which I can't set breakpoints as it's a compiled jar,
and I'm new to the whole Scala lib dependencies and SBT process.

Thanks forward

andreasWallner, 2023-11-08:
@oletf 
There are basically a few ways to do this:
- Clone current SpinalHDL dev, put a simple testcase into it's code and debug it from there (since you'd be in the SpinalHDL codebase it's easy to set a breakpoint)
- Use an editor that fetches the sources and allows setting breakpoints: IntelliJ for example does that and I can set a breakpoint in SpinalHDL core even though I'm depending on the release version
- Instead of depending on the released version, clone the project to disk and depend on the local checkout, a build.sbt could look like:
```
ThisBuild/version := "0.1"
ThisBuild/organization := "andreasWallner"
ThisBuild/scalaVersion := "2.13.6"

lazy val spinalHdlIdslPlugin = ProjectRef(file("../SpinalHDL"), "idslplugin")
lazy val spinalHdlSim = ProjectRef(file("../SpinalHDL"), "sim")
lazy val spinalHdlCore = ProjectRef(file("../SpinalHDL"), "core")
lazy val spinalHdlLib = ProjectRef(file("../SpinalHDL"), "lib")

lazy val spinalStuff = (project in file(".")).dependsOn(spinalHdlIdslPlugin, spinalHdlSim, spinalHdlCore, spinalHdlLib)
scalacOptions += (spinalHdlIdslPlugin / Compile / packageBin / artifactPath).map { file =>
  s"-Xplugin:${file.getAbsolutePath}"
}.value

libraryDependencies ++= Seq(
  "org.scalatest" %% "scalatest" % "3.2.5" % "test",
)

fork := true
```
can be a bit finicky, especially if you are not using the default version that SpinalHDL builds for.

Hope that helps - If you have any issues you're also quite welcome to join us in the gitter chat, for small setup issues it's normally quite quick to get help.

oletf, 2023-11-09:
Thanks a lot for this reply, definitely seems to have all the info I was missing, 
Don't really know when I'll start to debug this yet, but definitely will come back to report here once I'll have figured the problem on my side !

oletf, 2023-11-14:
Alright, so I've tried to debug the thing, 
took some time to get familiar with the vivado xsimk dll calls and all.

- I originally wanted to have the xsim.dir folder not copied to project root as it's [currently done here](https://github.com/SpinalHDL/SpinalHDL/blob/13dc1926658b9ff404e7d0733b0e60e75ffe2c1d/sim/src/main/scala/spinal/sim/XSimBackend.scala#L239) by using the xsim `-xsimdir` argument which I've tested by [setting it up here](https://github.com/SpinalHDL/SpinalHDL/blob/348767481028d3cb03a4b84e3dc9b4c6aaaa44b6/sim/src/main/resources/XSIIface.cpp#L56). 
  - from the Scala Spinal simulation it goes either 
    - well with no error nor exception, but the wdb which is (somehow) successfully produced is too small and Vivado can't even open it (I assume simk fails finding some relative path somehow, but never complains about it)
    - or there's a crash in native `librdi_simulator_kernel.dll` code
- so in the end I kept the `xsim.dir` copied to project root and only changed a few things :
  - I realized that the only moment we need to use Msys2 shell is when compiling the xsim kernel to get its `g++`
  and all other calls are only `cmd /c`
  - so as long there are the following in the Windows PATH, all works fine :
    - `%VIVADO_HOME%\bin` : to [find vivado](https://github.com/SpinalHDL/SpinalHDL/blob/13dc1926658b9ff404e7d0733b0e60e75ffe2c1d/sim/src/main/scala/spinal/sim/XSimBackend.scala#L211)
    - `%VIVADO_HOME%\lib\win64.o` : for the librdi dlls (`xsimk.dll` dep)
    - `%MSYS2_ROOT%\mingw64\bin` : for its `libstdc++` (`spinal_xsim.dll` dep)
 
So the `xsim.dir` stays in the project root until the next sim run (technically it's needed until the end of the simulation, but since deleting it after the change would need to be called after the `doSim` closure).

Finally I moved the wdb file to where `spinal_xsim.dll`'s location, as opening a wdb in Vivado makes it create all its `log`, `jou` and `.Xil`, so it doesn't clutter the project root.

I can make a PR if you're interested, 
I've only tested with scala `2.12.18` on Windows 10, so needs testing on Linux to make sure I didn't break anything, and xsimk.dll seems very sensitive to the cwd it's being run in.

oletf, 2023-11-14:
here's my current change
https://github.com/oletf/SpinalHDL/commit/08a8cef77e684afc8be0e24e84e74d2e0bb8089c

open to any other suggestions!

oletf, 2023-11-14:
re-Reading https://github.com/SpinalHDL/SpinalDoc-RTD/pull/230 now,
I realize that it'd be possible to use `${env:VIVADO_HOME}/bin/setEnvAndRunCmd.[bat|sh]` from the Process calls directly so that :
- windows user only needs to have `%MSYS2_ROOT%\mingw64\bin` in their path as mentioned above
- linux user doesn't need to source settings64 anymore

Readon, 2023-11-14:
> re-Reading [SpinalHDL/SpinalDoc-RTD#230](https://github.com/SpinalHDL/SpinalDoc-RTD/pull/230) now, I realize that it'd be possible to use `${env:VIVADO_HOME}/bin/setEnvAndRunCmd.[bat|sh]` from the Process calls directly so that :
> 
> * windows user only needs to have `%MSYS2_ROOT%\mingw64\bin` in their path as mentioned above
> * linux user doesn't need to source settings64 anymore

I did not tested that command script yet. However, msys2 could not use settings64.sh because of the path inside it，

andreasWallner, 2023-11-15:
Thanks for looking into it!

Ad running the script in MSYS: Is there a setting file for a linux shell contained in the windows installation? From what I remember from last time there was none?

I'm kind of wondering if we could get rid of the calls to some shell here, and just call the binaries directly...
I can offer to test your changes on linux, but that'll have to wait until tomorrow - it's quite late here already.

The common directory being used opens up one point though that we should document: the build.sbt should then not have `Test/testForkedParallel := true` configured I guess, otherwise the forks would all end up using the same directory?

oletf, 2023-11-15:
What I had in mind is to use native OS shell (and hence also native PATH) as much as possible,
so [currently in my latest change](https://github.com/oletf/SpinalHDL/commit/08a8cef77e684afc8be0e24e84e74d2e0bb8089c) :
- on linux : shouldn't have changed anything : either needs : 
  - source Vivado's `$VIVADO_HOME/settings64.sh`
  - or adding `$VIVADO_HOME/bin` to path manually as [mentioned here](https://github.com/SpinalHDL/SpinalDoc-RTD/blob/445e9eb0e40e6cc06a9aebfe2478e464c21cb868/source/SpinalHDL/Simulation/simulator_specifics.rst?plain=1#L80)
- on windows : since all calls are either 
  - `cmd /c` for everything that can be run natively (all vivado-related `.bat`)
    - [vivado project creation](https://github.com/oletf/SpinalHDL/blob/08a8cef77e684afc8be0e24e84e74d2e0bb8089c/sim/src/main/scala/spinal/sim/XSimBackend.scala#L234)
    - [running `compile` and `elaborate`](https://github.com/oletf/SpinalHDL/blob/08a8cef77e684afc8be0e24e84e74d2e0bb8089c/sim/src/main/scala/spinal/sim/XSimBackend.scala#L246)
  - msys2 mingw64 shell for its `g++` to build `spinal_xsim.dll`
    - which doesn't actually need any Vivado in PATH, as all needed includes are copied where needed
      - [`jniIncludes` from `JAVA_HOME`](https://github.com/oletf/SpinalHDL/blob/08a8cef77e684afc8be0e24e84e74d2e0bb8089c/sim/src/main/scala/spinal/sim/XSimBackend.scala#L268)
      - [`xsimIncludes` from `VIVADO_HOME`](https://github.com/oletf/SpinalHDL/blob/08a8cef77e684afc8be0e24e84e74d2e0bb8089c/sim/src/main/scala/spinal/sim/XSimBackend.scala#L276)
---
and what I thought when writing the previous message was that we could actually use :
- on linux : `doCmd` calls through `$VIVADO_HOME/bin/setEnvAndRunCmd.sh` where Vivado path needed
- on windows : `doCmd` calls through `%VIVADO_HOME%/bin/setEnvAndRunCmd.bat` (that the only call to msys2 shell that is done doesn't need)

although the shell calling `sbt` will still actually need to have `%VIVADO_HOME%\lib\win64.o` (and linux's equivalent) in path to find `librdi_[simbridge|simulator]_kernel.[dll|so]` needed by `xsimk.[dll|so]` (see next part for details)

---

> I did not tested that command script yet. However, msys2 could not use settings64.sh because of the path inside it，

> Ad running the script in MSYS: Is there a setting file for a linux shell contained in the windows installation? From what I remember from last time there was none?

So this (if I didn't miss anything) isn't a problem, in both cases where sbt is run from `cmd` or (any) `msys2` shell.

> I'm kind of wondering if we could get rid of the calls to some shell here, and just call the binaries directly...

msys2 will need to be called by its shell to get its `g++` env, but Vivado cmd could probably be called directly through  `%VIVADO_HOME%/bin/setEnvAndRunCmd.bat`.

> The common directory being used opens up one point though that we should document: the build.sbt should then not have Test/testForkedParallel := true configured I guess, otherwise the forks would all end up using the same directory?

As all `Process` calls are only done through the [`doCmd` method](https://github.com/oletf/SpinalHDL/blob/08a8cef77e684afc8be0e24e84e74d2e0bb8089c/sim/src/main/scala/spinal/sim/XSimBackend.scala#L112) to which all calls have the `cwd` arg being relative paths to project root, if it is not a problem in the current version, it doesn't change.

The only constraints relative to sbt (and hence java process)'s `cwd` (project root where build.sbt is located) :
- was : for the `wdb` being produced which [I moved there](https://github.com/oletf/SpinalHDL/commit/08a8cef77e684afc8be0e24e84e74d2e0bb8089c#diff-3d59e8e0118d8606eb948bd13b70b83d33b1ad056301212de48b9407ff18a083) from *project root* to *where `spinal_xsim.dll` is built* to avoid Vivado cluttering project root when wdb is being opened in gui
- is still : that when [`XSIIface`](https://github.com/SpinalHDL/SpinalHDL/blob/348767481028d3cb03a4b84e3dc9b4c6aaaa44b6/sim/src/main/resources/XSIIface.cpp#L45) is being [instantiated here](https://github.com/SpinalHDL/SpinalHDL/blob/13dc1926658b9ff404e7d0733b0e60e75ffe2c1d/sim/src/main/scala/spinal/sim/XSimBackend.scala#L326-L328) by [SimXSim](https://github.com/SpinalHDL/SpinalHDL/blob/13dc1926658b9ff404e7d0733b0e60e75ffe2c1d/sim/src/main/scala/spinal/sim/SimXSim.scala#L9) from `spinal_xsim.dll` : 
  - it inherits the current sbt (java)'s cwd (which is project root) which loads it
  - which in turns loads `xsimk.dll` from relative path `"xsim.dir/${toplevelName}_behav/xsimk.dll"` (generated by [`elaborate`](https://github.com/SpinalHDL/SpinalHDL/blob/13dc1926658b9ff404e7d0733b0e60e75ffe2c1d/sim/src/main/scala/spinal/sim/XSimBackend.scala#L235))  : which is the **only** reason why copying `xsim.dir` to the project root is needed.

---
I have to admit that's a lot of stuff going on there, 
(understanding that we got a *scala calling shell scripts to build a dylib to load it in its own java process as native code* was very fun to figure out)
it's pretty clear in my head now, 
but not sure the way I transcribed it here is clear enough,
please ask for clarification on any point if needed !

andreasWallner, 2023-11-16:
First off, I tested you version on linux and the simulation worked as expected.

Yes, `settings64.sh` does not set up any `LD_LIBRARY PATH` , so I had to do that for `librdi...` to be found.

> calls through `$VIVADO_HOME/bin/setEnvAndRunCmd.[sh|bat]`

I would see that as an improvement, less stuff to go wrong in the end (if we do that then we could on linux already inject the correct `LD_LIBRARY_PATH=$VIVADO_HOME/lib/lnx64.o`, so the user would only have to set `VIVADO_HOME`
If we do this change then we could also use the version of `Process` that takes a `Seq[String]` for the call, completely getting rid of the need to escape paths, etc.

> msys2 will need to be called by its shell to get its g++

Sure, but for Vivado those are not needed as you rightly said.

> As all Process calls are only done through the [doCmd method](https://github.com/oletf/SpinalHDL/blob/08a8cef77e684afc8be0e24e84e74d2e0bb8089c/sim/src/main/scala/spinal/sim/XSimBackend.scala#L112) [...]

I meant to say that the this would also create an issue in the current version: `Test/testForkedParallel := true` gets sbt to run tests fully in parallel, so the files that are generated in the `xsim.dir` in the root directory could be overwritten while in use?

I'd say go ahead with a PR, only minor early feedback: in places where you now used assert and then have to provide some empty strings that will actually never be used you should IMHO just throw an exception.

oletf, 2023-11-16:
> I meant to say that the this would also create an issue in the current version: `Test/testForkedParallel := true` gets sbt to run tests fully in parallel, so the files that are generated in the `xsim.dir` in the root directory could be overwritten while in use?

So I guess that means the problem already exists already without my change then ?
I didn't consider tests when doing this, I was only focused on "getting a single Vivado simulation run" as I'm new to both Scala as a language, sbt as well as Spinal codebase, so I lack awareness of more global interactions. 

I don't know how the spawning of the whole `compile` and `doSim` are currently managed, 
and changing the cwd of a process or even for a single thread spawning isn't possible.
But I realized that if there was a way to have the java process which [loads `XSIIface`](https://github.com/SpinalHDL/SpinalHDL/blob/bed4bc0127e606fd54333a7e45ed51a59f6437c9/sim/src/main/scala/spinal/sim/XSimBackend.scala#L327) (so I guess both in [`SimBootstrap` here](https://github.com/SpinalHDL/SpinalHDL/blob/bed4bc0127e606fd54333a7e45ed51a59f6437c9/core/src/main/scala/spinal/core/sim/SimBootstraps.scala#L1115) and [`TestXSim` here](https://github.com/SpinalHDL/SpinalHDL/blob/bed4bc0127e606fd54333a7e45ed51a59f6437c9/sim/src/test/scala/spinal/sim/TestXSim.scala#L18)) to be spawned as a new process (for it to be able to have its cwd changed, to satisfy `xsimk.[dll|so]`'s relative path to `xsim.dir` requirement) :
the copying of `xsim.dir` to project root would become unnecessary and could also help on the fact that tests are parallelized,
but I acknowledge that the refactor needed could be big, and I wouldn't be ready to try it myself.

Also I'm not sure exactly why the crash in the librdi dll when trying to give the `xsimdir` arg to `xsim` fails,
but getting it to work would be the cleanest way to do this and could also solve the parallelized test problem, 
so I might look again into it if I get some free time.

> I'd say go ahead with a PR, only minor early feedback: in places where you now used assert and then have to provide some empty strings that will actually never be used you should IMHO just throw an exception.

I literally copy pasted the `vivadoInstallPath` one which was [already there](https://github.com/SpinalHDL/SpinalHDL/blob/bed4bc0127e606fd54333a7e45ed51a59f6437c9/sim/src/main/scala/spinal/sim/XSimBackend.scala#L92-L98), 
but I could change both to call exception indeed, not sure about the most idiomatic way to do this, but will give it a try!

oletf, 2023-11-16:
Also if anyone has experience in Vivado tcl scripting and cli usage,
and has already used `xelab` and `xsim` : but with the specificity of running them from a `cwd` which isn't the Vivado xsim folder (where `xsim.dir` is created by default)
and had it working, we're probably just missing the correct set of tcl/cli arguments to properly set the whole paths as relative to the sbt root workspace.

I managed to have the call to `xsim` almost work by calling it as 
```
xsim MyTopLevel_behav -log xsimlog.log -wdb ./simWorkspace/MyTopLevel/xsim/MyTopLevel.wdb -xsimdir ./simWorkspace/MyTopLevel/xsim/MyTopLevel_xsim.sim/sim_1/behav/xsim
```
but there's still a librdi dll which tries to load `xsim.dir/MyTopLevel_behav/xsim.[dbg|rtti|svtype]` when all those files are properly created in `./simWorkspace/MyTopLevel/xsim/MyTopLevel_xsim.sim/sim_1/behav/xsim/...` by all the previous steps...

That's the last missing piece to be able to get rid of the copy of `xsim.dir` and possibility to have tests running with `testForkedParallel`.

andreasWallner, 2023-11-19:
@oletf haven't looked at those tcl use-cases specifically, but in some of my other scripts I use `cd` in the script to change the working directory, this might work here as well.

You mentioned above that you wanted to use the `setEnvAndRunCmd` script - is this something that you want to change for the PR, or did you drop that idea?

> So I guess that means the problem already exists already without my change then ?

Yes, the note above was one the one hand a bit of a check whether I understood correctly that it always uses the same folder, on the other hand a note to myself that we should add that info to the documentation. ;-)

> I don't know how the spawning of the whole compile and doSim are currently managed,

Either the way that you ran it until now (via some main method) or alternatively using the scalaTest unit test framework (which can do test enumeration/filtering/etc.)

> But I realized that if there was a way to have the java process which loads XSIIface [...] to be spawned as a new process

I don't think there is a way to reliably do that. The whole compilation/running the simulation is done during test execution. While we could start a new process for that, the test code itself would still be running in the original process - so that wouldn't sadly be any good.

oletf, 2023-11-27:
> but in some of my other scripts I use cd in the script to change the working directory, this might work here as well.

the problem I noticed is that `cwd` that would need to be changed is the one of the java process that'll load the librdi....dll/so, 
so sadly wouldn't be enough :/

> You mentioned above that you wanted to use the `setEnvAndRunCmd` script - is this something that you want to change for the PR, or did you drop that idea?

I still need to fix the sbt cmd/msys shell env detection as mentioned [here](https://github.com/SpinalHDL/SpinalHDL/pull/1246#discussion_r1396087871), so I think I might as well add this !

What's the way I should use to change the PR? 
should I : 
- force push to my fork branch
- or make a merge on the PR ? 

Thanks forward

oletf, 2023-11-27:
ok so I just pushed a new commit on my fork, PR seems to have updated it,

andreasWallner, 2023-11-28:
> What's the way I should use to change the PR?
I normally force push to my fork to have one or more meaningful commits in the PR - but we are not very strict on that ATM ;-)

> will need another round of test
Will do that in the evening.

> just need to keep in mind that %VIVADO_HOME%\lib\win64.o still needs to be in path
We should be able to modify the PATH for the call so that it's in there for sure? (but if you prefer we don't have to do that in this PR)

Is the PR then OK from your side or is there anything else you wanted to look at? I lost track a bit, sry.

oletf, 2023-11-28:
> Is the PR then OK from your side or is there anything else you wanted to look at?

PR is up to date on my side !
there's nothing much I had in mind to add, 
the only problem still being the librdi fixed relative path which is not our job to fix, which is the last reason why we need to copy `xsim.dir` to project root.

> We should be able to modify the PATH for the call so that it's in there for sure? (but if you prefer we don't have to do that in this PR)

`%VIVADO_HOME%\lib\win64.o` in PATH is not needed for any cmd nor shell call,
it needs to be in PATH from the pov of the java process which runs SBT, because that's where the `spinal_xsim.dll` is loaded, which in turn loads the `librdi` stuff...
so unless Java process has a way to change its own PATH Env Var at runtime, 
it'll probably be easier to make sure user has both `VIVADO_HOME` set, *and* `%VIVADO_HOME%\lib\win64.o` in PATH.

>  I lost track a bit, sry.

yeah there's lots of stuff involved,
and after a week of pause coming back to it was a little hard for me too, no worries :D

Dolu1990, 2023-12-29:
Hi ^^

What is the status on this issue ?

Readon, 2023-12-31:
> Hi ^^
> 
> What is the status on this issue ?

Wait for #1246 be tested.
Remains:
1. on some Windows environment it still report "UnsatisfiedLinkError".
2. ~~I suggest to add xcix file in test to keep it test by CI. (This can be ignored on CI due to the CI itself is on Ubuntu OS.)~~

oletf, 2024-01-10:
>     1. on some Windows environment it still report "UnsatisfiedLinkError".

Could you please check what `ldd` gives on `xsim.dir/[TopLevel]/xsimk.dll` on the environment which gives this error ?
So we can figure out which dll could be missing in the PATH ?

Readon, 2024-01-22:
> > ```
> > 1. on some Windows environment it still report "UnsatisfiedLinkError".
> > ```
> 
> Could you please check what `ldd` gives on `xsim.dir/[TopLevel]/xsimk.dll` on the environment which gives this error ? So we can figure out which dll could be missing in the PATH ?

So I find the problem. it still requires to put the MINGW64's binary path into Windows's PATH variable.

closed 2024-01-24 by oletf