SpinalHDL/SpinalHDL #1236: 『Requirement And Idea』SpinalHDL generated verilog code parameterization and generate block simplification requirements.

dreamflyings, 2023-11-05:
Hi~ 
SpinalHDL language itself has good generalization and maintainability,
But the quality of the verilog code it generates is not high, the maintainability is poor, and it is difficult to meet the verilog code delivery requirements.
For most verilog designers, integrating some large-scale spinalHDL generated verilog is a huge challenge, and a large amount of redundant code is also inconvenient for verilog developers to maintain, which is very disadvantageous for the continuous promotion of spinalHDL language.
In my actual engineering applications, I often need reasonable module division, and use more general small module design, in order to greatly reduce the amount of verilog code generated. But even so, the amount of verilog code generated is still huge, especially when designing register configuration module and so on.
For the quality of SpinalHDL generated verilog code, I hope to be able to greatly improve.
My understanding is that there are two ways to achieve this goal:
1. Modify the existing logic of generating verilog code;
2. Only regard the generated verilog code as a string, and let the user customize it by a similar BlockBox method.
For the first point, I think it may be more difficult, the development cycle is long, I suggest to improve gradually, and finally achieve the goal.
For the second point, I think the feasibility is very high. My understanding is: implement or extend BlockBox + inline verilog method, provide more generation parameterization, multi-dimensional array and generate block auxiliary functions; provide a framework for mapping spinalHDL related logic to strings (component interface signal printing, area logic printing, logic that meets the requirements printed into arrays and generate blocks, tags that meet the requirements mapped to parameterized code, such as important signal width)
In addition, the verilog mixed spinalHDL code string splicing design in the second point does not necessarily have to perform SpinalHDL related checks. Because this is only based on verilog stringization, and the embedded SpinalHDL printing code can be verified separately, and finally the mixed verilog plus spinalHDL code can generate the final verilog and simulate verification.
Thank you very much🙏 I hope SpinalHDL can do better and stronger.

Dolu1990, 2023-11-06:
Hi,

Here are my thought, note that it doesn't mean i do not agree, it can mean i would need some practical explanation of the cases which aren't good.

> the maintainability is poor

?

> But the quality of the verilog code it generates is not high

Does it matter ? I mean, by analogy, when you code C code, you don't realy need to look at the assembly it generates, unless you coded your whole life in assembly and are doing a transition.

> and it is difficult to meet the verilog code delivery requirements.

I'm not familiar with that world.
Is the issue that the verilog code delivery requirements are verilog centric / cursed ?
Or there there is something on the SpinalHDL side which is fundamentaly not good ?

> For most verilog designers, integrating some large-scale spinalHDL generated verilog is a huge challenge, and a large amount of redundant code is also inconvenient for verilog developers to maintain,

Could you explaine a bit more ?
Is the issue that mixing verilog / SpinalHDL code is a paine ?
What kind of project is it ?
From my perspective, i just do everything in SpinalHDL, so my flow i quite "clean".

> But even so, the amount of verilog code generated is still huge, especially when designing register configuration module and so on.

Could you give some example ?

> For the quality of SpinalHDL generated verilog code, I hope to be able to greatly improve.

From my perspective, i don't realy see / fell an issue there, but that maybe due to me working on different scale projects.

Can you provide some example / data ?

Thanks :)

dreamflyings, 2023-11-06:
> Hi,
>
> Here are my thought, note that it doesn't mean i do not agree, it can mean i would need some practical explanation of the cases which aren't good.
>
>
> ?
>
>
> Does it matter ? I mean, by analogy, [...]

@Dolu1990 
Thank you for your prompt reply! 
I raised this question based on the basic judgment that verilog/vhdl will still be the mainstream for a considerable period of time, and it is difficult to persuade those “old antiques” to adopt a new hardware design process. They have developed a set of verilog coding standards very early (as a latecomer, I can only comply); in addition, the greater resistance is those peers who do not know how to use and do not want to learn SpinalHDL (imagine, most of them did not learn advanced hardware design languages in school, many have family pressure, and there are several times or ten times more project output less than SpinalHDLer, do they have the willingness to integrate and modify the verilog code generated by SpinalHDL? They will not have the willingness to use the SpinalHDL code generator either, they do not want to maintain or modify it), these are just to the point. Using verilog to analogy assembly, maybe it can’t be simply analogized, because after C language came out, assembly language exited the historical stage very quickly, and software-minded people obviously are not used to the anti-human assembly language; verilog is different, although it is very bad, but it is small and refined, it has a low level of abstraction, but it can also meet the productivity requirements, and it can be used reluctantly. 
Does the SpinalHDL language itself have major flaws? I don’t think so! The language itself has a reasonable underlying design and the generated code is readable The biggest motivation for me to raise this question is: SpinalHDL language can be compatible with verilog language, if verilog language can also be compatible with SpinalHDL (string form compatibility), then we will have a strategic advantage, no matter how the future develops, the initiative will be in our hands. 
Why did I raise this question now? Mainly because, before I did FPGA design, now I do large-scale heterogeneous NOC (network on-chip) chip design (before I did not optimize the verilog code generated by SpinalHDL, the verilog code size reached more than 200 megabits, after optimization less than ten megabytes), the quality requirements for verilog code are different, and no one has the ability to maintain and modify my SpinalHDL code, so I simply only deliver verilog. How do you use SpinalHDL to develop hardware in a low-key way? I mainly divide the modules reasonably, only use Area under appropriate conditions, and use Component/Module in most cases; some modules use BlackBox + inline verilog code form; there are also some small SpinalHDL codes, first generate verilog code, and then throw it to GPT4, let it parameterize and simplify the verilog code (it can be fake [smiling face] ) haha, unfortunately there are still some limitations and errors. 
Is there any redundant generated verilog reference code available? The code is on the server, it may not be convenient to download, but I will sort it out when I have time and give some examples. 
Is it necessary to greatly improve the quality of the generated verilog code? If it is put in a larger scope, I think even if only looking at the code size, it is necessary to compress it within a reasonable range; there is a basic judgment here that verilog/vhdl will still be the mainstream in the short and medium term in many hardware design processes and eda support. In addition, better verilog code quality is more conducive to the large-scale use of SpinalHDL; imagine: a mid-to-high-level verilog engineer, who knows a little scripting language such as tcl and Python, now he does not need to learn the complex syntax of SpinalHDL, various libraries and frameworks, he only needs to regard SpinalHDL as a scripting language, learn a little bit of knowledge, and he can improve the efficiency of writing verilog code (this is the main thing), and he can also better integrate other people’s SpinalHDL code and some simple spinalHDL libraries, they have no reason not to learn a little fur. 
That’s all for now, I will also study the feasibility, thank you 🙏

Dolu1990, 2023-11-07:
> They have developed a set of verilog coding standards very early (as a latecomer, I can only comply); in addition, the greater resistance is those peers who do not know how to use and do not want to learn SpinalHDL 

Yes, i understand

> many have family pressure

Yes, enough worries on all the sides to not add another layer. I can feel paine

> Using verilog to analogy assembly, maybe it can’t be simply analogized

I meant from a purly technical point of view, but the historical aproache you have is also right. Also yes, verilog isn't as bad as asm.

> Only regard the generated verilog code as a string, and let the user customize it by a similar BlockBox method.

I'm not exactly sure what you meant by it.
Could your provide an example of fictive API / example of what you mean ?

> string splicing design

Do you mean generating verilog code which mix SpinalHDL design and verilog design ?

> the verilog code size reached more than 200 megabits

Hooo that's big.
It make me thing of two things : 
1) Either at some place, the way the SpinalHDL API is used isn't the best. For instance indexing a Vec using a UInt literal, which can very easly generate much unecessary verilog. Another example is using Vec instead of Mem. Another example is when you have array of registers to keep track of things happening, done "wrongly" using Bundle can scale up realy fast, here is an template : https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Examples/Advanced%20ones/slots.html
2) Either the design is just big, in such case, i was thinking that maybe, for such case, we could have some API to help blackboxifying SpinalHDL code itself, allowing easy multi stage elaboration.

So far i didn't personaly had issue, mostly because the biggest design i had was multi core NaxRiscv, but any feedback on what design patterns are generating too much verilog is welcome.
What did you optimized specificaly to generate smaller verilog ?

> when I have time and give some examples.

Thanks :D

So, overall i would say that i'm not personnaly exposed to most of the issue you are facing, so it is very necessary to open github issue when something not good is happening, so we can work on it and fix it :)

andreasWallner, 2023-11-07:
Maybe one small thing to add here w.r.t. generated code readability: we had an investigation here where it was decided that we should build our own HDL instead of using Spinal/Chisel/nMigen because of certification: to do certification with the SpinalHDL code directly would essentially mean that the SpinalHDL generator is trusted and would need to be certified beforehand (which is not gonna happen ;-)).
The solution that was thought up here is that the generator should output code in such a way that the Verilog can be certified directly. This is also something that the current SpinalHDL can't really do (and it's also really hard to "fix" I'd say if this would ever become a design goal)

(In this case this was about safety cert for automotive btw., but security might play out very similar)

One thing that is for sure a factor is that SpinalHDL can't (as we discussed many times before) output code with generics. That means for cases where we'd in a typical design use generics to make the code smaller (post elaboration it'd still have similar size I guess) we have "duplication" with SpinalHDL.

Readon, 2023-11-08:
There may be another way to introduce a GPT-like tool that simplifies the generated verilog code through reinforcement learning, verifying its correctness with an equivalence verification tool. This depends on the definition of "code quality". If this is an impossible task for the existing SpinalHDL base architecture, then the new tool is worth a try.

However, I can't find examples of SpinalHDL generating redundant code in our internal projects right now, so perhaps some simple examples would be better understood. Or it would be nice to generate one from existing open source code as an example.

Dolu1990, 2023-11-08:
One thing, if you have two equivalent module generated, SpinalHDL will only emit the verilog of one and reuse it, but for it to work, your scala/spinalHDL elaboration need to be deterministic.
One easy way to fail it, is to use things like HashMap / HashSet and then to iterate over their elements.
The solution is to use LinkedHashMap / LinkedHashSet which preserve insertion order.

dreamflyings, 2023-11-10:
> One thing, if you have two equivalent module generated, SpinalHDL will only emit the verilog of one and reuse it, but for it to work, your scala/spinalHDL elaboration need to be deterministic. One [...]

Yes, this is how it’s done! I wrote two simple examples in a hurry. The first example is about: a few lines of register configuration SpinalHDL code, after generating verilog, the code volume expanded by hundreds of times (ps: in this example, two lines are commented out, I don’t know why they are wrong [facepalm]). The second example is about: reasonable independent implementation as Module, which can reuse related logic (this example, just to illustrate the problem, the encapsulated Module scale is not large, and the Module instantiation cannot be made into a generate block, the code volume has not been reduced much). I will add more later when I think of something, and I will think about the possible improvement schemes for these features in the next two days.
1.
```scala
import spinal.lib._
import spinal.core._
import spinal.lib.bus.misc.BusSlaveFactory
import spinal.lib.bus.simple.{PipelinedMemoryBus, PipelinedMemoryBusConfig, PipelinedMemoryBusSlaveFactory}

object RegfileConfig extends App{
  SpinalConfig().generateVerilog({
    new Module{
      val mod = new Module{
        val offset = Seq(0x0000, 0x0100, 0x1000)
        val regfile:Vec[Vec[Vec[Bits]]]  = Reg(Vec(Vec(Vec(Bits(32 bits), 16), 8), 4))
        def driveFrom(busCtrl: BusSlaveFactory) = new Area{
          for (i <- 0 until 4) {
            for (j <- 0 until 8) {
              regfile(i)(j).foreach { g =>
                busCtrl.writeMultiWord(g, address = offset(0) + i * offset(2) + j * offset(1))
              }
            }
          }
        }
      }
//      val pmb = slave(PipelinedMemoryBus(PipelinedMemoryBusConfig(addressWidth = 14, dataWidth = 32)))
//      mod.driveFrom(new PipelinedMemoryBusSlaveFactory(pmb))
      mod.rework{
        val pmb = slave(PipelinedMemoryBus(PipelinedMemoryBusConfig(addressWidth = 14, dataWidth = 32)))
        mod.driveFrom(new PipelinedMemoryBusSlaveFactory(pmb))
        pmb
      }.toIo()
    }.setDefinitionName("RegfileTest")
  })
}
```
2.
```scala
import spinal.core._
import spinal.lib._

class ComponentCopy[T <: Data](val payloadType: HardType[T], val depth: Int, mode:String = "zip") extends Component {
  val io = new Bundle {
    val input = slave(Flow(payloadType))
    val outStreams = Vec(master(Stream(payloadType)), depth)
    val inStreams = Vec(slave(Stream(payloadType)), depth)
    val willOverflow = out(Bool())
  }

  def init() = {
    io.outStreams.map(_.ready := False)
    io.inStreams.map(_.valid := False)
  }

  def builder(prev: Stream[T], left: Int): List[Stream[T]] = {
    left match {
      case 0 => Nil
      case 1 => prev :: Nil
      case _ =>
        prev :: builder(
          {
            val streamId = depth + 1 - left

            val stream = Stream(payloadType)
            val rValid = RegNextWhen(prev.valid, stream.ready) init (False)
            val rData = RegNextWhen(prev.payload, stream.ready)
            stream.valid := rValid
            stream.payload := rData
            prev.ready := stream.ready
[... 81 lines cut ...]
  val full = CountOne(readyAvailable) === 0
  io.willOverflow := this.full && io.input.valid && !io.outStreams.sExist(_.fire)
  when(this.full) {
    cachedConnections.last.ready := io.input.valid
  }.otherwise {
    val readyId = OHToUInt(OHMasking.first(readyAvailable))
    cachedConnections(readyId).ready := io.input.valid
  }

  def findFirst(condition: Stream[T] => Bool): (Bool, UInt) = {
    val (exist, reversedId) = io.outStreams.reverse.sFindFirst(condition)
    val realId = depth - 1 - reversedId
    (exist, realId)
  }
}

object ComponentCopyX_gen extends App{
  SpinalConfig(targetDirectory = "./verilog").generateVerilog({
    new ComponentCopy(UInt(32 bits), 32).setDefinitionName("ComponentCopy")
  })
  SpinalConfig(targetDirectory = "./verilog").generateVerilog({
    new ComponentCopy(UInt(32 bits), 32, mode = "zip").setDefinitionName("ComponentCopyZip")
  })
}
```

Dolu1990, 2023-11-14:
Hi,

Thanks for the examples ^^

About 1)

Got it reduce by 45% by allowing resize node to simplify themself when useless.
I pushed a fix in a branch.

One question i have is :
How the regfile is readed ? (and so, if a Mem could be used instead)

About 2)

I would say, in some ways, it is part of the paradigme, SpinalHDL do the hardware elaboration in Scala, nothing is left for the Verilog. Maybe a first step for SpinalHDL would be to add a native Array of BaseType that would be preserved in the emited verilog, i'm mostly scared of limitation of array in Verilog VHDL. that would be the kind of feature which only work in a limited number of cases.

Dolu1990, 2023-11-14:
Merged as https://github.com/SpinalHDL/SpinalHDL/pull/1242

dreamflyings, 2023-11-15:
> Hi,
>
> Thanks for the examples ^^
>
> About 1)
>
> Got it reduce by 45% by allowing resize node to simplify themself when useless. I pushed a fix in a branch.
>
> One question i have is : How the regfile is [...]

@Dolu1990  Hi~

Thank you very much!

As a first step, you pushed a patch so quickly, awesome! regfile can be replaced by mem in some cases, and I hope to do so, but unfortunately I am not very familiar with the library API.

Raising this question or idea is a summary of the recent work and the IC design specification requirements, and of course I hope that you and the community will pay enough attention to this issue.

Through multi-party communication, I summarize as follows: If the following requirements can be met, it will be of great benefit to the promotion of the SpinalHDL language. 

This requirement is: The top-level verilog code generated by the Spinal language is required to be parameterized, and the verilog code with simplified arrays and generate blocks. A constraint condition can be added: The top-level code only contains modules and wires, and does not contain complex logic operations. As mentioned above, that is, for the specified verilog "top-level" corresponding to the SpinalHDL component that requires parameterization, array and generate block simplification, it is required to contain only BlackBox, component, simple assignment statements, and area only has simple assignment statements, or interface signals with automatic pull or automatic connection.

Of course, the above requirements can be achieved by writing BlackBox and inline verilog code in SpinalHDL, but the efficiency is very low and unnatural. This is why I had the previous idea: to see if it is possible to extend BlackBox and add some auxiliary functions to improve the writing efficiency.

No matter what your plan is for the next step, whether you will continue to follow up and optimize verilog simplification, I thank you very much for running SpinalHDL as an important cause. 

I continue to look forward to the development of SpinalHDL, and of course, I hope it can develop faster [surprise]

andreasWallner, 2023-11-15:
@dreamflyings 
With regards to parameterized code output: This has been discussed a few times and the short summary is that this is a problem that is very hard to solve, if we find a solution at all. The main reason for that is that once you add generation parameters or even generate blocks then most of the checks and guarantees that SpinalHDL now gives can't be checked anymore. It also open all kinds of issues, e.g. of how to connect those components that we don't know about, which would be generated in a generate block. While for your use-case you'd be fine with not having any checks performed on that code this is not what most users expect.
In short it's unlikely to happen, and would most likely need a very dedicated contributor to pull off.

Regarding additional simplifications of the generated verilog: this is an ongoing topic, though not highly prioritized (most of the time an implementation is chosen that minimizes the risk of producing invalid HDL or HDL that doesn't match the SpinalHDL description). If this is something that have great interest in then I can only extend an invitation to you to help and improve the generated output. Projects like this survive because of enthusiastic volunteers that spend time to improve the project. None of us do this as their job, we spend our spare time to work on SpinalHDL.

dreamflyings, 2023-11-16:
> @dreamflyings With regards to parameterized code output: This has been discussed a few times and the short summary is that this is a problem that is very hard to solve, if we find a solution at all. [...]

@andreasWallner 
Well, I agree! I have tried, and also used AI parameterization, and compared, and understood some of the design difficulties. So I said before, I prefer to use SpinalHDL + inline verilog + string auxiliary functions + GPT4 to achieve it.
Thank you also for your invitation, if the plan and design maturity are reached, I will also share my own plan with everyone.

dreamflyings, 2024-05-29:
> Hi,
> 
> > The top-level verilog code generated by the Spinal language is required to be parameterized
> 
> What is the reason for this requirement ? Is it because the verilog need to be shiped to a third party which may need to paramatrize it his own way ?
> 
> > The top-level code only contains modules and wires, and does not contain complex logic operations
> 
> Is this to make it ASIC flow friendly ? where the toplevel is a collection of module which are place / route individualy and then manualy floorplanned ?

Sorry, I just saw your message (ー_ー)!! 

Thank you for your attention!

Yes, indeed, it is to make the ASIC flow more user-friendly. As it stands, the SpinalHDL language still has a way to go before it can be widely adopted. A large number of Verilog engineers and a small number of SpinalHDL engineers might be a long-term talent layout. It requires that the top-level and sub-top-level (subsystem-level) codes generated by SpinalHDL remain stable and parameterizable. In this way, changes in the code below the subsystem level and changes in the interface parameter configuration will not affect the generated top-level and sub-top-level codes (Verilog engineers only need to change the instantiation parameters of the generated Verilog modules accordingly). This approach makes the collaboration between Verilog engineers and SpinalHDL engineers more friendly (because the intermediate parameterized top-level isolates compatibility). I find that the sv interface seems to be a good attempt (however, some companies still refuse to use sv for ASIC design).