SpinalHDL/SpinalHDL #1239: Ability to specify path to make binary

Tectu, 2023-11-06:
Is there any existing mechanism (or can we add one) that would allow specifying the path to the `make` binary that is being used within SBT to build the Verilator C++ model?

On non-GNU systems, `make` is typically the traditional non-GNU `make` whereas Spinal appears to expect GNU-make (i.e. `gmake`).
Currently I'm working around this by having a wrapper which maps `make` to `gmake` and putting that wrapper in front of `PATH`. However, that is a bit fuckly:
```shell
➜ cat $HOME/bin/make
#!/bin/sh
exec gmake "$@"
```

Is there a way to pass the full path to `gmake`/`make` via a setting or environment variable?

Dolu1990, 2023-11-06:
> within SBT to build the Verilator C++ model?

it isn't SBT which build the verilator, it is the scala code itself.

Which scala code are you running through SBT ?

Tectu, 2023-11-06:
The "issue" can be reproduced by just cloning the `SpinalTemplateSbt` project, firing up `sbt` and running: `runMain projectname.MyTopLevelSim`.

On a non-GNU system this errors because `make` is not GNU make:
```plain
sbt:projectname> runMain projectname.MyTopLevelSim
[info] running (fork) projectname.MyTopLevelSim 
[info] [Runtime] SpinalHDL v1.9.4    git head : 270018552577f3bb8e5339ee2583c9c22d324215
[info] [Runtime] JVM max memory : 14536.0MiB
[info] [Runtime] Current date : 2023.11.06 17:06:24
[info] [Progress] at 0.000 : Elaborate components
[error] [Thread-2] INFO net.openhft.affinity.Affinity - Using dummy affinity control implementation
[info] [Progress] at 0.156 : Checks and transforms
[info] [Progress] at 0.226 : Generate Verilog
[info] [Done] at 0.268
[info] [Progress] Simulation workspace in /usr/home/jbo/projects/spinalhdl_template/./simWorkspace/MyTopLevel
[info] [Progress] Verilator compilation started
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 60: Variable/Value missing from "export"
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 65: Invalid line type
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 66: Unknown modifier " GNU Make cannot build in directories containing spaces, build elsewhere"
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 66: Unknown modifier " '$(CURDIR"
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 66: Invalid line type
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 67: Invalid line type
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 88: Invalid line type
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 90: Invalid line type
[error] Exception in thread "main" java.lang.AssertionError: assertion failed: Verilator C++ model compilation failed
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 109: Invalid line type
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 111: Invalid line type
[error] 	at scala.Predef$.assert(Predef.scala:223)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 144: Invalid line type
[error] 	at spinal.sim.VerilatorBackend.$anonfun$compileVerilator$21(VerilatorBackend.scala:686)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 147: Invalid line type
[error] 	at spinal.sim.VerilatorBackend.cacheSynchronized(VerilatorBackend.scala:75)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 152: Invalid line type
[error] 	at spinal.sim.VerilatorBackend.compileVerilator(VerilatorBackend.scala:649)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 156: Invalid line type
[error] 	at spinal.sim.VerilatorBackend.<init>(VerilatorBackend.scala:770)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 159: Invalid line type
[error] 	at spinal.core.sim.SpinalVerilatorBackend$.apply(SimBootstraps.scala:144)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 160: Invalid line type
[error] 	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:1003)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 168: Invalid line type
[error] 	at spinal.core.sim.SpinalSimConfig.compileCloned(SimBootstraps.scala:934)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 169: Invalid line type
[error] 	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:897)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 171: Invalid line type
[error] 	at projectname.MyTopLevelSim$.delayedEndpoint$projectname$MyTopLevelSim$1(MyTopLevelSim.scala:7)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 172: Invalid line type
[error] 	at projectname.MyTopLevelSim$delayedInit$body.apply(MyTopLevelSim.scala:6)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 204: Invalid line type
[error] 	at scala.Function0.apply$mcV$sp(Function0.scala:39)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 214: Invalid line type
[error] 	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 219: Invalid line type
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 261: Invalid line type
[error] 	at scala.App.$anonfun$main$1$adapted(App.scala:80)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 288: Invalid line type
[error] 	at scala.collection.immutable.List.foreach(List.scala:431)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 297: Invalid line type
[error] 	at scala.App.main(App.scala:80)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 298: Invalid line type
[error] 	at scala.App.main$(App.scala:78)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 300: Invalid line type
[error] 	at projectname.MyTopLevelSim$.main(MyTopLevelSim.scala:6)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 302: Invalid line type
[error] 	at projectname.MyTopLevelSim.main(MyTopLevelSim.scala)
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 304: Invalid line type
[info] make: "/usr/local/share/verilator/include/verilated.mk" line 328: Invalid line type
[... cut ...]
[info] [Progress] Simulation workspace in /usr/home/jbo/projects/spinalhdl_template/./simWorkspace/MyTopLevel
[info] [Progress] Verilator compilation started
[info] [Progress] Verilator compilation done in 9490.887 ms
[info] [Progress] Start MyTopLevel test simulation with seed 985360247
[info] [Done] Simulation done in 22.594 ms
[success] Total time: 11 s, completed Nov 6, 2023 5:07:40 PM
```
Hence I think the only thing required would be to somehow & somewhere being able to specify the path to the `make` binary that is being invoked.

Dolu1990, 2023-11-06:
Ahh got it. So, this make command is comming from : 
https://github.com/SpinalHDL/SpinalHDL/blob/7314a3d140b89c5ddc1f8d46274c28c36b98af46/sim/src/main/scala/spinal/sim/VerilatorBackend.scala#L686
https://github.com/SpinalHDL/SpinalHDL/blob/7314a3d140b89c5ddc1f8d46274c28c36b98af46/sim/src/main/scala/spinal/sim/VerilatorBackend.scala#L689

Could add a environnement variable in that kind of way : 
https://github.com/SpinalHDL/SpinalHDL/blob/7314a3d140b89c5ddc1f8d46274c28c36b98af46/sim/src/main/scala/spinal/sim/VerilatorBackend.scala#L524

and replace the s"make ..." by s"$makeEnv ..."

Readon, 2023-11-06:
currently we have a lot of incompatibility with non-GNU system. Not only on make, but also verilator, due to "sh" is introduced.

Readon, 2023-11-06:
Do you refer non-GNU system == Windows? or MacOS?

Tectu, 2023-11-06:
@Dolu1990 Yeah exactly, something like that! I assume you could assign it a default value so existing consumers are unaffected by this change/feature.

> currently we have a lot of incompatibility with non-GNU system. Not only on make, but also verilator, due to "sh" is introduced.

`sh` would be more of a POSIX thing. You can have a non-GNU system that is still POSIX compatible.
So far the GNU make issue mentioned here is the only problem I encountered.

> Do you refer non-GNU system == Windows? or MacOS?

In my case it would be FreeBSD.

commit 34876748 (2023-11-07): fix #1239 Add SPINAL_MAKE_CMD to change make commands to something else
closed 2023-11-07 by Dolu1990