SpinalHDL/SpinalHDL #1248: How to understand the Simulated behavior in SpinalHDL?

xiaochuang-lxc, 2023-11-18:
There is a simple example:
`import spinal.core._
import spinal.lib._
import spinal.core.sim._
case class bootTest() extends Component {
  val pc=out UInt(8 bits) setAsReg() init(0)
  val inst=in UInt(8 bits)
  pc:=pc+1
}

object bootTestApp extends App {
  SimConfig.withFstWave.compile(bootTest()).doSim{dut=>
    dut.clockDomain.forkStimulus(10)
    fork{
      while(true){
        dut.clockDomain.waitSampling()
        println(s"time:${simTime()}\t value:${dut.pc.toInt}\t clk:${dut.clockDomain.clockSim.toBoolean}")
        dut.inst#=dut.pc.toInt+1
        sleep(0)
        println(s"time:${simTime()}\t value:${dut.pc.toInt}\t clk:${dut.clockDomain.clockSim.toBoolean}")
      }
    }
    dut.clockDomain.waitSampling(10)
  }
}`
 The purpose of the simulation is to feedback the inst port at the same cycle, but the result mismatch:
[image]
 the inst value is one cycle delayed .
I write two println in simulation,and notice that:
`time:170	 value:0	 clk:true
time:170	 value:1	 clk:true` 
Before sleep(0),the clk value has changed,but the pc is still the old value.
Systemverilog has blocking assignment and non-blocking assignment in simulation,and the clock driver usually use the blocking assignment in simulation environment。
My Question is：
How to understand the Simulated behavior in SpinalHDL? 
The relationship between the simualtion and verilator rtl model ?
Before Sleep(0),the clock rising edge has happened,why the pc value is sitll 0?

Dolu1990, 2023-11-18:
Hi,

dut.clockDomain.waitSampling() will wakeup in the delta cycle where the clock just did rising edge, but the flipflop didn't had yet updated their output value. (like in VHDL / verilog)

Also, when you assign a signal in simulation using \#= , the same kinda of thing happen, the write will only be visible in the next delta cycle (not the current one)

those two feature combined allows to write a testbench which can behave syncronously like flipflops and avoid race conditions

xiaochuang-lxc, 2023-11-18:
How to understand the delta cycle? Is it equal to time slot in systemverilog?
[image]
In SystemVerilog ,the execution sequence of time solt excution is sample,rtl behavior,simualtion region. As the same simualtion in systemverilog:
```
`timescale 1ps / 1ps
module dut(
    input clk,
    input rst,
    input logic [7:0] inst,
    output logic [7:0] pc
);
always @(posedge clk or posedge rst) begin
    if(rst)
        pc<=0;
    else
        pc<=pc+1;
end

endmodule

program sim(input clk,input [7:0] pc,output logic [7:0] inst );

initial begin
 forever begin
 @(posedge clk)
 inst<=pc+1;
 $display("rime=$%t\tclk=%d\tpc=%d\tinst=%d",$time(),clk,pc,inst);
 end
end

endprogram

module top();
    logic clk;
    logic rst;
    logic [7:0] pc,inst;
     initial begin
        clk=0;
        rst=1;
        #10;
        rst=0;
        #200;
        $stop();
    end
    initial begin
        forever begin
            #10; clk=!clk;
        end
    end
    dut dutInst(.clk(clk),.rst(rst),.pc(pc),.inst(inst));
    sim simInst(.clk(clk),.pc(pc),.inst(inst));
endmodule
```
after sampling the rising clk edge,the pc has updated.
I'm not understand the document https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Simulation/engine.html .
can you give more details about the Simulated behavior in SpinalHDL just like systemverilog spec?
Thanks

andreasWallner, 2023-11-25:
@xiaochuang-lxc 
A delta cycle is a simulation step that is done without increasing the simulation time. The term is taken from the VHDL simulation model where those are used to handle combinational propagation through the circuit. (simplified) In VHDL all statements that update signal state for a timestep are processed before the state is updated atomically.

That is what the SpinalSim engine mimics with inputs into the HDL before it lets the simulator do a timestep. It will keep calling scheduled tasks as long as there are any for the current timestep, In the SpinalSim context a delta cycle happens when all scheduled tasks are processed. What they change in the hardware is buffered so that back-to-back evaluations of a signal see consistent state. Once all scheduled tasks are processed, signals in the simulator are updated. This may cause some tasks to be triggered, so that another delta cycle is processed. When there are none it let's the simulator advance to the next time.

This is the reason for the behavior that you initially observed: The write to `dut.inst` was buffered (again, so that during processing in the testcase you see consistent state). You then forced a write and a new delta cycle with `sleep(0)`.

If you remove the `sleep(0)` from your code, the this verilog is the equivalent of your initial SpinalSim code:
```scala
 @(posedge clk)
 $display("rime=$%t\tclk=%d\tpc=%d\tinst=%d",$time(),clk,pc,inst);
 inst<=pc+1;
 $display("rime=$%t\tclk=%d\tpc=%d\tinst=%d",$time(),clk,pc,inst);
 end
```
Which shows the same behavior that you see from SpinalSim: the same value is printed twice since updates are buffered till the end of a process.

> can you give more details about the Simulated behavior in SpinalHDL

If you have a concrete question I'm sure we'll be able to provide more details.

xiaochuang-lxc, 2023-11-25:
I'm a little understand.Thanks!

closed 2023-11-25 by xiaochuang-lxc