SpinalHDL/SpinalHDL #1251: SpinalConfig.generate does not work as expected When define two modules with implicit parameter in "generate" func

lycfly, 2023-11-24:
I get "moduleA.v" generated in rtl folder when I run code below, but what I want to generate actually is "moduleB.v"

```scala
case class Configs()

class moduleA()(implicit conf: Configs) extends Component {
  val io = new Bundle {
    val ina = in Bool()
    val outb = out Bool()
  }
  noIoPrefix()
  io.outb := io.ina

}
class moduleB(implicit conf: Configs) extends Component {
  val io = new Bundle {
    val ina = in Bool()
    val outb = out Bool()
  }
  noIoPrefix()
  io.outb := ~io.ina
}

object moduleB_inst {
  def main(args: Array[String]): Unit = {
    SpinalConfig(
      mode = Verilog,
      targetDirectory = "rtl"
    ).generate({
        implicit val conf = Configs()
        val modA = new moduleA()
        val modB = new moduleB()
        modB
       })
  }.printPruned()
}
```

And when I modify the code in generate() to:

```scala
object moduleB_inst {
  def main(args: Array[String]): Unit = {
    SpinalConfig(
      mode = Verilog,
      targetDirectory = "rtl"
    ).generate({
        implicit val conf = Configs()
        val modB = new moduleB()
        val modA = new moduleA()
        modB
       })
  }.printPruned()
}
```
I get "moduleB.v" generated correctly. What changed when I swap these two line?
My scala Version  is '2.12.18' and spinal Version is '1.9.3'.

andreasWallner, 2023-11-24:
Hi lycfly,

The first component that is instantiated registers itself as the toplevel component, which is later used e.g. for the name of the module. (it has to so that components inside it can add themselves as subcomponents).
`modB` in your first example is never really inserted into the internal data model. 
I think we should add an error for this corner case.

What's your use-case, what would you like to achieve in the end?

lycfly, 2023-11-25:
Hi andreasWallner,
Actually I define an SpinalEnum in moduleA and some elements is dynamically added when moduleA is instanced according to the Config. ModuleB will use this dynamically added elements of Enum object so I need to instance moduleA before.
I am just testing whether ModuleB can get this elements, so I write this code in the generate function and find it does not generate the rtl code as I expected.

andreasWallner, 2023-11-28:
@lycfly As it is at the moment that doesn't work (for the reason I desribed above) - is it possible for you to move the "common code" outside of ModuleA so that both can call it?

lycfly, 2023-11-29:
> @lycfly As it is at the moment that doesn't work (for the reason I desribed above) - is it possible for you to move the "common code" outside of ModuleA so that both can call it?

Yes, these two modules can be instantiated in an top module.
The code above is for test, and I was just confused about the generate function's behavior.
Thanks for your reply and I will avoid this usage next time.

commit e24f9768 (2024-01-12): fix #1251 Catch multiple toplevel