SpinalHDL/SpinalHDL #1259: Issues with verilator 5

andreasWallner, 2023-12-14:
After discussing some verilator behavior with Gitter user solater we came up with this example that has some (I'd say)  weird behavior with verilator 5:
```scala

case class Repro() extends Component {
  val io = new Bundle {
    val address = in port UInt(1 bit)
    val value = in port UInt(8 bit)
    val cond = in port Bool()
  }
  val init = Seq.fill(2){U(0)}
  val mem = Mem(UInt(8 bits), init)
  when(io.cond) {
    mem(io.address) := io.value
    report(Seq("writing ", io.address, " = ", io.value))
  }
}

object Repro extends App {
  val dut = SimConfig.withFstWave.compile{
    val dut = Repro()
    dut.mem.simPublic()
    dut
  }
  for(_ <- 0 until 100)
    dut.doSim { dut =>
      dut.io.cond #= false
      dut.clockDomain.fallingEdge()

      dumpData()

      for(i <- 0 until 2) {
        assert(dut.mem.getBigInt(i) == 0)
      }

      def dumpData(): Unit = {
        println(simTime())
        for (ii <- (0 until dut.mem.wordCount).grouped(8)) {
          print(f"${ii(0)}%04d  ")
          for (i <- ii) {
            print(f"${dut.mem.getBigInt(i)}%08x ")
          }
          println()
        }
      }
    }
}
```
Which normally simulates fine, but will fail randomly like so:
```
[info] [Progress] Start Repro test_1 simulation with seed 1906628651
[info] 0
[info] 0000  00000000 00000000 
[info] [Done] Simulation done in 6.013 ms
[info] [Info] Test 'test' was reallocated as 'test_2' to avoid collision
[info] [Progress] Start Repro test_2 simulation with seed 1361515301
[info] NOTE writing 0 = 89
[info] 0
[info] 0000  00000089 00000000 
[info] [Error] Simulation failed at time=0
[error] Exception in thread "main" java.lang.AssertionError: assertion failed
[error]         at scala.Predef$.assert(Predef.scala:156)
[error]         at spinal.core.package$.assert(core.scala:497)
[error]         at vexiiriscv.scratchpad.Repro$$anonfun$6$$anonfun$apply$mcVI$sp$2$$anonfun$apply$4.apply$mcVI$sp(Play.scala:217)
[error]         at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:160)
[error]         at vexiiriscv.scratchpad.Repro$$anonfun$6$$anonfun$apply$mcVI$sp$2.apply(Play.scala:216)
[error]         at vexiiriscv.scratchpad.Repro$$anonfun$6$$anonfun$apply$mcVI$sp$2.apply(Play.scala:211)
[error]         at spinal.core.sim.SimCompiled$$anonfun$doSimApi$2.apply$mcV$sp(SimBootstraps.scala:611)
[error]         at spinal.sim.SimManager.spinal$sim$SimManager$$threadBody$1(SimManager.scala:230)
[error]         at spinal.sim.SimManager$$anonfun$2.apply$mcV$sp(SimManager.scala:233)
[error]         at spinal.sim.SimThread$$anonfun$1.apply$mcV$sp(SimThread.scala:93)
[error]         at spinal.sim.JvmThread.run(SimManager.scala:51)
```
It seems that sometimes clock triggered processes are triggered at time 0, before we run our initial assignments.

I have yet to check in more detail if this is just an order thing in our simulation driver.

andreasWallner, 2023-12-16:
yes, that the exact same issue...

g0t00, 2024-03-01:
I also stumpled upon this. This currently sadly makes verilator 5 completely unusable with Spinal, as asserts just randomly trigger at timepoint 0. See the simplified MWE:
```scala
object AssertsMWE extends App {
  SimConfig
    .compile(new Component {
      for (_ <- 0 until 100)
        assert(RegNext(RegInit(True), init = True))
    })
    .doSim { dut =>
      dut.clockDomain.forkStimulus(10)
      dut.clockDomain.waitSampling(10)
    }
}
```
Is there a current workaround for this (other than downgrading verilator to version 4). e.g. Disabling asserts for time = 0 or sth?
Internally we had thought about doing something ORing a big enough register to abuse the the randomized init state. this seems to work but feels hacky as hell:
```scala
object AssertsMWE extends App {
  SimConfig
    .compile(new Component {
      for (_ <- 0 until 100) {
        val guard = RegNext(RegInit(B(False, 100)), init = B(False, 100)).orR
        assert(guard || RegNext(RegInit(True), init = True))

      }
    })
    .doSim { dut =>
      dut.clockDomain.forkStimulus(10)
      dut.clockDomain.waitSampling(10)
    }
}
```

andreasWallner, 2024-03-04:
@g0t00 Dolu and I discussed a possible workaround with a reset sequence that would work for both synchronous and async resets. So progress so far on that.

If you know that you have only have synchronous resets then you can remove the `--x-initial-edge` option for the verilator execution, as mentioned in the linked verilator issue.

Dolu1990, 2024-03-04:
For the assert statement, we could maybe add (optionaly) some condition to not allow them to trigger at time 0

g0t00, 2024-03-04:
> For the assert statement, we could maybe add (optionaly) some condition to not allow them to trigger at time 0

For me this sounds like the cleanest solution for now. If thats possible would be great.

Dolu1990, 2024-03-04:
Seems good with that fix.
Kinda dirty workaround, but hooo well, let's focus on getting it to work XD

g0t00, 2025-07-07:
@Dolu1990 
So we funnily stumbled on this issue again, in the memory context.
So like in the original context RAM with initial content has a chance to randomly overwrite some of the content in simulation.
We currently use a guard condition with a reducing and like above. But this feels like a very dirty workaround. Any ideas on a more elegant fix?
If not shall I make a PR to make this at least permanent?

Dolu1990, 2025-07-09:
I don't have any fix. PR welcome :D

commit 9dd5285a (2024-03-04): Fix #1259 (verilator 5 asserts at time 0)