SpinalHDL/SpinalHDL #1262: Signals in simulation pruned

jjjt-git, 2023-12-15:
In my current project, I have a bunch of diagnostic signals inside of my modules, that are not connected to anything on purpose.
They are used to provide faster access to informations in traces or to allow the simulation-env to emulate required behaviour.
As expected these signals are pruned, which is what I want for the final implementation of my modules.

Is it possible, to disable signal pruning for a simulation, without modification to the component.

Thanks in advance,
Jacob

andreasWallner, 2023-12-17:
Hi,
Normally named signals are not removed (if `removePruned` is not `true` which is the standard).
If something is removed then you may give it a name (e.g. if it's unnamed or weakly named) via calling `setName` of the signal.
Signals also have a `dontSimplifyIt` member function that prevents optimization of them.

You can do both w/o modifying the component by calling those during generation, e.g. like so:
```scala
val dut = SimConfig.withFstWave.compile {
  val c = MyComponent()
  // depending on what you need:
  c.subcomponent.some_signal.setName("blub")
  c.subcomponent.some_signal.dontSimplifyIt()
  c
}

Dolu1990, 2023-12-18:
Normaly, as long as the signal has a name, it won't be removed

jjjt-git, 2023-12-20:
Thanks, now it works.

closed 2023-12-20 by jjjt-git