SpinalHDL/SpinalHDL #1264: Async programming in simulation

KireinaHoro, 2023-12-19:
I'm trying to build a Axi4 bus master simulation agent that looks roughly like [this](https://gist.github.com/KireinaHoro/41ede743e05f016268f5752c6cd76b5b).  The agent works ok so far, but in the actual test bench it is very quickly turning into a [callback hell](http://callbackhell.com/):

```scala
  dut.doSim { dut =>
    SimTimeout(1000)
    dut.clockDomain.forkStimulus(period = 4) // 250 MHz

    val master = Axi4Master(dut.io.s_axi, dut.clockDomain)
    // write global config bundle
    val rxBlockCycles = 100

    master.write(0, BigInt(rxBlockCycles).toByteArray) { () =>
      master.read(0, 8) { data =>
        assert(BigInt(data.reverse).toInt == rxBlockCycles, "global config bundle mismatch")

        master.read(0x1000, 8) { data =>
          val received = BigInt(data.reverse)
          println(f"Received status register: $received%#x")

          // ... more simulation steps follow
        }
      }
    }

    dut.clockDomain.waitActiveEdgeWhere(master.idle)
  }
```

Is it possible to write the simulation testbench asynchronously with the help of `Future[T]` and `Promise[T]`?  In this way it would be possible to write something a bit nicer:

```scala
async {
  await { master.write(0, BigInt(rxBlockCycles).toByteArray) }
  val data = await { master.read(0, 8) }
  assert(BigInt(data.reverse).toInt == rxBlockCycles, "global config bundle mismatch")
  val statusRegister = await { master.read(0x1000, 8) }
  println(f"Received status register: $statusRegister%#x")
  // ... more simulation steps follow
} onComplete { case Success(_) =>
  simSuccess()
}
```

I've tried to change `Axi4Master.read` and `write` to return `Future`s by internally having a `Promise` and calling `success`, instead of using a callback; I'm not sure what's wrong, but the futures never seem to get scheduled and the `waitActiveEdgeWhere` just times out.  What's the way ahead here?

Dolu1990, 2023-12-22:
Hi,

Yes callback hell should be avoided.
Here is an example of an Apb3 driver : 
https://github.com/SpinalHDL/SpinalHDL/blob/68b6158700fc2440ea7980406f927262c004faca/lib/src/main/scala/spinal/lib/bus/amba3/apb/sim/Apb3Driver.scala#L7

Where for instance you can do asyncronous :
```scala
fork{
   val x = apb.read(address)
   apb.write(address2, x + 1)
}
```

So the idea is more to fork thread explicitly

Here is an example another driver which this time support being called from multiple simulation thread at the same time : 
https://github.com/SpinalHDL/SpinalHDL/blob/68b6158700fc2440ea7980406f927262c004faca/lib/src/main/scala/spinal/lib/bus/bmb/sim/BmbDriver.scala#L12

But will only allow one outstanding transaction at the time

Finaly, a Driver which handle multiple thread having multiple outstanding pending transaction is : 
https://github.com/SpinalHDL/SpinalHDL/blob/68b6158700fc2440ea7980406f927262c004faca/lib/src/main/scala/spinal/lib/bus/tilelink/sim/MasterAgent.scala#L44

So overal, with the fork{ xx } you can fork thread, and you can also wait on the completion of another thread via someThread.join(), it should be enough to build it all ^^

KireinaHoro, 2024-01-05:
Good point.  Although I realized that the `StreamMonitor` and `StreamDriver` are all thread-based, which means it would be ok to block the current simulation driver thread.  I now have something like this:

```scala
def read(address: BigInt, totalBytes: BigInt, id: Int = 0, burst: Axi4Burst = Incr, len: Int = 0, size: Int = maxSize) = {
    val done = new AtomicReference[Array[Byte]](null)
    readCB(address, totalBytes, id, burst, len, size)(done.set)
    clockDomain.waitActiveEdgeWhere(done.get() != null)
    done.get()
  }

  def readCB(address: BigInt, totalBytes: BigInt, id: Int = 0, burst: Axi4Burst = Incr, len: Int = 0, size: Int = maxSize)(callback: Array[Byte] => Unit) = {
```

The only issue is that this cannot be used together with `delayed` or other callback-based APIs; I'm working around that with something like:

```scala
fork {
  sleep(20)
  master.read(...)
}
```

The full implementation is [here](https://gist.github.com/KireinaHoro/4f9d6fcec0b4c9ffd63cb0b3899e50ee) with some sample usage [here](https://gist.github.com/KireinaHoro/c765badefa1f96a8c2e3832653c6c558).  @Dolu1990 can you take a glance at this?  I can work on a PR if we'd need something like this in `spinal.lib`.

Dolu1990, 2024-01-11:
Hi,

> AtomicReference

This shouldn't be used, all the thread in SpinalSim work in their own "sandbox". You don't have to wory about multiple thread running at the same time, as they are always scheduled to have only one running at a given time on your PC.
You can for instance use the SimMutex instead.

> delayed or other callback-based APIs;

Right, the callback API is there to provide high speed stuff, but as you did, you would need to fork threads from it.
In general the callback API is fine for simple things, but become very verbose not nice when it is about complicated time based tasks.

> writeCB

What the CB mean ?

> I can work on a PR if we'd need something like this in spinal.lib.

Yes a controllable AXI master would be needed for sim.
The question then is more about how "feature full" it would need to be.

It seems that your code do it right : 

```scala
  val rQueue = Array.fill(idCount)(mutable.Queue[Axi4R => Unit]())
  val bQueue = Array.fill(idCount)(mutable.Queue[Axi4B => Unit]())
```

So yes, why not ^^

KireinaHoro, 2024-01-11:
Hi,

> You can for instance use the SimMutex instead.

So it's ok to keep a simple reference (unsynchronised) and just have the calling thread sleep on a SimMutex?

> What the CB mean ?

It means CallBack.  `writeCB` and `readCB` is intended as the non-blocking version that takes a callback, which would allow concurrent operations (e.g. read and write on two different busses at the same time).

> how "feature full" it would need to be

The current implementation allows narrow bursts and multi-transaction reads (to facilitate memory bulk read operation), but not uncommon burst modes (only INCR).  Are there commonly used features missing?

Dolu1990, 2024-01-12:
> So it's ok to keep a simple reference (unsynchronised) and just have the calling thread sleep on a SimMutex?

Yes, you never have to worry about software related race condition

Dolu1990, 2024-01-12:
>  (only INCR)

That's good i think ^^

PR #1288 (2024-01-22): AXI and AXI-Stream simulation bus masters

Closes #1264

# Context, Motivation & Description

Simulation bus source/sinks with better control for behavioural testbenches in SpinalSim.

# Checklist

- [ ] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`
closed 2024-01-22 by Dolu1990