SpinalHDL/SpinalHDL #1265: BusSlaveFactory driveFlow does not honor byteEnable

KireinaHoro, 2023-12-21:
It seems like the `driveFlow` function in `BusSlaveFactory` does not honour `byteEnable`:

https://github.com/SpinalHDL/SpinalHDL/blob/68b6158700fc2440ea7980406f927262c004faca/lib/src/main/scala/spinal/lib/bus/misc/BusSlaveFactory.scala#L512-L516

I had the impression that the `valid` bit in the flow should not be on when `byteEnable` was all zero for the width of the flow. This scenario is tricky however, since it's not clear how to properly handle partial `byteEnable`s on a `Flow`:
- do we trigger the Flow but with false data on those that the byte is not enabled (similar to current situation), or
- do we drop the write completely?

A real-scenario where this would matter:  I am using a Axi4SlaveFactory after an AXI width converter.  For a wide write request with some bytes disabled with `strb`, the AXI width converter issues requests that contains beats with zero `strb`.  This then falsely triggers the downstream `Flow`.

Dolu1990, 2023-12-29:
Flow them self aren't made for byteenable, could add additional versions which would work on specific payload type.

> A real-scenario where this would matter: I am using a Axi4SlaveFactory after an AXI width converter. For a wide write request with some bytes disabled with strb, the AXI width converter issues requests that contains beats with zero strb. This then falsely triggers the downstream Flow.

Yes you are right, all bytemask being low could / should be used to not have valid = '1'

which transactions make the axi width converter make all strb going low in some transactions (example) ?

KireinaHoro, 2024-01-03:
I'm not actually using the built-in width converter / unburstifier / ..., but [an external IP](https://github.com/alexforencich/verilog-axi/blob/master/rtl/axi_adapter.v) instead.  An example of the all low strb transaction is as follows in a downsize from 512 to 64:

[image]

You can see that `s_axi_wstrb` is `0xff000000`, which gets translated to the 4th beat having `m_axi_wstrb` being `0xff`.  The other beats have `wready && wvalid` but `wstrb == 0`.

Dolu1990, 2024-01-08:
I would say, you may get peripherals which trigger actions when a register is written and on which byte mask make no sense, for instance the PLIC interrupt reservation.
So there is a few way to fix the situation i would say : 
- patching every peripherals to not trigger side effect when byte mask is 0, but that seems the worst option to me
- but more fondamentaly, i think the main issue is the s_axi transaction we can see on the wave, with its awsize of 64 bytes.

What kind of master generates it ?

KireinaHoro, 2024-01-18:
The offending request comes from the [Xilinx XDMA bridge](https://docs.xilinx.com/r/en-US/pg195-pcie-dma); it issues narrow writes with 64B AWSIZE but only partial WSTRB (in this case 8 bytes).  Per my understanding this is allowed by AXI (see [this thread](https://community.arm.com/support-forums/f/soc-design-and-simulation-forum/2046/the-usage-of-wstrb-signal): WSTRB is allowed to go low during a transaction to signal only some bytes are valid).

Regarding a possible solution, I agree that it's bad to change behaviour of existing peripherals that expect the current behaviour.  I'm working on a pull request to add an optional parameter to the method (that defaults to the old behaviour); what do you think?

Dolu1990, 2024-01-22:
Ok for me ^^

Else, a general fix would be to implement an AXI bridge which filter this kind of accesses, so you would just have to put the bridge in between the width adapter and peripherals.

PR #1289 (2024-01-22): driveStream on BusSlaveFactory

Closes #1265

# Context, Motivation & Description

The bus slave factory `driveFlow` does not currently honour `byteEnable`.  As discussed, this PR adds an optional mode where it checks for all-zero `byteEnable` and does not trigger valid on the flow.  For backwards compatibility we keep the old behaviour as default.

It also adds `driveStream` and `readStreamBlockCycles` as two blocking-mode operations on the bus control class.

# Checklist

- [ ] Unit tests were added
- [x] API changes are or will be documented:
  - using Scaladoc comments: `/** */`
closed 2024-01-22 by Dolu1990